gemu2015/Sonoff-Tasmota #1: sonoff/xsns_91_sml.ino Q3B

futschikato, 2019-01-03:
Hallo gemu

Bitte die zwei Werte beim Q3B wieder ändern.
von
#define SFAC1 1000
#define SFAC2 1

in
#define SFAC1 100
#define SFAC2 100

Gruss ow

Ormerod2, 2019-05-12:
The Q3B reports numbers with wrong decimals. I had to change ':
#if METER==Q3B
#define METERS_USED 1
struct METER_DESC const meter_desc[METERS_USED]={
  [0]={0,'s',0,SML_BAUDRATE,"SML"}};
const uint8_t meter[]=
//0x77,0x07,0x01,0x00,0x01,0x08,0x01,0xff
"1,77070100010800ff@**10000000**," D_TPWRIN ",KWh," DJ_TPWRIN ",4|"
//0x77,0x07,0x01,0x00,0x02,0x08,0x01,0xff
"1,77070100020801ff@100," D_TPWROUT ",KWh," DJ_TPWROUT ",4|"
//0x77,0x07,0x01,0x00,0x01,0x07,0x00,0xff
"1,77070100010700ff@**1**," D_TPWRCURR ",W," DJ_TPWRCURR ",0";
#endif

tested on 2 Meters, both same result. I can provide pictures as well if needed.

gemu2015, 2019-05-13:
die Definition habe ich von einem User bekommen.
dass sich bei current der Wert auf 1 verändert ist in Ordnung, das hatte mit einer Anpassung der scaler im Decoder zu tun.
aber die 1 Million ist sehr suspekt. das würde ja eine Auflösung von 1 Milliwatt bedeuten.
kannst du mir mal einen Konsolen Dump des Hex Outputs dieses Zählers schicken ?

Ormerod2, 2019-05-13:
wenn du mir sagst wie, kann ich das gerne machen. Google hat mir jetzt auf die schnelle nix ausgeworfen.

gemu2015, 2019-05-14:
hast du eine ältere Version ? das habe ich erst kürzlich repariert.
mit sensor95 d1  sollte dein Zähler einen Hexdump machen.
diesen Dump in einen Texteditor kopieren und dort nach dem entsprechenden String suchen => 77070100010800ff
dann diesen Teil und noch ca 2 weitere Zeilen mir schicken

Ormerod2, 2019-05-14:
ja lag wohl an der nicht aktuellen Version. Hier der Hex:

00:01:16 : 77 01 0b 06 45 53 59 01 02 72 68 4f 00 01 72 62 01 65 0e 9e 8d 25 76 
00:01:16 : 77 07 81 81 c7 82 03 ff 01 01 01 01 04 45 53 59 01 
00:01:17 : 77 07 01 00 01 08 00 ff 01 01 62 1e 52 fc 69 00 00 00 33 2f 2b 58 49 01 
00:01:17 : 77 07 01 00 01 08 01 ff 01 01 62 1e 52 01 65 00 21 8a cb 01 
00:01:17 : 77 07 01 00 01 08 02 ff 01 01 62 1e 52 01 65 00 00 00 82 01 
00:01:17 : 77 07 01 00 01 07 00 ff 01 01 62 1b 52 fe 55 00 00 98

gemu2015, 2019-05-15:
ok danke, der Zähler liefert tatsächlich eine irrwitzige Auflösung und eine interne Skalierung von 10000. habe die fehlenden internen Skalierungen (scaler) eingebaut.
wenn du jetzt nochmal kompilierst und testest kannst du mir die dann gefunden richtigen Skalierungen für die SML Definition durchgeben und dich werde den Descriptor korrigieren.

Ormerod2, 2019-05-15:
Also immer noch um etwas flasch.

Zählerstand in Sonoff IST: 219930.1882 KWh  SOLL: 21993.0188 
Verbrauch in Sonoff IST: 4 W SOLL 400W

Sprich so wird es dann korrekt angezeigt:
#if METER==Q3B
#define METERS_USED 1
struct METER_DESC const meter_desc[METERS_USED]={
  [0]={0,'s',0,SML_BAUDRATE,"SML"}};
const uint8_t meter[]=
//0x77,0x07,0x01,0x00,0x01,0x08,0x01,0xff
"1,77070100010800ff@1000," D_TPWRIN ",KWh," DJ_TPWRIN ",4|"
//0x77,0x07,0x01,0x00,0x02,0x08,0x01,0xff
"1,77070100020801ff@100," D_TPWROUT ",KWh," DJ_TPWROUT ",4|"
//0x77,0x07,0x01,0x00,0x01,0x07,0x00,0xff
"1,77070100010700ff@1," D_TPWRCURR ",W," DJ_TPWRCURR ",0";
#endif

gemu2015, 2019-05-16:
ok habe den Q3B Descriptor angepasst

closed 2019-05-16 by gemu2015