gemu2015/Sonoff-Tasmota #2: missing renderer.h in universal branch

ozett, 2019-01-04:
hi,
i came over from the [tasmota-thread](https://github.com/arendst/Sonoff-Tasmota/issues/2915#issuecomment-450739474) to try the rfid-pn532 driver.
as you mentioned to copy only the pn532.ino and the lib over,
i did this but i got errors. i had arduino-ide1.611 comipling my tasmota 6.x fine, 
but only copying lib and your .ino showed errors.

[image]

i took your complete universal-branch,
made a new arduino-ide v1.88, made the portable-dir and put all your source and libs in (as i alyway normally do), but got a different error.

[image]

```
In file included from C:\Users\olaf\AppData\Local\Temp\arduino_build_1307\sketch\ILI9488.cpp:18:0:

D:\daten\olaf.arduino\arduino-1.8.8.qmu2015\portable\sketchbook\libraries\ILI9488-master/ILI9488.h:27:22: fatal error: renderer.h: No such file or directory

 #include <renderer.h>

                      ^

compilation terminated.

```

may you have a hint for me?

ozett, 2019-01-04:
for my first version, i made a mistake:
first i used the "only" the xsns_40_pn532_ic,
and "not" from universal-branch.

i corrected this, and found the xsns93_pn532.
i used this also in my existing tasmota and now i compiled fine.

[image]

(the renderer.h is still missing in your universal branch on my arduino-install. may you still have a hint for me..)

ozett, 2019-01-04:
now i flashed the wemos-mini-d1 succesfull,
got all the pn532.inos and libs, in,
but you have to get me going, i guess..

[image]

what are my next steps to get the pn532 reader active?
configure i2s on the pins?
what about the reset-line (you mentioned in the tasmota-thread)? 
how can i configure this?

thanks for helping me out..

gemu2015, 2019-01-05:
i would recommend to use the pn532 branch. it is the newest tasmota release and an updated driver from andrethomas . the pn532_89 ist no longer supported.
this is work in progress anyhow but the more testers the better.
we try to avoid using the reset line currently, we are not sure if we succeed.
so simply connect SDA,SCL and it should run.
in the config you must enable 
#define USE_PN532_I2C
and if you want to test the learning feature also
#define LEARN_TAGS

ozett, 2019-01-05:
thanks for the info, i will try it now.

just for comparison: on espeasy i was used to do the following settings, to have my pn532 running (with reset) since 2016:

[image]

[image]

[image]

ozett, 2019-01-05:
@gemu2015 

> i would recommend to use the pn532 branch. 

just by the way:
i think there is a typo-mistake in the branch-spelling ?!

ozett, 2019-01-05:
maybe also for reference,
my settings on espeasy for my wiegand-fingerprint...

(sda is not needeed, but an impression of the actual interface lookout)
[image]

[image]

ozett, 2019-01-05:
👍  running...
[image]

ozett, 2019-01-05:
only as an idea: very helpful on the espeasy for debugging device-error is  the i2c-scan:

[image]

==>

🔦 

==>

[image]

ozett, 2019-01-05:
this one for completness of report:
on the espeasy the same TAG-ID is somehow different diplayed. (no problem for me)
but maybe worth a look?
[image]

ozett, 2019-01-05:
> this is work in progress anyhow but the more testers the better.
> we try to avoid using the reset line currently, we are not sure if we succeed.

just a an observation: back in the old-days of espeasy in 2016 the developer martinus build a tag-robot in his lab (as he posted in the forum), to have automatic testing for the need of the reset line.
as i have only readings from time-to-time, i guess i will not see if i need the reset-line

> and if you want to test the learning feature also
> #define LEARN_TAGS

just as an idea for learning-tags: 
its difficult to have the console at my housedoor to activate read-in.
on my fingerprint-reader i have to designate a tag, which activates the read-in of the next tag.
the first tag is made a master-tag to activate reading-in of the next tag, which is presented to the reader.
ne need for console-cmds.
also there can be a master-tag for deletion.. 📦

gemu2015, 2019-01-05:
espeasy only displays the first 4 hex bytes as decimal value, while tasmota displays all 7 hex bytes.

i really forgot that we had this mastertag thing already 20 years ago! thanks for remembering.

i implemented it in the source, you may add your master tags into a list there, (as many as you like)

console entry is not a problem however.
you enter the code in console where ever  you want, the reader waits for the next tag that is applied and learns it.

ozett, 2019-01-05:
> console entry is not a problem however. you enter the code in console where ever you want, the reader waits for the next tag that is applied and learns it.

just for my handling i would have to walk 2 min. up to the door. 
if anybody in the meantime (>2min 😄 ) holds a tag in front of the reader, 
he was faster than me.

if i would have master-tag, i could stand directly in front of the reader,
present the master tag to start teach-in, than the new tag.
if i wirte a rule, i could light up green,orange or red-lights on a ws2812 to visualize the process.
without any console-access  (or device for console) at all. 🎺 
what do you think? 
this method is the normal process on all my fingerprint (and face! )-reader. yep.
i mean: could it be an option?

gemu2015, 2019-01-05:
ok, then try it, its already implemented

you simply have to enter a master UID in the source code
currently 2 are foreseen one for adding und one for deleting

but you can add as many masters as you want.

ozett, 2019-01-06:
just for pleasure (or wiki?):
 a picture of my reader setup

[image]

gemu2015, 2019-01-07:
now you can read and write all sectors of the tag
you can format it to NDEF and add records.

// current commands
======================== 
a relay mode time => wait for tag to learn
A UID relay mode time => enter UID in list directly
AM UID relay mode time => enter UID in list directly as a learning master
AD UID relay mode time => enter UID in list directly as a deleting master
d index delete tag of index
D wait for tag to delete
e erase tag list
s index show list entry of index

// for these commands the tag must be placed onto the reader
// format a card to NDEF
f  => format to NDEF
// reformat an NDEF card to Mifare
fm =>reformat an NDEF tag to Mifare
// add a new NDEF record
n sector (1-15) type record  => (up to 38 asci chars (currently without spaces))
type can be found on the net NDEF Type definitions 
n=new entry nu=update existing entry

// write a block to Mifare card
w blocknr 16 bytes of hex date without spaces
// read a block from Mifare card
r blocknr 

in stead of the internal simulated EEPROM you can now use an i2c eeprom of type AT24C256

gemu2015, 2019-01-08:
RM6300 driver added

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015