gemu2015/Sonoff-Tasmota #3: EPaper 2.9 does not work

chkbln, 2019-01-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : nodemcu, EPaper2.9
- [ ] Provide the output of command ``status 0`` :

Hello gemu2015

I have started several times now to get an EPaper 2.9 up and running.
It does not work and I do not know why that is.
It works for you. Could you possibly provide me your branch that works for you with all the files I need to compile to provide.
In which files must be made changes. Only in the USER_CONFIG.H? The CS, MOSI and CLK terminals can be freely assigned on the ESP8266 e.g. D5, D6 and D7?
many Greetings

gemu2015, 2019-01-06:
Hello chkbln

use the branch =>displays

in user_config.h 
define the following =>
#define USE_DISPLAY
#define USE_DISPLAY_EPAPER29
#define USE_EPD_FONTS
#define USE_SOFTSPI
#define USE_SPI
#define SHOW_SPLASH
#undef USE_DISPLAY_MODES1TO5

now ist should compile.

in the web gui of tasmota define the pins for soft spi

SSPI_MOSI, SSPI_CS, SSPI_SCLK

assure that your epaper 29 board is jumpered to 3 wire spi

in the tasmota console =>

DisplayModel 5 

should give no error

now the screen should come up with a splash message

chkbln, 2019-01-07:
Hi,
I got it now.
Your settings for the User_config.h I have adopted.

Everything else commented out.
// #define USE_DISPLAY // Add I2C Display Support (+ 2k code)
// #define USE_DISPLAY_MODES1TO5 // Enable display mode 1 to 5 in addition to mode 0
// #define USE_DISPLAY_LCD // [DisplayModel 1] Enable Lcd display (I2C addresses 0x27 and 0x3F) (+ 6k code)
// #define USE_DISPLAY_SSD1306 // [DisplayModel 2] Enable SSD1306 Oled 128x64 display (I2C addresses 0x3C and 0x3D) (+ 16k code)
// #define USE_DISPLAY_MATRIX // [DisplayModel 3] Enable 8x8 matrix display (I2C adresseses see below) (+ 11k code)
// #define MTX_ADDRESS1 0x71 // [DisplayAddress1] I2C address of first 8x8 matrix module
// #define MTX_ADDRESS2 0x74 // [DisplayAddress2] I2C address of second 8x8 matrix module
// #define MTX_ADDRESS3 0x75 // [DisplayAddress3] I2C address of third 8x8 matrix module
// #define MTX_ADDRESS4 0x72 // [DisplayAddress4] I2C address of the fourth 8x8 matrix module
// #define MTX_ADDRESS5 0x73 // [DisplayAddress5] I2C address of fifth 8x8 matrix module
// #define MTX_ADDRESS6 0x76 // [DisplayAddress6] I2C address of sixth 8x8 matrix module
// #define MTX_ADDRESS7 0x00 // [DisplayAddress7] I2C address of seventh 8x8 matrix module
// #define MTX_ADDRESS8 0x00 // [DisplayAddress8] I2C address of eigth 8x8 matrix module

#endif // USE_I2C

// - SPI sensors ---------------------------------
// # define USE_SPI // SPI using library TasmotaTFT

// # ifdef USE_SPI
// #ifndef USE_DISPLAY
// #define USE_DISPLAY // Add SPI display support for 320x240 and 480x320 TFT
// #endif
// #define USE_DISPLAY_ILI9341 // [DisplayModel 4] Enable ILI9341 Tft 480x320 display (+ 19k code)
// #define USE_DISPLAY_EPAPER // [DisplayModel 5] Enable e-paper display (+ 19k code)
// # endif // USE_SPI

In the directory of the esp-epaper-29-ws-20171230-gemu-1.0 library, I have copied the library.properties file from the Arendst Branch. This will also clear the error message of the Arduino IDE that an invalid library was found.

Does the Arendst Branch also work or is your work not yet fully implemented?

Best regards and thank you

gemu2015, 2019-01-07:
HI,

 no Theo Arends did not find the time to adapt my Pull Request.  My branch contains many extensions to the current Tasmota release but the basic epaper also works in Tasmota main branch.
as my version works without problems in my home i do not intend to hold it up to date with the tasmota main branch.

PS strange my compilation of my display branch does compile without errors.

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015