gemu2015/Sonoff-Tasmota #5: Hager eHZ-K Werte werden falsch dargestellt

fleischtorte, 2019-03-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [x] Hardware used : D1 mini 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hallo Gemu,

erstmal vielen Dank für deine Arbeit. Leider habe ich Probleme mit meinem Zähler Hager eHZ-K.
Der Output liefert beim Parsen einen Zählerstand von 566.5290 und eine Einspeisung von 0.0010, laut Debug sollte er aber folgende Daten liefern:

> 
> 56 00 00 00 08 a5 01 -> 2213 KWh
> 
> 56 00 00 00 00 00 01 -> 0KWh

diese sind laut Zählerdisplay auch korrekt. Könnte da etwas beim Parsen schieflaufen? Allerdings wird die Zählernummer richtig angezeigt und übertragen.  

Konsole:

> {"Time":"2019-03-08T09:38:17","SML":{"Total_in":566.5290,"Total_out":0.0010,"Power_curr":0,"Meter_number":"Zählernummer"}}

Output sensor95 D1:

> 09:34:08 : 77 07 01 00 00 00 09 ff 01 01 01 01 0b Zählernummer 01 
> 09:34:08 : 77 07 01 00 01 08 00 ff 64 01 01 82 01 62 1e 52 03 56 00 00 00 08 a5 01 -> Zählerstand Tariflos
> 09:34:08 : 77 07 01 00 02 08 00 ff 64 01 01 82 01 62 1e 52 03 56 00 00 00 00 00 01 -> Einspeisung Tarfilos
> 09:34:09 : 77 07 01 00 01 08 01 ff 01 01 62 1e 52 03 56 00 00 00 08 a5 01 -> Zählerstand Tarif 1 
> 09:34:09 : 77 07 01 00 02 08 01 ff 01 01 62 1e 52 03 56 00 00 00 00 00 01 
> 09:34:09 : 77 07 01 00 01 08 02 ff 01 01 62 1e 52 03 56 00 00 00 00 00 01 
> 09:34:09 : 77 07 01 00 02 08 02 ff 01 01 62 1e 52 03 56 00 00 00 00 00 01 

Woran könnte das liegen?
Danke für deine Hilfe.

Gruß

gemu2015, 2019-03-08:
Hallo fleischtorte,

versuchs mal der neuen Version. sollte jetzt gehen. 

Gruß gemu

fleischtorte, 2019-03-08:
Hallo gemu,

hat funktioniert 👍 vielen Dank für deine schnelle Unterstützung.

Gruß fleischtorte

closed 2019-03-08 by fleischtorte