gemu2015/Sonoff-Tasmota #7: Universal 3 doesnt compile

Jason2866, 2019-05-02:
Reason: user_config_override is enabled and the needed file is missing.

gemu2015, 2019-05-02:
for unknown reason -DUSE_CONFIG_OVERRIDE in platformio.ini didn't work for me
so i enabled the override in sonoff.ino by uncommenting  the #ifdef

#ifdef USE_CONFIG_OVERRIDE
  #include "user_config_override.h"         // Configuration overrides for my_user_config.h
#endif

i have updated this to the normal behavior and will try the normal way going.

Jason2866, 2019-05-02:
Tried with Gitpod (not on my PC) and i get:

```

Compiling .pioenvs/sonoff/lib7df/Ticker/Ticker.cpp.o
Archiving .pioenvs/sonoff/lib4c4/libAdafruit_SH1106-master.a
lib/ILI9488-master/ILI9488.cpp:40:17: fatal error: spi.h: No such file or directory

*************************************************************
* Looking for spi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:spi.h"
* Web  > https://platformio.org/lib/search?query=header:spi.h
*
*************************************************************

#include <spi.h>
^
compilation terminated.
Archiving .pioenvs/sonoff/libb16/libArduinoOTA.a
*** [.pioenvs/sonoff/libee3/ILI9488-master/ILI9488.cpp.o] Error 1
Indexing .pioenvs/sonoff/lib384/libesp-epaper-29-ws-20171230-gemu-1.0.a
Indexing .pioenvs/sonoff/lib4c4/libAdafruit_SH1106-master.a
Indexing .pioenvs/sonoff/libb16/libArduinoOTA.a
================================================== [ERROR] Took 46.24 seconds ==================================================

=========================================================== [SUMMARY] ===========================================================
Environment sonoff              [ERROR]
```

gemu2015, 2019-05-02:
on platformio it compiles with no error. if you delete the ili9488 folder it should work.

as i have only platformio i could not determine why this happens

Jason2866, 2019-05-02:
Thx, latest commits (upper lower case) has solved

closed 2019-05-02 by Jason2866