gemu2015/Sonoff-Tasmota #8: 'ShowSource' was not declared in this scope

AndyWo83, 2019-06-07:
Hello,

i would like use the Universal 4 with my Wemos D1 Mini to read my powermeters.
But in Arduino i have a error message.
How can I fix the error?
[image]

gemu2015, 2019-06-07:
as i use atom to compile i have no idea why this happens.

if you are able to compile the standard tasmota source with arduino IDE you may simply take over sml95 and TasmotaSerial folder (in lib). no other changes necessary .

AndyWo83, 2019-06-07:
Hmm, with the Tasmota 6.5.0.13, copy/paste the xsns_95 and your serial driver, everything is OK...

[image]

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015