gemu2015/Sonoff-Tasmota #9: Connection Display (SS1351 128x128 OLED Display)

chkbln, 2019-07-01:
Hello gemu2015
I would like to use the display (SS1351 128x128 OLED Display).
How is the display connected?
About 4-SPI or I2C?
I2C
SLC = GPIO 5
SDA = GPIO 4
Would that be correct?

How is the connection via 4-SPI?
What must the necessary entries look like in my_user_config.h?

many Greetings

gemu2015, 2019-07-01:
hi chkbln,

the display is connected via 3 wire hardware spi. (you must set up the jumper on the display for 3 wire spi)

in user_config_overwrite specify #define USE_SSD1351

then in Tasmota config GPIO  define 

MOSI to SSPI_MOSI on GPIO13
SCLK to SSPI_SCLK on GPIO14
and CS to SSPI_CS on GPIO15

Gerhard

chkbln, 2019-07-01:
Hello gemu2015
I have to specify the DisplayMode in the Tasmota Console.
Which one would that be?

Unfortunately, they gave me the monochrome display in the store.
That's why I have I2C and 4-SPI 4.
Must now exchange the display once.

many Greetings

gemu2015, 2019-07-02:
ok, you must specify =>

#define USE_SPI
#define USE_DISPLAY
#define USE_DISPLAY_SSD1351
#undef USE_DISPLAY_MODES1TO5 // because i do not support modes 1-5
#define SHOW_SPLASH  // nice to see when display comes up

chkbln, 2019-07-02:
Hi,
unfortunately my display stays dark.
Do you have any idea how to find the mistake?
Unfortunately, SHOW_SPLASH is not displayed.
greeting

gemu2015, 2019-07-03:
Hardware:
ssd1351module => ESP
===========
GND => GND
VCC => 3,3 V  (5 Volt will damage the display)
DIN => MOSI (GPIO13)
CLK => SCLK (GPIO14)
CS => SPI_CS (GPIO15)
DC => GND
RST => 3,3 V

jumper BS on the display soldered to "1"  (is at "0" at delivery)

Tasmota GPIO selector (note must set to SSPI not SPI, because software spi may be a later option)
GPIO13 => SSPI_MOSI
GPIO14 => SSPI_SCLK
GPIO15 => SSPI_CS

in Tasmota console
check displaymodel , should show 9
displayrotate should show 0
then enter displaydimmer 100

you should see "SSD1351" centered on the screen

greeting

chkbln, 2019-07-03:
Hi,
Now I have SSD1351 on the display lights up.
Supi
When compiling with the Arduino IDE I got the following error message.
[image]
 
I then commented out the entry. What effects does this have or how can I get rid of the error?
 
[image]

How do I get values, e.g. Temperature displayed on the display.
Does it work like the e-paper?
You had once published a Word document.

many Greetings

gemu2015, 2019-07-04:
ok, this include was actually not needed, i removed it. (otherwise it hat to be uppercase <SD.h>)

my drivers do only support display mode 0.

you have to display data with the cmd "displaytext"

you may either do this remotedly with home automation or with rules or easier with scripter.
see scripter.md for doc and examples

to enable scripter:
#undef USE_RULES
#define USE_SCRIPT

in the WEBUI config menu you find an entry "edit script"
there you can enter scripts 

here is an example:

; define variables
>D
temp=0
hum=0

; called during boot
>B
; clr screen
=>displaytext [z]

; called during telemetry
>T
; catch sensor values
temp=BME280#Temperature
hum=BME280#Humidity

;print values to console for debugging
=>print %temp% : %hum%

; called every second
>S
; now print to display
; print at position x=0,y=20 with index color 2 (red) and font 1 (epd font 12)
=>displaytext [x0y20Ci2f1] Temperature=%temp%
; print with index color 3 (green) at y pos 60
=>displaytext [x0y60Ci3] Humidity=%hum%

for detailed explanation of displaytext see displaytext.md

remarks on OLED displays:  
oled display modules have an expected lifetime of about 10000 hours (416 days) which means they should not be on permanently. you may use a proximity sensor to turn the display on and off.
dimming the display will also expand the lifetime. (dimming to 50% expands the lifetime to about 25000 hours)  

so best to dim as low as possible with e.g. displaydimmer 50

have fun

chkbln, 2019-07-05:
Hi

I wanted to recompile the sketch today to try scripter.
Unfortunately, I get the following error messages.

xdrv_10_scripter: 2979: 62: error: 'FAT_TIME' was not declared in this scope
'FAT_DATE' was not declared in this scope

Is the error in the xdrv_10_scripter.ino?
many Greetings

chkbln, 2019-07-05:
When I comment out this line, the compilation runs through

  //#define USE_SCRIPT_FATFS 4

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015