gemu2015/Sonoff-Tasmota #14: Calculate the filled volume of a horizontal cylinder tank

littlebilly, 2019-11-06:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**
yes
**Describe the solution you'd like**
Is it possible to "Calculate the filled volume of a horizontal cylinder tank" with the  script editor for Tasmota?

I get the the water-level of my lying cistern via the analog value of a Wemos D1 flashed with Tasmota.
It would be nice to calculate the Volume (Liter) via Tasmota Scripting Language and show the result on a ssd1306 Display.

gemu2015, 2019-11-06:
since the formula is very easy  (V = π · r2 · h)
it is no problem to calculate the volume right away
you may precalculate pi *r2 

>D
v=0
h=0
pir2=7.23456 (e.g.)

>T
; enter your sensors MQTT designators here
h=Yoursensor#level

>S

v=pir2*h

print Volume=%v% liters

littlebilly, 2019-11-06:
Thanks a lot for your quick reply.
As I mentioned I  have to Calculate the filled volume of a **horizontal** cylinder tank like in this link.
[https://www.calculatorsoup.com/calculators/construction/tank.php](url)
Your formula is for "Vertical Cylinder Tank"
The question is whether "Horizontal Cylinder Tank" could be calculate with your scripting language.

littlebilly, 2019-11-06:
Sorry closing was a mistake

gemu2015, 2019-11-06:
Ok, horizontal cylinder is currently not supported because It needs arccos and sin
will think about implementing

gemu2015, 2019-11-06:
#define USE_ANGLE_FUNC
enables sin and acos

given r,l and h
this should calculate the volume

a=2*acos((r-h)/r)
v=l * ((r*r)/2)*(a-sin(a))

did not test it yet you may give it a try

littlebilly, 2019-11-06:
You are the greatest!
Thanks a lot I will test it as soon as possible.
But this will take some time, because I startet using your Scripting Language today.

Question? May I use your fork universal7 equal to version 6.71

gemu2015, 2019-11-07:
yes sure, if this is verified i will make a pull request to Tasmota dev branch.

littlebilly, 2019-11-07:
Do I have to activate 
#define USE_ANGLE_FUNC
via console or in my_user_config.h?

For testing I tried this without any sucess.

`>D
r=100
l=207
h=0
a=0

>T
h=ANALOG#A0/3
>S

a=2*acos((r-h)/r)
v=l * ((rr)/2)(a-sin(a))

=>print Flaeche A:%a% qcm!
=>print Volumen :%v% qcm!
endif`

littlebilly, 2019-11-07:
A little step forward.
a=2*acos((r-h)/r)        has to be 
a=2 * acos((r-h)/r) 
otherwise i get the message --> var not found: *acos((r-h)/r)
-------------
Result 
Hoehe: 176.00 cm
Flaeche A:2.00 cm*cm
Volumen :207.00 cm*cm*cM
I think there is still something wrong.

littlebilly, 2019-11-07:
[image]
Would it be possible to use this formula

gemu2015, 2019-11-07:
hier hab ich die Formel her (Bartsch)
[Beh-ltervolumen.pdf](https://github.com/gemu2015/Sonoff-Tasmota/files/3818749/Beh-ltervolumen.pdf)

#define USE_ANGLE_FUNC muss in my_user  config.
in Formeln darf kein Leerzeichen zwischen Mathezeichen sein weil nach dem Leerzeichen abgebrochen wird.

der Fehler kommt weil acos nicht kompiliert wurde

littlebilly, 2019-11-07:
ok. ich hatte #define USE_ANGLE_FUNC in der user_config_override.h
Werde das ganze nochmals sauber bauen.
Danke nochmals!

gemu2015, 2019-11-07:
ok, hab jetzt selbst noch mal getestet da war noch ein Fehler bei den beiden Funktionen, ist jetzt behoben. Jetzt mal die Formel auf plausible Ergebnisse testen

gemu2015, 2019-11-07:
hab jetzt noch sqrt (wurzel) eingebaut damit man auch die andere Formel testen kann

littlebilly, 2019-11-07:
Super danke werd ich tun! Und melde mich auf jeden Fall.
Falls es dich interessiert hier mein Beitrag zur Zisternenlösung!
[https://forum.fhem.de/index.php/topic,93308.0.html](url)

gemu2015, 2019-11-07:
mit diesen Werten kommt das richtige Ergebnis raus
muss man jetzt mal mir verschiedenen anderen Werten testen 

\>D
r=100
l=200
h=100
a=0
v=0

\>T
;h=ANALOG#A0/3

\>S

a=2*acos((r-h)/r)
v=l*((r*r)/2)*(a-sin(a))

print Winkel Alpha:%a%
print Volumen :%v% qcm!

littlebilly, 2019-11-07:
Sorry habe gerade große Probleme beim flashen!
Ergebnis mit obigem script bei mir!
16:06:06 Winkel Alpha:2.00
16:06:06 Volumen :0.00 qcm!
Volumen null kann eigentlich nicht  sein. Mache jetzt mal eine längere Pause.

gemu2015, 2019-11-07:
sorry, wenn man ein script hier postet gehen Zeichen verloren.
oben sind die * Zeichen verschwunden.
habe das editiert

littlebilly, 2019-11-07:
Super, danke die Volumenberechnung geht jetzt.
Wieso a0 nicht mehr angezeigt wird und wieso Display jetzt verschwnden ist macht mir noch Probleme.

littlebilly, 2019-11-08:
Works perfectely now! 
I copied your "xdrv_10_scripter.ino" to Tasmota 6.71 master, activated #define USE_ANGLE_FUNC
in my_user_config.h!!! After compilation I got what I need. I learned a lot regarding your phantastic scripting language for Tasmota. Many thanks for your help.

By the way, is it possible to suppress the two positions behind the decimal point?

[image]

gemu2015, 2019-11-08:
setting the precision of a replacement variable

either globally for all following vars with dpx  (x 0..9)
or individually by a preceding digit

e.g.
displaytext volume: %0v% liter

littlebilly, 2020-05-20:
Is there a bug in your calculates exponential powers functionality?
pow(x y) = calculates exponential powers x^y
---------------------------------------------------------------
**19:59:46 CMD: script >print%sqrt(4)% --> the result is 4 and is ok!
19:59:46 2.00**

**20:00:14 CMD: script >print%pow(2 2)% --> the result is 3.88 instead 4 ????
20:00:14 3.88**

gemu2015, 2020-05-20:
i simply call pow from the tasmota math utility it was changed in the corse of time to a smaller and faster form. will check if it is that imprecise.

gemu2015, 2020-05-21:
i checked it now. it is indeed that imprecise
Arends replaced all pow() functions in Tasmota with FastPrecisePow which tends to be NOT precise at all.

gemu2015, 2020-05-21:
what is the impact on the above formula ?
if it is too hight you may file a bug report to tasmota.

littlebilly, 2020-05-21:
Hi Gemu,
there is no impact at all to the above formula. I just used pow function in another context and noticed that it does not work correctly and thought it is scripting related.
Thanks a lot for your quick response and again for your Tasmota scripting version.

closed 2020-05-21 by littlebilly