gemu2015/Sonoff-Tasmota #15: Is it possible to issue a Subscribe command from inside a script?

YvesLev, 2019-11-29:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Hello @gemu2015,
I managed to compile latest dev branch of tasmota and enable Scripting (and disable rules) and enable also Subscribe/Unsubscribe.

In the console, issuing a Subscribe command with a Variable name defined in the script works very well.

But I have not manage to issue the very same Subscribe command inside the script with:
=>Subscribe SubVar, Topic/stat/POWER

This in the console gives: {"Command":"Unknown"}

Since I lose the subscribe when power cycle off/on, I would have like to issue this subscribe command from the Script.

So I would like to know if there is another way of writing the command from the script or if this is just not implemented yet?

Thank you.
(And thanks to have brought this scripting option to tasmota!)

Yves

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU board for testing purpose
- [X] Tasmota binary firmware version number used: 7.0.0.4
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: GitPod
- [X] Flashing tools used: NodeMCU PyFlasher
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  Configuration output here:
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  Rules output here:
- [ ] Provide the output of this command: ``Status 0``:
  STATUS 0 output here:
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  Console output here:
**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2019-11-29:
you may NOT execute subscribe in the \>B section because MQTT is not yet running there
you must send it in \>S and assure that it is issued only once

\>S
if mqttc>0
then
=>Subscribe SubVar, Topic/stat/POWER
endif

YvesLev, 2019-11-29:
Thanks,

That is exactly what I did but had no luck yet:
Console with debug info:
13:06:33 QPC: Reset
13:06:33 **Script: performs "Subscribe MqtR, BulbChambreAmi/stat/POWER"**
13:06:33 SRC: Rule
13:06:33 CMD: Group 0, Index 1, Command "SUBSCRIBE", Data "MqtR, BulbChambreAmi/stat/POWER"
13:06:33 MQT: stat/McuTest/RESULT = **{"Command":"Unknown"}**
13:06:35 APP: Boot Count 9
13:06:35 CFG: Saved to flash at FB, Count 73, Bytes 4096
13:06:37 MQT: tele/McuTest/STATE = {"Time":"2019-11-29T13:06:37","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"PuttPutt","BSSId":"1C:87:2C:68:7C:A8","Channel":11,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:06:39 HTP: Main Menu
13:06:45 HTP: Console
13:07:42 CMD: subscribe
13:07:42 SRC: WebConsole from 192.168.2.58
13:07:42 CMD: Group 0, Index 1, Command "SUBSCRIBE", Data ""
13:07:42 MQT: stat/McuTest/RESULT = {"Subscribe":""}

gemu2015, 2019-11-29:
this works for me without error

\>D
MqtR=0

\>S

if mqttc>0
then
=>Subscribe MqtR, BulbChambreAmi/stat/POWER
endif

YvesLev, 2019-11-29:
Can you tell me which version of tasmota you compile with?

YvesLev, 2019-11-29:
In the user_config_override.h, I have the following:

`#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif

#ifndef USE_BUTTON_EVENT
#define USE_BUTTON_EVENT
#endif
#ifndef SUPPORT_MQTT_EVENT
#define SUPPORT_MQTT_EVENT
#endif`

gemu2015, 2019-11-29:
the very last version of my fork but is identical in scripter to the current Tasmota dev

YvesLev, 2019-11-29:
I'll try the very latest one in tasmota, and if not working, I'll try yours.

I'll be back soon :-)
Thanks

YvesLev, 2019-11-29:
With the latest Tasmota Dev Branch, I still have same problem.

Gonna try your fork now.

YvesLev, 2019-11-29:
So I compiled using universal8 branch with GitPod and I have the same problem.

So I guess I must be doing something wrong in the parameters for the compilation...
Or do I have to use some SetOption commands?

All I do to compile, is go to gitub of tasmota, use de defaut branch which is develop, then I click on GitPod on the right which I believe opens the latest version of develop onti the GitPod IDE.

Then I edit Platform.ini to have Tamsota uncommented and all other language commented.
Also I uncomment the line to use User_Config_Override.h.

I attached in a previous post the content of my user_config_override.h file.

Let me know if I forgot something.

Thanks again!

gemu2015, 2019-11-29:
strange, i now get the error too ???
it was working several times ???

try this variant =>

if mqttc>0
then
+>Subscribe MqtR, BulbChambreAmi/stat/POWER
endif

YvesLev, 2019-11-29:
Ok, +> works !!!

Funny, cause I tried yesterday -> variant and had same problem and stopped there with the varaints lol.

Big Thanks!

Should we close this issue?
I think at least the documentation should be revised and also the cook book to show an example of use with the mqttc thing etc...

gemu2015, 2019-11-29:
ok, leave it open for others.
i will examine why that variant works and others not.
+> allows for recursion but i am sure it also worked with => when i developed the call.

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015