gemu2015/Sonoff-Tasmota #16: Keyfob not working

lacrimosa2k2, 2019-12-26:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
i have the HM-17 module and the cheap keyfob that can beep.
the hm-17 work with the tasmota, i can see the scan.
but when i power on one of the keyfob i see in the debug info only >>00000000:0000 and no data on mqtt or so.
on the site it says this keyfob works, i have buy this, ok another color.

so i tried in the driver file change from DISI to DISC, so i can see the mac of the keyfob.

what can i do to get this work?

thanks

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): wemos d1 mini & HM-17
- [ x] Tasmota binary firmware version number used: 8.1
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: Arduino IDE
- [ ] Flashing tools used: _____
- [x ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
18:23:05 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:23:06 MQT: stat/sonoff/RESULT = {"Module":{"0":"Generic"}}
18:23:06 MQT: stat/sonoff/RESULT = {"GPIO0":{"9":"Switch1"},"GPIO1":{"168":"iBeacon TX"},"GPIO2":{"157":"LedLink"},"GPIO3":{"169":"iBeacon RX"},"GPIO4":{"1":"DHT11"},"GPIO5":{"106":"RFrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"42":"Counter1"},"GPIO14":{"0":"None"},"GPIO15":{"9":"Switch1"},"GPIO16":{"0":"None"}}
18:23:06 DIS0 OK

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [x ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
18:22:24 CMD: STATUS 0
18:22:24 MQT: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:22:24 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:29","StartupUTC":"2019-12-26T17:10:55","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":103,"SaveAddress":"3F4000"}}
18:22:24 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(tasmota)","BuildDateTime":"2019.12.26 12:27:26","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(5ab15d1)","Hardware":"ESP8266EX","CR":"333/699"}}
18:22:24 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["0011001000110011",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
18:22:24 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":1524,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000407","8F802197","000100A0","020004C5","000000C0","00000904"],"Drivers":"1,2,3,4,8,9,10,17,27,29","Sensors":"1,2,3,5,6,9,22,52"}}
18:22:24 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2261","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"BC:DD:C2:B6:68:D5","Webserver":2,"WifiConfig":4}}
18:22:24 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"iobroker.local","MqttPort":1883,"MqttClientMask":"Sicherungskasten","MqttClient":"Sicherungskasten","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:22:24 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 26 17:22:24 2019","Local":"Thu Dec 26 18:22:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:16","Sunset":"16:11"}}
18:22:24 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-12-26T18:22:24","Switch1":"OFF","COUNTER":{"C1":86},"DHT11":{"Temperature":23.7,"Humidity":38.0},"TempUnit":"C"}}
18:22:24 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-12-26T18:22:24","Uptime":"0T00:11:29","UptimeSec":689,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"0011001000110011","BSSId":"00:17:13:15:65:98","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}
18:22:25 hm17cmd 6

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2019-12-26:
did you initialize the hm17 ?

you must execute these commands only once (reconfigure the hm17 to master mode)
sensor52 1
sensor52 2

then switch debug mode on

#define IBEACON_DEBUG in compiler

and then

sensor 52 d1

lacrimosa2k2, 2019-12-26:
yes, i have do this.
in the console with debug i can see the scan result.
when i turn on the smart-tag it comes >>00000000:0000

gemu2015, 2019-12-27:
you should NOT use DISC but DISI

below is my debug log for 2 ibeacons and one keyfob 
there are many other bluetooth devices listet that are not compatible
compatible devices have the first field set OR MAC starts with "FFFF" (keyfob)

09:43:34 hm17cmd 3
09:43:35 DISCS OK
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:989E633D04A9:-062
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:5AEEDA4CA928:-062
09:43:35 DISC: OK
09:43:35 >>4C000C0E:00B381FFCE3E6DFD455B9D5329BD1005:4B1C4EF4C1:59F59CC93F4A:-078
09:43:35 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T09:43:35","IBEACON_59F59CC93F4A":{"RSSI":-78}}
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:604057C2E87F:-068
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:779D778DE593:-063
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:FFFF000F893F:-064
09:43:35 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T09:43:35","IBEACON_FFFF000F893F":{"RSSI":-64}}
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:48921D1EB1D8:-064
09:43:35 DISC: OK
09:43:35 >>00000000:00000000000000000000000000000000:0000000000:0024E4204295:-078
09:43:39 DISC: OK
09:43:39 >>4C000215:FDA50693A4E24FB1AFCFC6EB07647825:0005000622:E788BEC6FBF0:-075
09:43:39 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T09:43:39","IBEACON_E788BEC6FBF0":{"RSSI":-75}}
09:43:39 DISC: OK
09:43:39 >>00000000:00000000000000000000000000000000:0000000000:989E633D04A9:-066
09:43:40 DISC: OK
09:43:40 >>00000000:00000000000000000000000000000000:0000000000:5AEEDA4CA928:-065
09:43:40 DISCE OK

web menu shows the 3 devices

[image]

lacrimosa2k2, 2019-12-27:
This is my console output with debug enabled:
11:27:42 CMD: sensor52 d1
11:27:42 MQT: stat/sonoff/RESULT = {"Sensor52":debug:1}
11:27:47 hm17cmd 3
11:27:47 DISCS OK
11:27:50 DIS0 OK
11:27:50 >>00000000:000
11:27:57 hm17cmd 3
11:27:57 DISCS OK
11:27:59 DIS0 OK
11:27:59 >>00000000:000
11:28:07 hm17cmd 3
11:28:07 DISCS OK
11:28:08 DIS0 OK
11:28:08 >>00000000:000
11:28:17 hm17cmd 3
11:28:17 DISCS OK
11:28:19 DIS0 OK
11:28:19 >>00000000:000
11:28:27 hm17cmd 3
11:28:27 DISCS OK
11:28:29 MQT: tele/sonoff/STATE = {"Time":"2019-12-27T11:28:29","Uptime":"0T00:01:12","UptimeSec":72,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"0011001000110011","BSSId":"00:17:13:15:65:98","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}
11:28:29 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T11:28:29","Switch1":"OFF","COUNTER":{"C1":86},"DHT11":{"Temperature":19.8,"Humidity":40.0},"TempUnit":"C"}

but i found, at the moment when i send the reset command, it shows the right data:
11:43:10 hm17cmd 3
11:43:10 DISCS OK
11:43:12 DIS0 OK
11:43:12 >>00000000:000
11:43:14 CMD: sensor52 s AT+RENEW
11:43:14 MQT: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+RENEW}
11:43:14 296:-052OK+RENEW0000000000000:0000000000:FFFF365C3396:-057OK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF1A843

gemu2015, 2019-12-27:
may be there is a different firmware in the module
please check firmware version with

sensor52 s AT+VERR?

my module was sold with remark "newest firmware version"
however it reports V120 which is in conflict with version numbers of the documents

after resetting the module to factory defaults it should only work in role 0 (peripheral role)
but it MUST be in role 1 (central role)
also it must be in IMME 1 mode

lacrimosa2k2, 2019-12-27:
12:24:19 CMD: sensor52 s AT+VERR?
12:24:19 MQT: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+VERR?}
12:24:19 296:-057OK+DISCEOK+Get:HMSoft V110000000:FFFF365C3396:-052OK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF1A843

since it the v 110

gemu2015, 2019-12-27:
pay attention with raw commands!

when you place eg AT+DISCE cmd which takes some time it may be that the regular scab interferes (which is the case in you logging)

before testing with raw command you should switch of the scheduler

by sensor52 u0

lacrimosa2k2, 2019-12-27:
yes it is the v110

12:26:22 MQT: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+VERR?}
12:26:22 OK+Get:HMSoft V110

gemu2015, 2019-12-27:
ok your software is older, there my be the reason why you get different results.
We have to find a datasheet which refers to the version differences.

lacrimosa2k2, 2019-12-27:
on the manufacture site v110 is the actual firmware
http://www.jnhuamao.cn/download_rom_en.asp?id=

gemu2015, 2019-12-27:
very confusing , i definitely have the hm17 (at least printed on board) with v120
hm19 has more pins

in the data sheet they refer to V539 as latest revision  ????

no the hm10 can't take central role only peripheral role that's the main difference.

hm19 board looks exactly the same.

[image]
[image]

that's what i have
[image]

gemu2015, 2019-12-27:
dann schreiben wir doch besser in Deutsch.

Dein logging ist komplett anders, da muss es Formattierungs  Unterschiede in der Modulausgabe geben.

du solltest die rohen Ausgaben loggen dann kann ich vergleichen wo die Unterschiede sind und das anpassen.

Viel kann es nicht sein.

gemu2015, 2019-12-27:
merkwürdig, in deinem Link mit den ganzen Modulen gibt es das hm17 und auch hm19 sowohl mit v110 und v120

lacrimosa2k2, 2019-12-27:
ok, your module have the cc2541
my hm-17 have the CC2640R2f
my hm-19 module have a TI chip

my hm-10 (als solches verkauft) - deutsch its besser ja :D
hat einen cc2541 chip, es ist ein fake, aber nach einem firmwareflash auf die offizielle firmware kann das auch zentralle role.
hier eine ausgabe des "hm-10" über usb-ttl
OKOK+Set:1OK+Set:1OK+DISCSOK+DIS0:FFFF365C3396OK+RSSI:-048
OK+NAME:iTAG            
OK+DIS1:FEF4FD96F07COK+RSSI:-081
OK+NAME:SumUp 606
OK+DISCEOK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:FEF4FD96F07C:-095OK+DISC:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-047OK+DISC:00000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-049OK+DISCE

der hm-17 über den komme ich leider im moment nur über die tasmota konsole da er verlötet ist gibt bei DISC? das aus
2:55:55 CMD: sensor52 s AT+DISC?
12:55:55 RSL: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+DISC?}
12:55:55 OK+DISCS
12:55:56 OK+DIS0:FFFF365C3396:-044:i
12:55:56 TAG            

12:55:56 OK+DIS0:FFFF1A843
12:55:56 296:-032:iTAG            

12:55:58 OK+DISCE

und DISI?
2:55:55 CMD: sensor52 s AT+DISC?
12:55:55 RSL: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+DISC?}
12:55:55 OK+DISCS
12:55:56 OK+DIS0:FFFF365C3396:-044:i
12:55:56 TAG            

12:55:56 OK+DIS0:FFFF1A843
12:55:56 296:-032:iTAG            

12:55:58 OK+DISCE

das sieht für mich erstmal gleich aus

gemu2015, 2019-12-27:
du hast dich vertippt hast 2 mal disc? abgeschickt

lacrimosa2k2, 2019-12-27:
oh
hier nochmal richtig
CMD: sensor52 s AT+DISI?
13:02:23 RSL: stat/sonoff/RESULT = {"Sensor52":hm17cmd:AT+DISI?}
13:02:23 OK+DISCS
13:02:24 OK+DIS0:00
13:02:24 000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-032
13:02:26 OK+DIS0:00000000:000
13:02:27 00000000000000000000000000000:0000000000:FFFF365C3396:-043
13:02:27 OK+DISCE

gemu2015, 2019-12-27:
ja den hm10 habe ich auch aber meiner geht nur in peripheral role.

versuche mal die Unterschiede zwischen unseren Version zu checken
melde mich später wieder

lacrimosa2k2, 2019-12-27:
Ich fass das nochmal Formatiert zusammen.
beide module über einen USB-TTL adapter abgefragt
hm-10 (mit firmwareflash) chip: cc2541
AT+VERR?
HMSoft V609

AT+ROLE1
OK+Set:1

AT+IMME1
OK+Set:1

AT+DISC?
OK+DISCSOK+DIS1:FEF4FD96F07COK+RSSI:-086
OK+NAME:SumUp 606
OK+DIS0:FFFF365C3396OK+RSSI:-062
OK+NAME:iTAG            
OK+DIS0:FFFF1A843296OK+RSSI:-044
OK+NAME:iTAG            
OK+DISCE

AT+DISI?
OK+DISISOK+DISC:00000000:00000000000000000000000000000000:0000000000:FEF4FD96F07C:-086OK+DISC:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-054OK+DISCE

HM-17 chip:CC2640R2f

AT+VERR?
OK+Get:HMSoft V110

AT+ROLE1
OK+Set:1

AT+IMME1
OK+Set:1

AT+DISC?
OK+DISCSOK+DIS0:FFFF365C3396:-046:iTAG            
OK+DIS0:FFFF1A843296:-046:iTAG            
OK+DISCE

AT+DISI?
OK+DISCSOK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-046OK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-052OK+DISCE

identisch sind die antworten nicht.
aber zb bei DISI? die antwort, wenn man alles vor dem ersten : rausfiltert sind sie gleich.
warum der cc2541 chip das sumup gerät sieht und der andere nicht ist mir nicht ganz klar.

egal welches von dem modulen, in der tasmota console kommt immer dieses bild (ist jetzt vom HM-17)
13:36:50 CMD: sensor52 d1
13:36:50 MQT: stat/sonoff/RESULT = {"Sensor52":debug:1}
13:36:51 DIS0 OK
13:36:51 >>00000000:000
13:36:59 hm17cmd 3
13:36:59 DISCS OK
13:37:00 DIS0 OK
13:37:00 >>00000000:000
13:37:09 hm17cmd 3
13:37:09 DISCS OK
13:37:11 DIS0 OK
13:37:11 >>00000000:000

gemu2015, 2019-12-27:
hab eine neue Version auf Github gestellt. Versuchs mal damit

lacrimosa2k2, 2019-12-27:
betrifft das nur die ibeacon.ino datei?
die habe ich gerade übernommen, leider immer noch das selbe bild

gemu2015, 2019-12-27:
ja nur die iBeacon.ino.
soweit ich das sehe gibt es doch nur einen Unterschied:

DISO versus DISC

dein HM17
AT+DISI?
OK+DISCS
OK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-046
OK+DIS0:00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-052
OK+DISCE

mein hm17
OK+DISCS
OK+DISC:00000000:0000000000:0000000000:52C04423F76D:-078
OK+DISC:00000000:00000000000000000000000000000000:0000000000:5C9940634CCB:-073
OK+DISC:400000000000000000000000000000000:0000000000:78A9EF4ADEBB:-066
OK+DISC:00000000:00000000000000000000000000000000:0000000000:0024E4204295:-078
OK+DISC:00000000:00000000000000000000000000000000:0000000000:52C04423F76D:-076
OK+DISC:00000000:00000000000000000000000000000000:0000000000:5C9940634CCB:-064
OK+DISCE

lacrimosa2k2, 2019-12-27:
scheinbar ist es nur der unterschied, wenn ich das at comand manuel sende listet er mir auch die sachen. nur scheint die auswertung bzw das dekodieren der antwort irgendwie nicht richtig.
ich bin kein c# profi, aber für mich liest sich zeile 192 if (!strncmp(ib->MAC,"FFFF",4) || strncmp(ib->FACID,"00000000",8)) { seltsam, wenn nicht fff oder 0000... dann, das ist richtig so?

gemu2015, 2019-12-27:
Ja dieser Teil ist richtig.
Was sagt denn dein Log jetzt. Vergleiche mal mit meinem von Oben.
erst wenn der Log gleich aussieht geht es weiter

gemu2015, 2019-12-27:
oh war noch ein Fehler drin, Versuchs jetzt nochmal

lacrimosa2k2, 2019-12-27:
jetzt hats das teil beim flashen wieder abgeshcossen, muss es erst holen.
du meinst sicher DIS0 vs DISC? meiner schreibt eine 0 dahin wo deiner ein c hat

gemu2015, 2019-12-27:
ja, ist nur dieser Unterschied, hatte es aber falsch korrigiert vorhin

lacrimosa2k2, 2019-12-27:
juhu, da tut sich was

17:05:56 CMD: sensor52 d1
17:05:56 MQT: stat/sonoff/RESULT = {"Sensor52":debug:1}
17:05:59 hm17cmd 0
17:05:59 AT OK
17:06:09 hm17cmd 3
17:06:09 DISCS OK
17:06:10 DISC: OK
17:06:10 >>00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-052
17:06:10 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T17:06:10","IBEACON_FFFF365C3396":{"RSSI":-52}}
17:06:12 DISC: OK
17:06:12 >>00000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-063
17:06:12 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T17:06:12","IBEACON_FFFF1A843296":{"RSSI":-63}}
17:06:13 DISCE OK
17:06:19 hm17cmd 3
17:06:19 DISCS OK
17:06:21 DISC: OK
17:06:21 >>00000000:00000000000000000000000000000000:0000000000:FFFF1A843296:-061
17:06:21 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T17:06:21","IBEACON_FFFF1A843296":{"RSSI":-61}}
17:06:22 DISC: OK
17:06:22 >>00000000:00000000000000000000000000000000:0000000000:FFFF365C3396:-053
17:06:22 MQT: tele/sonoff/SENSOR = {"Time":"2019-12-27T17:06:22","IBEACON_FFFF365C3396":{"RSSI":-53}}
17:06:23 DISCE OK
17:06:29 hm17cmd 3
17:06:29 DISCS OK

gemu2015, 2019-12-27:
das mit dem Beepen geht so (in der console)
z.B.
IBEACON_FFFF000F893F 99

du musst ja die Adresse angeben

gemu2015, 2019-12-27:
leider hält die Batterie bei den Dingern nur ca einen Monat.
Diese "echten" ibeacons (ohne beeper) halten ein Jahr

https://www.ebay.de/itm/NRF51822-2V-3-3V-Bluetooth-4-0-Funkmodul-fur-ibeacon-Basisstation-YB-W-HV/283697479562?_trkparms=ispr%3D1&hash=item420dafff8a:g:TT4AAOSwaMtcnXbr&enc=AQAEAAACQBPxNw%2BVj6nta7CKEs3N0qXzzM%2Bppa7LaGtlUJmFkFx2Dyixb39xNQkLltRP1Yl44xFgHWActzdl9BdxIQ%2FJSFeMJkGRE2zyksquXDJYYlrN14JmTCCFG2lQNPMmiISPQiIDNs5%2FgE7t%2BDqQw%2FCmtp5co7yggozRg61GEs%2FjGhrvzcaVjcJYZJY73T9F6F7Bdz1HoJOc218N93yfPapR9mP%2Bfesb0%2B5JKcZ00uw620kkeokbYjTEi9E%2B5mpSbVqaS4joolfn7q%2FnbYqAtzpzHmOMtaqtuRFS2pzoKV6nXHhXbn%2FK5BHft8PQXKsZ2P1SxTmYByg4shBswpXnwddC2UvV4bPwCBgx1s6j%2FnAS5EDWQbKW3qSHYWJ67bW%2F%2Btu7Ld10ae0l94S424VIQ3cqazf8JEPY49uubxNhbaSwhd5mN9havCBtWIMmji2U88gsenDQX0yr7CS11F46fEBlOk8zrpJORriFxL%2FGoF6Uu7Y%2FbHoSiez%2BONFtuZ1lSlMLoUA2hN9jNDy9ggpYYsegxtOR0pI9qINUPAT9v7WZGrQLXlrk%2B1b5WJyOsYa%2FQPv1gC%2FiOU9pF8Vax03dbXzfXMTjlztMQxP%2BIk7D8oU4hmtqjS2zD2m%2F2z99EmyFXirYmu6rmxoBvbYZA9cAQ0j3faXzzcyuFD5G86uhIjPejj%2FA%2Fre20GU5aX8lI2fwNMQBFDw1T82nxxr8NkUJ4ADhQG5E64NVrzWWJ4Wqv75rwtXeiJ8evJWD5zpS8z1WYtaGqw%3D%3D&checksum=2836974795628aa06bdee0fd4a0987b8d70a04838aff&enc=AQAEAAACQBPxNw%2BVj6nta7CKEs3N0qXzzM%2Bppa7LaGtlUJmFkFx2Dyixb39xNQkLltRP1Yl44xFgHWActzdl9BdxIQ%2FJSFeMJkGRE2zyksquXDJYYlrN14JmTCCFG2lQNPMmiISPQiIDNs5%2FgE7t%2BDqQw%2FCmtp5co7yggozRg61GEs%2FjGhrvzcaVjcJYZJY73T9F6F7Bdz1HoJOc218N93yfPapR9mP%2Bfesb0%2B5JKcZ00uw620kkeokbYjTEi9E%2B5mpSbVqaS4joolfn7q%2FnbYqAtzpzHmOMtaqtuRFS2pzoKV6nXHhXbn%2FK5BHft8PQXKsZ2P1SxTmYByg4shBswpXnwddC2UvV4bPwCBgx1s6j%2FnAS5EDWQbKW3qSHYWJ67bW%2F%2Btu7Ld10ae0l94S424VIQ3cqazf8JEPY49uubxNhbaSwhd5mN9havCBtWIMmji2U88gsenDQX0yr7CS11F46fEBlOk8zrpJORriFxL%2FGoF6Uu7Y%2FbHoSiez%2BONFtuZ1lSlMLoUA2hN9jNDy9ggpYYsegxtOR0pI9qINUPAT9v7WZGrQLXlrk%2B1b5WJyOsYa%2FQPv1gC%2FiOU9pF8Vax03dbXzfXMTjlztMQxP%2BIk7D8oU4hmtqjS2zD2m%2F2z99EmyFXirYmu6rmxoBvbYZA9cAQ0j3faXzzcyuFD5G86uhIjPejj%2FA%2Fre20GU5aX8lI2fwNMQBFDw1T82nxxr8NkUJ4ADhQG5E64NVrzWWJ4Wqv75rwtXeiJ8evJWD5zpS8z1WYtaGqw%3D%3D&checksum=2836974795628aa06bdee0fd4a0987b8d70a04838aff

lacrimosa2k2, 2019-12-27:
ja das mit dem peeb geht, der broker hat wohl die id noch nicht richtig übernommen oder ich hab ausgerechnet die 99 gesendet als er aktuialisierte.

die beeper werden nur genutzt wenn man sie braucht, also einschalten, in die nähe gehen bis er vom broker eine meldung bekommt zum piepsen usw... eine art anwesenheit/sicherheitsystem

jap von den dingern hab ich mir noch welche bestellt.
auf der tasmota wiki seite sind die ja auch verlinkt bei aliexpress, dort bieten sie aber zwei versionen an, das evt genauer beschreiben welche version man nehmen sollte.

gemu2015, 2019-12-27:
warum die auch hier wieder was anders machen ist mir ein Rätsel.
nach DISI kommt da statt DISCS DISIS sonst ist alles gleich
bau dir das direkt ein.

werde auch mal versuchen meinen zu Flaschen.

gemu2015, 2019-12-27:
ok ist auf GitHub

lacrimosa2k2, 2019-12-27:
super, werd es morgen mal testen mit einem anderen aufbau.

welcher von den nrf chips ist den eig der richtige bei aliexpress, die chinesisch-deutsch übersetzung verstehe ich mal garnicht
https://de.aliexpress.com/item/32885909449.html?spm=a2g0o.productlist.0.0.76004832Hfv6v5&algo_pvid=4b3b6ba8-5607-405d-a287-e974ff95412c&algo_expid=4b3b6ba8-5607-405d-a287-e974ff95412c-0&btsid=b55e7608-ea87-448e-a059-f835093ba95b&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_53
einmal beacon und einmal locate rssi

lacrimosa2k2, 2019-12-28:
HM-10 mit Firmware V609 funktioniert auch

gemu2015, 2019-12-29:
danke für das Doku update .
Die neue xsns_52_ibeacon.ino Version ist gerade bei Tasmota im pr.

lacrimosa2k2, 2019-12-29:
Gern doch.
kleine frage am rande, kann es sein das diese anhänger die piepsen können eine mieserable reichweite haben? mit frischen batterien ist nach 3m freie distanz schluss

gemu2015, 2019-12-29:
ja die Reichweite is nicht sehr hoch.  die Orientierung der Antenne des hm17 spielt aber eine Rolle 
Habe festgestellt dass mein iPhone (app nRF Connect) mit den Buttons  eine deutlich höhere Reichweite hat.

lacrimosa2k2, 2019-12-30:
ich hab jetzt zu dem HM-10 gewechselt da er eine merklich bessere reichweite hat, er hat auch eine invertet-F antenne, der HM-17 hat nur eine "einfache"

Jason2866, 2020-01-29:
Hallo, habe nun einen Clon Hm-10 auf die Version 609 geflasht.
Ich bräuchte (bin kein Programmierer) eine kleine Anleitung wie ich den HM-10 in den Such Modus
bringe das er Beacons anzeigt.
Das initialisieren scheint geklappt zu haben:
```
17:33:08 CMD: sensor52 1
17:33:08 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":hm17cmd:1}
17:33:16 CMD: Sensor52 2
17:33:16 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":hm17cmd:2}
```
Danke!

funnyfrish, 2020-01-29:
Wie hast du die hm10 geflasht? Woher die firmware? Bei mir will es
irgendwie nicht funktionieren.

lacrimosa2k2, 2020-01-29:
> Hallo, habe nun einen Clon Hm-10 auf die Version 609 geflasht.
> Ich bräuchte (bin kein Programmierer) eine kleine Anleitung wie ich den HM-10 in den Such Modus
> bringe das er Beacons anzeigt.
> Das [...]

was sagt den die debug ausgabe? (muss im script aktiviert werden vor dem kompilieren)

Jason2866, 2020-01-29:
@funnyfrish https://github.com/Jason2866/CCLoader

gemu2015, 2020-01-30:
@Jason2866 
#define IBEACON_DEBUG 
und dann sensor52 d1
schaltet den debug mode ein.
In der Konsole sieht man dann welche Befehle gesendet werden und welche Geräte gefunden werden

Jason2866, 2020-01-30:
Danke :-)  Der HM-10 (clone) funktioniert auch mit der neuesten firmware v.707
```
16:17:26 CMD: sensor52 1
16:17:26 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":hm17cmd:1}
16:17:35 CMD: Sensor52 2
16:17:35 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":hm17cmd:2}
16:17:42 CMD: sensor52 d1
16:17:42 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":debug:1}
16:17:43 hm17cmd 3
16:17:44 DISIS OK
16:17:45 DISC: OK
16:17:45 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-086
16:17:45 DISC: OK
16:17:45 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-048
16:17:46 DISC: OK
16:17:46 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-044
16:17:47 DISCE OK
16:17:54 hm17cmd 3
16:17:54 DISIS OK
16:17:55 DISC: OK
16:17:55 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-050
16:17:56 DISC: OK
16:17:56 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-086
16:17:57 DISCE OK
16:18:04 hm17cmd 3
16:18:04 DISIS OK
16:18:04 DISC: OK
16:18:04 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-086
16:18:05 DISC: OK
16:18:05 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-049
16:18:06 DISC: OK
16:18:06 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-043
16:18:07 DISCE OK
16:18:13 hm17cmd 3
16:18:14 DISIS OK
16:18:14 DISC: OK
16:18:14 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-049
16:18:16 DISC: OK
16:18:16 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-044
16:18:17 DISCE OK
16:18:24 hm17cmd 3
16:18:24 DISIS OK
16:18:24 DISC: OK
16:18:24 >>00000000:00000000000000000000000000000000:0000000000:FFFFC0209126:-035
16:18:24 MQT: tele/sonoff-6111ED/SENSOR = {"Time":"2020-01-30T16:18:24","IBEACON_FFFFC0209126":{"RSSI":-35}}
16:18:25 DISC: OK
16:18:25 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-043
16:18:25 DISC: OK
16:18:25 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-050
16:18:26 DISC: OK
16:18:26 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-086
16:18:27 DISCE OK
16:18:34 hm17cmd 3
16:18:34 DISIS OK
16:18:35 DISC: OK
16:18:35 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-049
16:18:35 DISC: OK
16:18:35 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-043
16:18:35 DISC: OK
16:18:35 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-087
16:18:37 DISCE OK
16:18:43 hm17cmd 3
16:18:44 DISIS OK
16:18:44 DISC: OK
16:18:44 >>00000000:00000000000000000000000000000000:0000000000:C54A74D1D3E5:-049
16:18:45 DISC: OK
16:18:45 >>00000000:00000000000000000000000000000000:0000000000:DB2592AEA2A3:-086
16:18:45 DISC: OK
16:18:45 >>00000000:00000000000000000000000000000000:0000000000:A4C1384284FC:-044
16:18:47 DISC: OK
16:18:47 >>00000000:00000000000000000000000000000000:0000000000:FE01BA254A92:-088
16:18:47 DISCE OK
16:18:50 CMD: sensor52 d0
16:18:50 MQT: stat/sonoff-6111ED/RESULT = {"Sensor52":debug:0}
16:18:54 MQT: tele/sonoff-6111ED/SENSOR = {"Time":"2020-01-30T16:18:54","IBEACON_FFFFC0209126":{"RSSI":-36}}
16:19:06 MQT: tele/sonoff-6111ED/SENSOR = {"Time":"2020-01-30T16:19:06","IBEACON_FFFFC0209126":{"RSSI":-47}}
16:19:15 MQT: tele/sonoff-6111ED/SENSOR = {"Time":"2020-01-30T16:19:15","IBEACON_FFFFC0209126":{"RSSI":-50}}

```

funnyfrish, 2020-01-30:
bei dem befehl  Sensor52 1 und sensor52 2  stürzt mein wemos und  auch dernodemcu ab. Ich bekomme dann folgende Fehlermeldung:
:56:28 MQT: tele/tasmota/INFO1 = {"Module":"Sonoff Basic","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_532472_fb/","GroupTopic":"cmnd/tasmotas/"}
20:56:28 MQT: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-1138","IPAddress":"192.168.1.187"}
20:56:28 MQT: tele/tasmota/INFO3 = {"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4020f4f5","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4021f25b","4021f434","4022cc84","4020f7ea","4027ca78","4027c70c","4027c820","4020f810","4027c820","402029f4","4021fe80","40247290","4023edb4","40216c41","401016d3","40101448","401016d3","4023ec10","40232f04","4021ffa4","4021fee4","402283a1","4022ddd9","40240884","4023ff21","4020fa40","4022fabc","40240884","40201378","40243da6","40201378"]}}
20:56:28 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}

Hat jemand eine idee woran das liegen könnten? der HM10 ist über rx/tx verbunden und die nodemcu wird über die pinleiste eingespeist

lacrimosa2k2, 2020-01-30:
> bei dem befehl Sensor52 1 und sensor52 2 stürzt mein wemos und auch dernodemcu ab. Ich bekomme dann folgende Fehlermeldung:
> :56:28 MQT: tele/tasmota/INFO1 = {"Module":"Sonoff [...]

kannst du den hm über serial sauber ansprechen (usb-ttl am pc) und die version abfragen?

funnyfrish, 2020-01-30:
ja das funktioniert. ist eine original HM10. Habs geflasht. Version ist HMSoft V707

Aber da stürzt Tasmota ab und macht dann ein neustart. Wieso schmiert Tasmota ab wenn ich sensor52 befehle auf der weboberflöche in die console eingebe?. Habs mit wemos und nodemcu getestet.

lacrimosa2k2, 2020-01-30:
> 
> 
> ja das funktioniert. ist eine original HM10. Habs geflasht. Version ist HMSoft V707
> 
> Aber da stürzt Tasmota ab und macht dann ein neustart. Wieso schmiert Tasmota ab wenn ich sensor52 befehle auf der weboberflöche in die console eingebe?. Habs mit wemos und nodemcu getestet.

was sagt der hm den wenn du ihn per serial manuel in den master mode schubst mit AT+ROLE1 und AT+IMME1 da müsste er ok sagen

funnyfrish, 2020-01-30:
bei AT+ROLE1
->OK+Set:1
AT+IMME1
->OK+Set:1

Wenn Ich Sensor 52 2 ausführe dann schmiert mein ESP ab und es erscheint folgende Meldung:
```
 {"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4020f4f5","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4021f25b","4021f434","4022cc84","4020f7ea","4027ca78","4027c70c","4027c820","4020f810","4027c820","402029f4","4021fe80","40247290","4023edb4","40216c41","401016d3","40101448","401016d3","4023ec10","40232f04","4021ffa4","4021fee4","402283a1","4022ddd9","40240884","4023ff21","4020fa40","4022fabc","40240884","40201378","40243da6","40201378"]}}
20:56:28 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
```

lacrimosa2k2, 2020-01-30:
den exp code kann ich dir nicht entschlüsseln.
rx und tx in der config eingetragen? auch mal vertauscht probiert?
debug modus eingeschaltet, wenn er startet müsstest du dann die befehle sehen die gesendet/empfangen werden
wen du role und imme einmal gesendet hast musst du es im tasmoate oder sonstwo nicht nochmal machen mit sensor521 und 2, der hm merkt sich diese einstellung

funnyfrish, 2020-01-30:
Wo sind die Stellen für RX+TX?

Serial_log_level habe ich soeben auf none gesetzt
#define SERIAL_LOG_LEVEL       LOG_LEVEL_NONE 
mein HM10 läuft mit Baudrate 115200 statt die 9600. 
Das hab ich in der xsns_52_ibeacon.ino noch angepasst

verbunden ist der HM-10 direkt mit RX und TX auf dem nodemcu. Die Stromversorgung geht direkt über die pinleiste 3.3V und GND

lacrimosa2k2, 2020-01-30:
stelle dein hm auf 9600 und korrigiere die ibeacon wieder.
rx und tx auf der webseite von tasmota in der pin configuration
https://user-images.githubusercontent.com/17022949/50584268-93da0b80-0e94-11e9-84cb-6671288ec198.png
dort unter serial in und out musst du noch das entsprechende wählen, und serial log kannst du in der tasmota config auch über web abschalten

Jason2866, 2020-01-30:
Gpio Zuordnung sollte so aussehen
https://user-images.githubusercontent.com/17022949/50584268-93da0b80-0e94-11e9-84cb-6671288ec198.png

funnyfrish, 2020-01-30:
ich bin dir so dankbar. Nirgends ist das beschrieben....
Das war das Problem....
Habs zurück auf 9600 gesetzt mit AT+BAUD0 und im code wieder auf 9600 zurück. Ab es lag an der Konfiguration der seriellen Schnittstellen.....

Wie ist es kann man damit auch Handys (Android, IPhone) tracken? schicken die Handys auch solche ibeacons raus?

lacrimosa2k2, 2020-01-30:
iPhone senden oder haben Mal dieses beacon gesendet allerdings mit einer random Mac.
Für Android gibt es beacon Emulatoren Apps, die haben bei mir bisher aber nicht funktioniert (nur mit einem hm-17 Mal getestet, der sieht aber nicht alles) soll aber wohl gehen.

funnyfrish, 2020-01-30:
was für sender kann man hierführ nehmen? Betreibe SmartHome und mir gehts um die Anwesenheitserkennung.

lacrimosa2k2, 2020-01-30:
Die in den Wikis beschriebenen beacons gehen ganz gut, die Schlüsselanhänger. Nimm die ohne Piepser da halten die Batterien wesentlich länger, gibt's für paar Euro das Stück beim Chinesen, müsste auch irgendwo verlinkt sein

funnyfrish, 2020-01-30:
ok bestell ich mir gleich. Die Teile schicken dann immer die gleiche ID? Also eine Zuordnung wäre möglich oder?
Wie ist die Reichweite?

Jason2866, 2020-01-30:
Bei mir funktioniert (Android) der Ibeacon Simulator (Google Appstore)

lacrimosa2k2, 2020-01-30:
> ok bestell ich mir gleich. Die Teile schicken dann immer die gleiche ID? Also eine Zuordnung wäre möglich oder?
> Wie ist die Reichweite?

Die Reichweite ist solange, allerdings kann man sich damit eine Raumerkennung basteln wenn man ein paar hm an jeweils ein Gerät bastelt was in einem Raum ist.
Ja die senden immer die selbe id

[12 more replies]

closed 2026-10-28 by gemu2015