gemu2015/Sonoff-Tasmota #17: S0 Zähler auslesen, Landis & Gyr

Kludi2019, 2019-12-27:
Hallo gemu2015,
sorry, wenn ich irgendetwas falsch mache, aber für mich ist es das erste Mal, einen Issues zu veröffentlichen.
Ich möchte einen Landis & Gyr Zähler mit OBIS Datenübertragung auslesen. Ich habe das entsprechend des Forums https://forum.creationx.de/forum/ mehrmals durchgearbeitet. Es handelt sich um einen Zähler, der mit 300 Baud zum senden "angesprochen" werden muss. Das läuft auch inzwischen, mit folgendem Script:
>B

=>sensor53 r

>M 1
+1,5,o,0,300,OBIS,16,20,2F3F210D0A,063030300D0A
1,0.0.1(@1,Zählernummer,,Meter_number,0
1,0.9.1(@1,Zeitstempel,Uhr,time_stamp,0
1,=h===================
1,1.8.1(@1,HT_Verbrauch,KWh,Total_in,3
1,1.8.2(@1,NT_Verbrauch,KWh,Total_in,3
1,1.8.0(@1,Gesamtverbrauch,KWh,Total_in,3
1,=h===================
1,36.7.0(@1,Leistung_L1,kW,Kilowatt_L1,2
1,56.7.0(@1,Leistung_L2,kW,Kilowatt_L2,2
1,76.7.0(@1,Leistung_L3,kW,Kilowatt_L3,2
1,16.7.0(@1,Leistung_L1+L2+L3,kW,Kilowatt_L1+L2+L3,2
1,=h===================
1,31.7.0(@1,Strom_L1,A,Ampere_L1,2
1,51.7.0(@1,Strom_L2,A,Ampere_L2,2
1,71.7.0(@1,Strom_L3,A,Ampere_L3,2
#
Das 1. Problem, welches ich habe, ist nach der Rückmeldung des Zählers, möchte ich auf 9600 Baud die Datenübertragung erhöhen, um schneller die Daten zu übertragen. Im Forum bekomme ich keine Antwort darauf.
2. Thema: Wie kann ich den Zeitstempel übernehmen? 1,0.9.1(@1,Zeitstempel,Uhr,time_stamp,0
gesendet wird: 
17:03:55 : 0.9.2(19-12-27)
17:03:55 : 17:03:56 : 0.9.1(17:04:59)
Bei mir wird nur die Stunde (hier die 17) angezeigt.
[image]

Vielen Dank im Voraus
Gerhard

gemu2015, 2019-12-27:
Hi Gerhard 
Also Punkt 1 geht leider nicht. Es gibt nur eine Baudrate keine Umschaltung.
Ist etwas langsam. Bei anderen Zählern gibt es ein Autobauding und die schalten automatisch auf 9600 um wenn mit 9600 die Anforderung gesendet wird.

Punkt 2
Uhrzeiten werden nicht unterstützt. Aber Pro Zähler kann man einen String abfragen und damit kannst du die Zeit auslesen. 

versuchs mal damit
1,0.9.1(@#),Zeitstempel,Uhr,time_stamp,0

das # Zeichen steht für Text und die Klammer ist das Endezeichen

Gruß Gerhard

PS

du kannst die Nachrichtenfunktion im Forum nutzen, da wird auch eine email geschickt

Kludi2019, 2019-12-27:
Hallo Gerhard,
danke, für die superschnelle Antwort... jetzt kommt aber der Spannungswert mit ins Datum.
[image]
Gruß
Gerhard

Kludi2019, 2019-12-27:
Vielen Dank für die schnelle Hilfe und Deinen tollen Treiber, den Du geschrieben hast!

Gruß
Gerhard

closed 2019-12-27 by Kludi2019