gemu2015/Sonoff-Tasmota #18: Please Update your fork

Hedgehog61, 2020-03-06:
Hello

To the related Issue #3405 (Platformio-core Ivan Kravitz ask me to inform you about updating your fork. Here the Original text:

No option 'platform_packages' in section: 'common''"
This bug was fixed in original repository https://github.com/arendst/Tasmota
Please ask @gemu2015 to update his fork.

Please be so kind and Update this fork.

thx

kugelkopf123, 2020-03-06:
Why don’t you use the original tasmota. The driver is integrated in it. This fork is more for development.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Hedgehog61, 2020-03-06:
But the platformio.ini has changeda view days ago. I only want to inform you about this

Von: kugelkopf123
Gesendet: Freitag, 6. März 2020 17:44
An: gemu2015/Sonoff-Tasmota
Cc: Hedgehog61; Author
Betreff: Re: [gemu2015/Sonoff-Tasmota] Please Update your fork (#18)

Why don’t you use the original tasmota. The driver is integrated in it. This fork is more for development.
Sent with GitHawk

gemu2015, 2026-10-28:
The fork was merged with the current Tasmota development branch on 27.10.2026 (see the `testing` release 1.6.75), so this request is done. Closing.

Der Fork wurde am 27.10.2026 mit dem aktuellen Tasmota-Entwicklungszweig zusammengefuehrt; damit ist das erledigt.

closed 2026-10-28 by gemu2015