gemu2015/Sonoff-Tasmota #20: Dokumentation ESP32CAM

chkbln, 2020-05-10:
### PROBLEM DESCRIPTION
Gibt es eine Dokumentation zu ESP32CAM.
Was muß ich wo einstellen?
Wird in der Web Ansicht von Tasmoto das Bild der Cam dargestellt?

Viele Grüße

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

gemu2015, 2020-05-11:
man kann die CAM sehr weitgehend mit scripts steuern. (siehe Tasmota Docs Script mit Beispiel)
oder mit Konsolen Befehlen:
wc  zeigt alle parameter an
wcstream 0 oder 1 schaltet stream aus oder an
wcresolution 0 .. 10 stellt die Auflösung ein
wcmirror 0 oder 1 spiegelt das Display
wcflip 0 oder 1 vertauscht oben und unten

es können bis zu 4 aufgenommene Bilder und ein Stream auf der Webseite angezeigt werden

[image]

chkbln, 2020-05-11:
Hi
danke für deine Infos. Leider funktionieren die Konsolen Befehle nicht. Ergebnis : {"Command":"Unknown"}
Kann es sein, dass ich beim compilieren einen Fehler gemacht habe? Mus in irgendeiner Datei angeben werden das die CAM benutzt werden soll?
Gruß

gemu2015, 2020-05-11:
ja in user_config_overwrite
#define USE_WEBCAM

chkbln, 2020-05-11:
ok hab ich gemacht. Nun bekomme ich eine Fehlermeldung. Was könnte das sein?
Init failed with error 0x20003
[image]

gemu2015, 2020-05-12:
der Fehler heißt das keine compatible Kamera gefunden wurde, aber kann auch heissen das die I2C Schnittstelle der Kamera auf den falschen Pins liegt.
du hast gar keine Pins für die Webcam definiert.
 dann wird die AITHINKER als default genommen. (default template)
Wenn du eine andere hast must du die Pins im Tasmota WEBUI definieren. Dazu musst du im Datenblatt nachsehen mit welchen GPIO deine Kamera verbunden ist.

Das ist meine (AITHINKER) 
Da ich auch eine SD Karte verwende sind hier auch SPI Pins definiert (kannst du weglassen)

[image]

riker65, 2020-09-13:
Hallo,

ich wollte die SD Card verwenden.
Verstehe nicht die PIN Belegung.

sie widerspricht der Doku hier: 
[https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/](url)

oder übersehe ich etwas

Danke T

gemu2015, 2020-09-13:
Hi,

Mit meiner CAM funktioniert die SD Card mit obiger Konfiguration

Die SD card funktioniert aber nur mit scripts. (siehe script doku)

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015