gemu2015/Sonoff-Tasmota #21: Fast relay switching based on power increase

tobetobe66, 2020-05-21:
Hi Gemu,

thanks a lot for the beautiful scripting feature.

I am using a Gosund SP1, GPIO4: 134, GPIO5: 132, GPIO12: 131, GPIO14: 21.
I am monitoring power and switch off the relay as soon as power is over 1 Watt using the following code:

>T
_pow=ENERGY#Power

>F
.
.
.
if ((_pow>1) and (_s==S)) {
->power1 0
e_eP=1
}

This works perfectly fine, however it takes sometimes way above 1 second for the relay to switch off. This is a bit too slow for me. The load is 800W, power rise should be instant.
Is there a way to do this faster? Maybe it's not possible at all, because sampling rate of power measurement is too slow.

I would be grateful for some ideas.
Thank you very much.

gemu2015, 2020-05-21:
Hi,

you get the power in the >T section. this is normally sampled on teleperiod time only at max every 10 seconds.
you can verify that by printing a time stamp in the >T section
e.g. print tele var at %upsecs%

i dont know the internals of the gosund driver and how we could get a faster access to the power register.

as far as i know there is a tasmota cmd to set limits also.

you might set such a limit and receive the alarm in the >E section

this approach may be faster

tobetobe66, 2020-05-21:
Thank you.
I will check.

tobetobe66, 2020-05-21:
OK, I found some cmds that might be of help. Have to do a bit of reading and testing. Will report back once done.

gemu2015, 2020-05-21:
because if found this an interesting application i  implemented direct access to some energy registers which are updated instantly. try my new version!
i however don't know how they are mapped in gosund

enrg[0] = should be total energy
enrg[1] = voltage p1
enrg[2] = voltage p2
enrg[3] = voltage p3
enrg[4] = current p1
enrg[5] = current p2
enrg[6] = current p3

gemu2015, 2020-05-21:
you need to define
#define USE_ENERGY_SENSOR

gemu2015, 2020-05-21:
updated again after testing with my MODBus device
added 3 power regs
enrg[7] = power p1
enrg[8] = power p2
enrg[9] = power p3

for a single phase meter only p1 values should be > 0

tobetobe66, 2020-05-21:
OK,

what I tried so far is using cmd powerdelta and cmd powerhigh.
I catch the event in the E-Section:
>E
print EVENT
if ((_pow>PON) and (_s==S)) {
->power1 0
e_eP=1
}

_pow is still in T-Section.

The result is comparable to what I had before doing polling in the F-Section.
0.5-2 seconds.

My application is a network of Gosunds to do load managment of the devices attatched to them. I need this. because I am partly OFFGRID and my inverter is limited in power. Therefor I have to detect if a device attached to a Gosund is switched on by the user. This has to be done in a very short time, as the inverter can take some overload for a short amount of time.

tobetobe66, 2020-05-21:
Thank you for the energy registers. I will give it a try and report.
Thanks a lot.

gemu2015, 2020-05-21:
you may also speed up relay switching by directly setting the relay GPIO by spin(x 1)
don't know how fast ->power1 is

tobetobe66, 2020-05-21:
I already compared spin vs ->power, there is not much difference. However I still had relay 1 defined, because otherwise power measurement didn't work. Just found out today that there is a switch to enable power measurement when power is off. Might give that a try later and compare spin vs ->power again, however I do not expect any speedup. Just compiled your version, will check energy registers now.

tobetobe66, 2020-05-21:
Hi Gemu,

holy cow, it works!! It is fast and reliable with respect to switching time.
I am way below 1 second all the time.

>T
;_pow=ENERGY#Power

>F
if ((enrg[4]>0.01) and (_s==S)) {
;->power1 0
spin(14 0)
e_eP=1
}

In the above example I used current instead of power, however that does not make a difference.
Using the registers I can skip the T-Section, saving script space, which is prime to me. Will you include the registers feature in a PR? That would be very helpful.

Thank you so much.

I have another issue which is not bothering me anymore because I found a work around. Anyway, I noticed a strange behavior using the short form of if, then, else, i.e. if {} else {}. Once enclosed in a subroutine it was gone. I will do some further testing with respect to this and maybe come back to you.

Best regards
Tobi

gemu2015, 2020-05-21:
glad that it worked!

if script size is important, use the new compress option!
#define USE_SCRIPT_COMPRESSION
(script is deleted after 1. install, so have a copy)
gives you about 50% more space

i do frequent prs every 2 or 3 days, until then use my fork

tobetobe66, 2020-05-21:
I read about rule buffer being compressed. Was hoping for that to happen for scripting also. Great.
Thanks a lot.
Shall I close the Issue?

gemu2015, 2020-05-21:
leave open, better to find for others

tobetobe66, 2020-05-22:
Hi Gemu,

Just noticed that every time I access a register a mqtt message is sent:

tele/DVES_XXXXX/SENSOR = {"Time":"2020-05-22T17:34:38","ENERGY":{"TotalStartTime":"2019-04-07T18:00:28","Total":0.085,"Yesterday":0.003,"Today":0.042,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":238,"Current":0.000}}

Is there a way to switch this off as it is causing a lot of traffic? Am I missing a Tasmota switch to turn it off?

Thanks.
Cheers tobi

gemu2015, 2020-05-22:
Hi tobi,
accessing the register should not generate a MQTT message.
it is just the same as reading an internal variable.
must be caused by something else.
you may switch off MQTT completely but i guess that is not what you want.

this is a tele message that should be generated every teleperiod seconds.

Greets Gerhard

tobetobe66, 2020-05-22:
Hmmm, strange.
Checking again. I was accessing the register in the >S section and I got mqtt messages every second. I will do some further testing. Teleperiod is definitely switched off (0).
I cannot switch off mqtt, as this is the way may Gosunds communicate.
I will be back.
Thanks for fast reply

tobetobe66, 2020-05-22:
OK, got it. Running three Gosunds in parallel, one had teleperiod on 300. thats the one I checked.
Sorry for bothering you.

tobetobe66, 2020-05-23:
Hi Gemu,

is it possible to change the last will and testament to a different topic from within the script?
The standard seems to be: tele/DVES_XXXXXXX/LWT.
I would like to change it for example to: abc/LWT.

Thanks a lot
Tobi

gemu2015, 2020-05-23:
Hi Tobi

i have not much knowledge about this MQTT stuff. scripter may only manipulate those settings via tasmota cmds.
if there is no Tasmota cmd for changing these items you might make a feature request to Tasmota dev.

Greets Gerhard

tobetobe66, 2020-05-23:
OK.
Thx a lot. I will check the code first. Maybe I can hard code it.
Tobi

tobetobe66, 2020-05-23:
Hi Gerhard,

the following script gives me a hard time:

>D
m:p_array=0 10
x=1
cnt=1

>B
for cnt 1 10 1
if cnt<4
then
print %cnt% 
else
break
endif
next

print end
print %cnt%

It seems like the break terminates the whole B section, as I get the following output:
1
2
3

and nothing else.
I really do not get it.

Thx
Tobi

gemu2015, 2020-05-24:
Hi Tobi,

This was a bug!
is fixed now.

Gerhard

tobetobe66, 2020-05-24:
Thanks Gerhard.
glad I found one! :-)

Tobi

tobetobe66, 2020-05-25:
Hello Gerhard,

see below the result of a fast polling of enrg7 and enrg3 on my gosunds after switching off.
Power (left column) still gives values long after switch off and is oscillating, whereas current is zero as it should be. As I use current it does not bother me, however the behaviour of enrg7 seems strange.

Tobi

08:25:26 P: 0.00  I: 0.00
08:25:26 P: 13.80  I: 0.00
08:25:26 P: 13.80  I: 0.00
08:25:26 P: 13.80  I: 0.00
08:25:26 P: 13.80  I: 0.00
08:25:26 P: 13.80  I: 0.00
08:25:26 P: 0.00  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 0.00  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:27 P: 13.80  I: 0.00
08:25:28 P: 13.80  I: 0.00
08:25:28 P: 0.00  I: 0.00

gemu2015, 2020-05-25:
Hi Tobi

i get the values from the energy driver, do not know how this is wired with your device
may be another register would be better, there a plenty of them, see below

could it be that your devices resolution is only 13.8 Watts?
 then 1 bit noise could produce this result

Gerhard

scripter =>
switch ((uint32_t)fvar) {
            case 0:
              fvar=Energy.total;
              break;
            case 1:
              fvar=Energy.voltage[0];
              break;
            case 2:
              fvar=Energy.voltage[1];
              break;
            case 3:
              fvar=Energy.voltage[2];
              break;
            case 4:
              fvar=Energy.current[0];
              break;
            case 5:
              fvar=Energy.current[1];
              break;
            case 6:
              fvar=Energy.current[2];
              break;
            case 7:
              fvar=Energy.active_power[0];
              break;
            case 8:
              fvar=Energy.active_power[1];
              break;
            case 9:
              fvar=Energy.active_power[2];
              break;

=> energy driver struct

struct ENERGY {
  float voltage[3] = { 0, 0, 0 };               // 123.1 V
  float current[3] = { 0, 0, 0 };               // 123.123 A
  float active_power[3] = { 0, 0, 0 };          // 123.1 W
  float apparent_power[3] = { NAN, NAN, NAN };  // 123.1 VA
  float reactive_power[3] = { NAN, NAN, NAN };  // 123.1 VAr
  float power_factor[3] = { NAN, NAN, NAN };    // 0.12
  float frequency[3] = { NAN, NAN, NAN };       // 123.1 Hz

//  float import_active[3] = { NAN, NAN, NAN };   // 123.123 kWh
  float export_active[3] = { NAN, NAN, NAN };   // 123.123 kWh

  float start_energy = 0;                       // 12345.12345 kWh total previous
  float daily = 0;                              // 123.123 kWh
  float total = 0;                              // 12345.12345 kWh total energy

  unsigned long kWhtoday_delta = 0;             // 1212312345 Wh 10^-5 (deca micro Watt hours) - Overflows to Energy.kWhtoday (HLW and CSE only)
  unsigned long kWhtoday_offset = 0;            // 12312312 Wh * 10^-2 (deca milli Watt hours) - 5764 = 0.05764 kWh = 0.058 kWh = Energy.daily
  unsigned long kWhtoday;                       // 12312312 Wh * 10^-2 (deca milli Watt hours) - 5764 = 0.05764 kWh = 0.058 kWh = Energy.daily
  unsigned long period = 0;                     // 12312312 Wh * 10^-2 (deca milli Watt hours) - 5764 = 0.05764 kWh = 0.058 kWh = Energy.daily

  uint8_t fifth_second = 0;
  uint8_t command_code = 0;
  uint8_t data_valid[3] = { 0, 0, 0 };

  uint8_t phase_count = 1;                      // Number of phases active
  bool voltage_common = false;                  // Use single voltage
  bool frequency_common = false;                // Use single frequency
  bool kWhtoday_offset_init = false;

  bool voltage_available = true;                // Enable if voltage is measured
  bool current_available = true;                // Enable if current is measured

  bool type_dc = false;
  bool power_on = true;

#ifdef USE_ENERGY_MARGIN_DETECTION
  uint16_t power_history[3] = { 0 };
  uint8_t power_steady_counter = 8;  // Allow for power on stabilization
  bool power_delta = false;
  bool min_power_flag = false;
  bool max_power_flag = false;
  bool min_voltage_flag = false;
  bool max_voltage_flag = false;
  bool min_current_flag = false;
  bool max_current_flag = false;

#ifdef USE_ENERGY_POWER_LIMIT
  uint16_t mplh_counter = 0;
  uint16_t mplw_counter = 0;
  uint8_t mplr_counter = 0;
  uint8_t max_energy_state  = 0;
#endif  // USE_ENERGY_POWER_LIMIT
#endif  // USE_ENERGY_MARGIN_DETECTION
} Energy;

tobetobe66, 2020-05-25:
Energy resolution is far better than 13.8Watts. I don't think its because of noise.
I can only use two registers, as it is a one phase system. I can live with the current register. 

Was just wondering about the readings of the power register. It reads 13.8W long after the device is switched off and also when current is already 0 for some time.

tobetobe66, 2020-05-26:
Dear Gerhard, 

the docu states: 
There are two syntax alternatives for conditional statements (if, then, else or if {} else {}). You may NOT mix both formats.

Does this NOT mixing refer to the whole script or just to not nesting both formats?

Thx
Tobi

gemu2015, 2020-05-26:
Hi Tobi,

just messing in one condition is not supported.

Gerhard

tobetobe66, 2020-05-26:
Hi Gerhard,

I have a strange behavior in the following code if I change the highlighted section from if then to if{}. The function FSM is called from the F Section.
It does not work with, however both variants should be equivalent. 
if e_eP==1 {
_s=N
sc=1
}

If I use {} the part with
e_Te=0
e_Mr=0
e_eP=0

seems to be skipped. I.E. in the print statement e_Mr is 1 instead of zero.
I do not see where I could have made an error as it works with if then.

Thx Tobi

#FSM
print BEG FSM: sc:%sc% State:%_s% e_eP: %e_eP%, e_Te: %e_Te% e_Mr: %e_Mr%
sc=0
switch _s
case S
print S
->power1 1
**_if e_eP==1
then
_s=N
sc=1
endif_**

case N
print N
->power1 0
if r==1
then
_s=O
sc=1
endif

case O
print O
->power1 1
ends

print clear events
e_Te=0
e_Mr=0
e_eP=0

print EOF FSM: sc:%sc% State:%_s% e_eP: %e_eP%, e_Te: %e_Te% e_Mr: %e_Mr%

gemu2015, 2020-05-26:
Hi Tobi,

Strange, i checked both variants and there is no difference.
However i simply checked the comparison itself with your var names, not within the switch case statements
From the source code i could not figure out why there should be a difference when inside switch ends
must be a complex interaction i cant explore.
if you can provide a complete scipt that reproduces the problem i will check it in detail.

Cheers

 Gerhard

tobetobe66, 2020-05-26:
What is the best way to send you the script?
tobi

gemu2015, 2020-05-26:
maybe attach it as a zip file

tobetobe66, 2020-05-26:
ok, i attached a text file.
its a simple state machine, reacting on two events e_Mr and e_eP.
if you need any explanation, let me know
thx a lot tobi
[gemu_ifthen.txt](https://github.com/gemu2015/Sonoff-Tasmota/files/4684028/gemu_ifthen.txt)

gemu2015, 2020-05-26:
very complex script!

ok to get it fired i had to put
_s=S
e_eP=1
=#FSM

into the >F section

and it behaves like it should, no difference between the 2 forms.
so its not related to the switch case

by the way. a great problem is that there is nearly no syntax check (because of limited code space) and errors are often discarded but produce strange behavior.

i discussed that with another user who also writes complex scripts (with strange behavior
due to syntax errors) that it would be nice to at least have a syntax colorizer to better find misspelled items.
a complete external syntax check would be a bit of hard work but a simple colorizer could be easily made with notepad++ or similar.

e.g. you have a =#TimerT1Expired subroutine call with a missing subroutine. not shore what happens in this case.

still possible that if your complex script runs with all inputs there is an interference but since i can't reproduce it easily i have no guess as to use the working form for the time being.

tobetobe66, 2020-05-26:
Dear Gerhard,

thanks for checking. I know its complex, I was aware of the missing sub once i posted it. I checked it, its not the missing subroutine that makes a difference. Also I noticed that in my file WORKING and NOT WORKING was the same.
But as I unterstood you noticed that and checked both. Sorry for that.
The strange behaviour starts once the script reaches case O.
I had that before when I use if {}, no probs sofar when I use if then.

If you find the time and are willing to spend some more time (which to be honest I do not expect, because I know of the complexity) I attached an updated file with both versions, one commented out correctly.
If you run each version and input on the console the two following lines you will see the difference. (I checked it)
[gemu2_ifthen.txt](https://github.com/gemu2015/Sonoff-Tasmota/files/4684525/gemu2_ifthen.txt)

script >e_eP=1 ;takes you from case S to case N
Backlog script >r=1; script>e_Mr=1. ; takes you from N to O

The non working version continuously keeps on printing on the console.

Thanks again. 
Tobi

gemu2015, 2020-05-29:
Hi Tobi,

Danke gerne geschehen. Hatte die Syntax von Rules nie gemocht und deshalb den Scripter so zum Spass komplett vom Scratch codiert.
Du macht auch echt komplexe Sachen damit. Ich selbst bin ein Fan von autarken Geräten die im Zweifelsfall auch ohne Broker funktionieren sollen. Habe zahlreiche Geräte mit Display (inzwischen einige auf ESP32 umgerüstet) die ich als Grafikanzeige einiger Wechselrichter und Umweltdaten verwende. Da sind die Scripte nicht ganz so komplex.
 
Bezüglich deiner Anwendung mit dem Wildcard '+' bei MQTT ist das übrigens der einzige Kode den ich von Rules übernommen habe (SUPPORT_MQTT_EVENT) und der kann das Wildcard auch bei Rules nicht.

Grüße

Gerhard

tobetobe66, 2020-05-30:
Hallo Gerhard,

das mit Wildcard '+' habe ich mir fast gedacht. Bekommst wohl auch alles mit. :-)

Ich bevorzuge auch autarke Systeme, aber ich brauche eben einen Kommunikationskanal für meine Gosunds SP1, da diese die Leistung meines Victron Batterie-Wechseltrichters nach Bedarf aufteilen. Wüsste nicht wie ich das ohne Kommunikation machen sollte. Da bietet sich mqtt natürlich an. Dein Scripting feature funktioniert tadellos, auch bei komplexen Dingen. Seit der Möglichkeit Skripte zu komprimieren verwende ich konsequent if then else und habe keine Probleme mehr.

Was ich jetzt noch rausfinden muss ist, ob ich auch mit QOS1 arbeiten kann. Das würde die Kommunikation zwischen den Knoten deutlich vereinfachen. Angeblich muss man nur die library wechseln.

Grüsse
Tobi

gemu2015, 2020-05-30:
Hi Tobi
Das mit QOS1 wurde schon mal irgendwo bei Tasmota chats oder so diskutiert und Theo wollte das nicht einbauen. Kenn mich leider kaum mit MQTT aus.
Was genau brauchst du eigentlich ? Du hast mehrere SP1 die du koordinieren willst aber ohne sie per IP oder so anzusprechen oder abzufragen ? Warum das ? Bei mir haben die Geräte fixe IPs. 
Ich habe solche Dinge schon mit WebSend gemacht, entweder direkt Werte von anderen Knoten abgeholt oder mir von dort was schicken lassen.  And dem Timestamp kann man dann sehen ob der Knoten seit einiger Zeit nicht gesendet hat und ausgefallen ist.

Gruß 
Gerhard

tobetobe66, 2020-05-30:
Hi Gerhard,

die Aufgabe ist folgende: 
1. An jedem SP1 ist ein Verbraucher eingesteckt (Waschmaschine, Fön, Mikrowelle, Spülmaschine etc...)
2. Die elektrische Leistung in meinem Hausnetz ist begrenzt.
3. Die Verbraucher haben unterschiedliche Priorität (Fön z.B Prio 9, Waschmaschine Prio 1 etc...)
4. Wird ein Verbraucher eingeschaltet bekommt der Gosund das durch den Stromanstieg mit. Dann schaltet er zunächst den Verbraucher ab und fordert die maximal vom Verbraucher benötigte Leistung an. Hierzu gibt es 2 Lösungsansätze. Dies kann entweder zentral (ein Master Gosund der alles verwaltet) oder dezentral bei allen anderen Gosunds geschehen.
5. Bei der Anforderung eines Gosund wird geprüft ob noch genügend Netzleistung zur Verfügung steht. Ist dies der Fall, darf er sich einschalten, falls nicht, werden solange andere Gosunds mit niedrigerer Priorität abgeschaltet, bis genügend Leistung zur Verfügung steht. Wird der Verbraucher abgeschaltet oder seine Leistung fällt für eine gewisse Zeit unter einen einstellbaren Wert, gibt er die Leistung wieder frei. Die abgeschalteten Gosunds konkurrieren dann wieder um Leistung und schalten sich wieder ein.

Anforderungen:
- zuverlässig (definierter Zustand auch bei WLAN-Ausfall, wenn ein Gosund ausgesteckt wird muss dies zu einem definierten Gesamtzustand führen, z.B. muss falls der Gosund eingeschaltet war die Leistung wieder freigeben werden.)
- skalierbar, einfach einen weiteren Gosund einstecken und läuft (es ist nur die Priorität sowie im Falle von mqtt die Adresse des Brokers zu konfigurieren)
- wenig Administrationsaufwand

Zu deiner Frage mit festen IPs:  
Ja, kann man machen, die muss dann jeder client oder der master (im Falle zentraler Ansatz) kennen. Bisschen mehr administrativer Aufwand.
Zunächst hatte ich auch WebSend auf dem Schirm. Dann hatte ich mir mqtt angesehen und festgestellt, dass es eigentlich ideal zu meinem Problem passt. Leider sind einige mqtt Features nicht in Tasmota implementiert bzw. nicht voll konfigurierbar (z.B. QOS1, LWT), dies macht es für mich skriptseitig etwas aufwändiger. Habs aber hinbekommen und bin jetzt an der Optimierung (Dank mehr Platz wegen script compression).

Der mqtt-Broker läuft auf einem alten HTC-Handy, hat sozusagen die USV an Bord.

Für Verbesserungen oder alternative Ansätze bin ich immer dankbar. :-)

Gruss
Tobi

gemu2015, 2020-05-30:
Hi Tobi,

Das ist wirklich ne coole Sache!  Macht bestimmt Spass das zu programmieren. Stelle ich mir aber auch schwierig vor wenn ich mich föhnen will und das Ding geht nicht an weil die Waschmachine läuft.

Habe selbst schon lange Photovoltaik und hatte dafür eine große LED Anzeige mit rot und grüner Skala. Habe meiner Frau beigebracht sie soll die großen Verbraucher erst einschalten wenn die Skala soundso hoch ist. Im Broker habe ich in der Heizperiode sogar einen Eheizer automatisch angeschaltet wenn es Überschuss gab.
Inzwischen haben wir ne 13.5 KWh Pufferbatterie und damit ist Ruhe. Habe praktisch 8 Monate im Jahr gar keinen Verbrauch mehr aus dem Netz trotz Klimaanlage und Poolwämepumpe.

Eine andere Methode für dich wäre eventuell mit UDP broadcasts zu arbeiten.

Naja wenn du was brauchst frag nach und ich versuch es dir einzubauen.

Gruß
Gerhard

tobetobe66, 2020-05-30:
Hi Gerhard,

ja, macht Spass.
Zum Fön:
Der Fön hat eine ganz hohe Prio. Der schaltet alle anderen ab falls nötig (nach Prio). Bedeutet, Du schaltest den Fön ein, er läuft für max 1 Sekunde, schaltet sich ab und in der Regel nach ca. 1 Sekunde wieder ein und Du föhnst solange Du willst. Die Waschmaschine hat eine niedrige Prio und wird in dieser Zeit abgeschaltet. Wenn der Fön wieder aus ist, schaltet sich die Wama wieder ein und macht da weiter wo sie war, da sie dankenswerterweise nach Stromausfall da weitermacht wo sie war. Das gilt übrigens auch für Spülmaschinen. Die ca. 3 Sekunden Totzeit halte ich für vertretbar. Dafür muss ich meinen 3 Damen im Haus nix beibringen:-).

zum Einbauen für weitere Features:
Das Scripting ist praktisch perfekt. Konnte bisher alles mit vorhandenen Funktionen lösen. Vielen Dank. Einziger kritischer Punkt ist bisher das schnelle Abschalten. Mal sehen ob das in der Praxis reicht. Ansonsten müsste man mal die Tasmota Cracks  fragen ob sich die Samplingrate bei der Strommessung erhöhen lässt. Das hat aber noch Zeit.

Jetzt bin ich neugierig auf dein Solarsetup.
- Hast Du reinen Batteriewechselrichter oder einen mit Netzanbindung der bei Bedarf umschaltet?
- Welche Batterietechnologie nutzt Du?

Bei mir:
- reiner Batteriewechselrichter von Victron, 24/3000
- 300AH LiFePo, Systemspannung  25,6V
- Solar 3,25kWp, kein MPPT, direkt auf Batterie, gesteuert von BMS

Bin gespannt.
Gruss Tobi

gemu2015, 2020-05-30:
habe auf dem Dach 5,4 KW + Solarthermie, auf der Garage 3,4 KW und auf dem Gartenhaus 3 KW und eine Tesla Powerwall 2 (Lipo)

gemu2015, 2020-05-30:
PS die Tasmota Energie Steuerung hab ich mir angesehen die ist mit einer Sekunde getaktet das kann man wohl nicht schneller machen

tobetobe66, 2020-05-30:
Hab was vergessen:
- Will auch 8 Monate ohne Verbrauch sein (Offgrid)
- 8 Stromkreise manuell zwischen Netz und Batterie umschaltbar (Übergangszeit)

tobetobe66, 2020-05-30:
Respekt. Bist zufrieden mit der Powerwall?
Bei mir ist alles Selbstbau, Zellen aus China.

Wo finde ich die Energiesteuerung, welches File?

gemu2015, 2020-05-31:
Ja die Powerball ist pflegeleicht alles prima.
Habe auch lange überlegt und recherchiert. 
Off Grid hat den Vorteil das man nichts zulassen muss und alles selber bauen kann.
Aber so eine Lösung hätten meine Frauen nicht akzeptiert. Die Gartenhausanlage habe ich selbst konzipiert und aufgebaut. War aber schwierig einen Solateur zu finden der mir das angemeldet hat.
Bei der Batterie gab es bis dato keine preiswertere als die Tesla Powerwall 2 (5000,- für 13.5 Kwh, MWSt habe ich ja zurück bekommen) Da ist ja alles drin auch ein 4,6 kW Batteriewechselrichter.
Völlig transparente Steuerung wird einfach parallel angeschlossen mit Stromzangen für alle Phasen als Sensoren.
 
Selbst bei den Chinesen hab ich nichts preiswerteres gesehen.
Was hast du denn für deine nackte Batterie bezahlt (7.5kwh)

Die Tasmota Energy files:

xnrg_01_hlw8012.ino sonoff pow .
xnrg_02_cse7766.ino sonoff pow r2
xnrg_04_mcp39f501.ino für shelly 2
xnrg_07_ade7953.ino für shelly 2.5

Laut kurzer Recherche ist wohl dein Chip derselbe wie beim pow (HLW8012)

tobetobe66, 2020-05-31:
Die 24 Zellen haben inkl. Lieferung 1600€ (inkl. MWSt.) gekostet.
Dann ca. 250€ für BMS (electrodacus.com).
Wechselrichter 3kW 1200€.
Das sind so die wesentlichen Kosten. Dann noch Batteriekabel etc.. ca. 300€. Muss das mal genau zusammenrechnen.
Würde ich die Batteriekapazität verdoppeln, läge ich dann bei ca. 5000€. Wechselrichter ist aber kleiner als deiner.

Bin mal in xnrg_01_hlw8012.ino rein. Ich glaube die Energie wird im Sekundentakt abgefragt. Spannung, Strom, Leistung im 200ms Takt.

Naja, ich bringe jetzt mal die Gosunds zum Laufen und dann teste ich mal mit echten Verbrauchern.

Grüsse Tobi

tobetobe66, 2020-05-31:
Gerhard,

ich bin ein Esel. Habe glaube ich meinen Fehler bei if{}else{} entdeckt.
Sind Kommentare nur in einer eigenen Zeile erlaubt?
Also z.B nicht:

>D
a=0; meine super Variable

Ich muss das später mit dem alten Code mal testen.

Gruss Tobi

gemu2015, 2020-05-31:
Ja so ist es, bei ; mitten in einer Zeile weis man nicht genau was passiert!

tobetobe66, 2020-05-31:
Verständnisfrage
Wenn ich t:xyz als timer verwende geht das nur auf Sekundenbasis oder?
Wenn ich was in ms will, dann muss ich mir über upsec,uptime was in der F-Section bauen, das wäre dann   in 0,1s Schritten oder?
Danke
Gruss Tobi

gemu2015, 2020-05-31:
Ja die t:xx timer sind in Sekunden.

in der >F Sektion kannst du halt in 100ms Schritten selbst zählen. Darauf kannst du dich aber nicht wirklich verlassen, weil die >F section gepollt wird es können auch mehr als 100ms zwischen 2 polls vergehen.

tobetobe66, 2020-06-03:
Hallo Gerhard,

wollte mal kurz Rückmeldung geben.
Also es funktioniert alles. Heute mal mit 3 Gosunds getestet.
Gosund 1: niedrige Prio mit 800W Wasserkocher
Gosund 2: mittlere Prio mit 2000W Heizofen
Gosund 3: hohe Prio mit 2000W Fön
Maximale Leistung 2800W eingestellt.
Wasserkocher und Fön eingeschaltet, die loggen sich ein und laufen.
Danach den Fön dazu. In etwa 1 Sekunde hat der Fön sich eingeschaltet und die beiden anderen rausgeschmissen. Danach hat sich der Wasserkocher wieder zugeschaltet, da noch Lesitung frei war.
Das hat absolut alltagstauglich funktioniert, dank der tollen Scriptingmöglichkeit.

Folgende Störfälle kann ich erkennen und zeitnah 10 sec. bearbeiten:
-MQTT Verbindung reist ab, dann wird alles wieder neu gestartet
-Ein oder mehrere Gosunds fallen aus, dann wird alles wieder neu gestartet

Falls bei bestehender MQTT-Verbindung Nachrichten verloren gehen, führt dies zu einem instabilen System. Aber ich denke die Wahrscheinlichkeit geht bei gutem WLAN wegen darunterliegendem TCP gegen Null. Evtl. baue ich noch ein, dass verlorene Nachrichten nach 10 Sekunden wieder zu einem stabilen Zustand führen.

Danke nochmals für deine Hilfe.
Beste Grüsse
Tobi

tobetobe66, 2020-06-03:
Hallo Gerhard,

habe bereits ein neues Scripting-Projekt auf dem Schirm:
DCF 77-Funkuhr mit Script.

Dazu muss ich 1 mal pro Sekunde eine Pulsdauer messen ansteigende Flanke, fallende Flanke.
Ist das im Script möglich?
Ich muss noch mal genauer prüfen was Tasmota hier bietet. Bin mir nicht sicher, ob die Zeit zwischen steigender und fallender Flanke messbar ist. Du stellst ja im Script mit pc[x] was zur Verfügung.

Für Erleuchtung bin ich stets dankbar.

Grüsse
Tobi

gemu2015, 2020-06-03:
Hi Tobi,
Freut mich wirklich dass es so gut funktioniert! Ist ne tolle Anwendung!
Vielleicht hast du ja Lust das mal bei den Beispielen einzustellen.

Der Tasmota Counter kann Pulslängen messen das weis ich.
Das Ergebnis kannst du dann mit pc[x] abholen.

Eigentlich wundert es mich dass es für DCF77 noch keinen eigenen Tasmota Treiber gibt.

Grüsse

Gerhard

tobetobe66, 2020-06-03:
Hi Gerhard,

stelle das gerne bei den Beispielen ein. Muss aber noch ein bisschen "beautyfyen".
Ausserdem möchte ich es gerne bei mir min. 2 Wochen live testen.
Die Skripte sind auch ziemlich lang. Kann das jeder einstellen?
Habe mir auch überlegt, ob ich das mal in ein Solarforum einstelle. Was meinst Du?

Bezüglich DCF77, ich probiere mal, ob ich das im Skript hinbekomme. Dann kann da gerne jemand einen Treiber für GPIO xyz erstellen.

Mein Anspruch ist es im Skript zu schaffen.
:-)

Gruss
Tobi

gemu2015, 2020-06-03:
im Prinzip kann das jeder einstellen! Ist ein pull request bei Tasmota Docs.
Da ich Schreibrechte habe könnte ich es auch für dich einstellen.

Ja macht Spass das "Scripten".  Leider hat der ESP8266 nicht so viele Möglichkeiten wie der ESP32.  Zur Zeit arbeite ich fast nur noch damit. Da ist halt ein Realtime Multitasking Betriebssystem darunter (FreeRTOS). Da kannst du z.B. Scripter Tasks als Multitasking starten.
Die laufen weiter auch wenn Tasmota hängt.

tobetobe66, 2020-06-03:
Das mit dem ESP32 ist natürlich toll. Das werde ich mir als nächstes ansehen.
Die 8266 sind halt in vielen Steckdosen, Schaltern etc.. verbaut. Fertige Hardware eben.
Die 32er kommen hoffentlich bald in kommerzielle Plattformen (Stecker dran:-))

Darauf freue ich mich jetzt schon.
Bezüglich der Anwendung mit den Gosunds teste ich und mache die Doku fertig. dann melde ich mich bei Dir.

tobetobe66, 2020-06-03:
Hi Gerhard,

brauche deine Hilfe. Für DCF77 muss ich an einem GPIO die Dauer eines Pulses (Dauer zwischen steigender Flanke und fallender Flanke) messen. So wie das aussieht sprechen die Counter nur auf fallende Flanken an. Habe ich was übersehen oder muss man da mit Switches arbeiten?

Hast Du eine Idee?

gemu2015, 2020-06-03:
soweit ich das verstanden habe kann man doch die Periodendauer messen ?
CounterType = 1  => Pulse Timer liefert die Pulszeit des Low Pulses 
Da die Pulse bei DCF immer im Sekundenabstand kommen kannst du ja 1 sec - counter Wert nehmen. du musst ja nur zwischen 100 und 200 ms unterscheiden, wären dann halt 800 oder 900
ms

[37 more replies]

closed 2026-10-28 by gemu2015