gemu2015/Sonoff-Tasmota #22: TuyaReceived values under >E

blakadder, 2020-07-17:
Maybe I was doing it wrong but it seems I cannot capture TuyaReceived events into a variable. 

Zigbee is documented as supported but I assume Tuya isn't?

Unrelated question: Is there any way to truncate a variable string, for example I only need the last digit from "0302000400000000"?

gemu2015, 2020-07-17:
TuyaReceived will arrive in >E too

i own no Tuya device but

strvar=TuyaReceived#Data

should do the job

substring:
svar=sb(src_svar p n)

p=position (if <0 counts from end)
n=number of chars

so svar=sb(src_svar -1 1)

should give the last digit

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015