gemu2015/Sonoff-Tasmota #24: very frequent reboots due to exceptions while processing script

bs-github, 2020-07-28:
### PROBLEM DESCRIPTION
I'm using a script to read the ORNO OR-WE-516 3-phase energy meter with tasmota on a D1-mini via MODBUS RS485.
It works in principle, but I have reboots every few seconds.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1-mini
- [x] Tasmota binary firmware version number used: 8.3.1(sensors) 2_7_1/2.2.2-dev(38a443e)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO, version 4.3.4
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:18:26 CMD: Backlog Template; Module; GPIO 255
15:18:26 MQT: stat/orno-smartmeter/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:18:26 MQT: stat/orno-smartmeter/RESULT = {"Module":{"0":"Generic"}}
15:18:26 MQT: stat/orno-smartmeter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [x] If using script, provide the script:
I've stripped the script down to three values to limit the output in this issue.
```
>D
>B
=>sensor53 r
>M 1
+1,3,M,0,9600,ORNO516,1,1,01030014,0103000E,01030010,01030012
1,010304ffffffff@i0:1,Grid Frequency,Hz,GridFrequency,2
1,010304ffffffff@i1:1,Voltage P1,V,Voltage_P1,2
1,010304ffffffff@i2:1,Voltage P2,V,Voltage_P2,2
1,010304ffffffff@i3:1,Voltage P3,V,Voltage_P3,2
#
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:21:13 CMD: Status 0
15:21:13 MQT: stat/orno-smartmeter/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"orno-smartmeter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:21:13 MQT: stat/orno-smartmeter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.188.253:8001/tasmota-sensors-orno.bin","RestartReason":"Exception","Uptime":"0T00:42:46","StartupUTC":"2020-07-28T13:38:27","Sleep":50,"CfgHolder":4617,"BootCount":198,"BCResetTime":"2020-07-25T18:56:24","SaveCount":266,"SaveAddress":"FB000"}}
15:21:13 MQT: stat/orno-smartmeter/STATUS2 = {"StatusFWR":{"Version":"8.3.1(sensors)","BuildDateTime":"2020-07-17T09:06:25","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"371/699"}}
15:21:13 MQT: stat/orno-smartmeter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["adrion.netgaroo.com",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:21:13 MQT: stat/orno-smartmeter/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","87DAA787","041DA3A5","B7FFBFCD","01DA9BC4","64367CCF","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,53,55,56,58,59,62,64,66,67,91"}}
15:21:13 MQT: stat/orno-smartmeter/STATUS5 = {"StatusNET":{"Hostname":"smart-meter","IPAddress":"192.168.188.27","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.1","Mac":"84:F3:EB:5A:46:96","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:21:13 MQT: stat/orno-smartmeter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A4696","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:21:13 MQT: stat/orno-smartmeter/STATUS7 = {"StatusTIM":{"UTC":"2020-07-28T14:21:13","Local":"2020-07-28T15:21:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:19","Sunset":"20:33"}}
15:21:13 MQT: stat/orno-smartmeter/STATUS10 = {"StatusSNS":{"Time":"2020-07-28T15:21:13","ORNO516":{"GridFrequency":49.97,"Voltage_P1":225.70,"Voltage_P2":226.00,"Voltage_P3":226.50}}}
15:21:13 MQT: stat/orno-smartmeter/STATUS11 = {"StatusSTS":{"Time":"2020-07-28T15:21:13","Uptime":"0T00:42:46","UptimeSec":2566,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"Wifi":{"AP":1,"SSId":"adrion.netgaroo.com","BSSId":"98:9B:CB:A2:75:3A","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}
15:21:13 MQT: stat/orno-smartmeter/STATUS12 = {"StatusSTK":{"Exception":9,"Reason":"Exception","EPC":["4022e2bb","00000000","00000000"],"EXCVADDR":"3fff5501","DEPC":"00000000","CallChain":["4022e2b9","4023d5a3","4023d03c","4021439e","40238f2b","4024f8a4","4010106d"]}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
```
  Console output here:

00:00:00 CFG: Loaded from flash at F6, Count 271
00:00:00 QPC: Flag 7E
00:00:00 Script: nv=0, tv=0, vns=0, ram=16
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1(sensors)-2_7_1
00:00:00 Script: performs "sensor53 r"
00:00:00 SRC: Rule
00:00:00 CMD: Group 0, Index 53, Command "SENSOR", Data "r"
00:00:00 SNS: Hardware Serial
00:00:00 RSL: stat/orno-smartmeter/RESULT = {"Time":"1970-01-01T00:00:00","SML":{"CMD":"restart"}}
00:00:00 SNS: Hardware Serial
[... 89 lines cut ...]
14:34:27 : 01 03 04 43 62 19 9a c5 92
14:34:27 : 01 03 04 43 61 e6 66 74 23
14:34:27 : 01 03 04 42 48 1e b8 67 8f
14:34:28 : 01 03 04 43 61 19 9a 35 92
14:34:28 : 01 03 04 43 62 19 9a c5 92
14:34:28 : 01 03 04 43 61 e6 66
14:34:28 : 74 23
14:34:28 : 01 03 04 43 62 00 00
14:34:28 : 4e 69
14:34:28 : 01 03 04 42 48 1e b8 67 8f
```

### ADDITIONAL CONTEXT
I think the problem might be in the code that processes the data from the meter/modbus/serial.

I do see valid results every time. But I see data missing often as well (zero values for where should be higher meter readings). 
It might be that the rs485 line is not very clear and there might come in wrong values? Or is that ruled out with a checksum (not to my knowledge)?
Bad data that could not be parsed correctly would explain that there are missing values.

The reboot problem vanishes when I turn on the debug dump (CMD: sensor53 d1), but then I do not get the desired data obviously. The data as well as the reboots immediately come back when I turn the dump off again (CMD: sensor53 d0). 
So as long as the data is not processed (but fetched) the WeMo D1-mini and tasmota work reliable, but as soon as the data is processed (for web display?) the exception occurs.

Side note: sometimes the watchdog initiate the restart.

What can I do to debug this?

gemu2015, 2020-07-28:
your descriptor looks OK
you should start with a minimal version to trace down the bug.
use a single value at first

the only thing in doubt you use only 100ms for value transmissions. For some meters this is too short.
try with larger values eg 300 ms = 3
+1,3,M,0,9600,ORNO516,1,3,01030014,0103000E,01030010,01030012

however still strange that this should cause the failure

transmissions and received data are checksummed so only valid values are processed. 

there are many users using this driver with modbus devices and no issues so i doubt there is a general software issue.

gemu2015, 2020-08-03:
did you solve the issue ?

bs-github, 2020-08-03:
I tried with larger values 3,5,7 and even 12 for value transmissions and the restarts/exceptions do happen less frequent with higher values, and do not occur at 12 anymore.
I still do think there is something that can be done in the code to prevent the restarts. 
How would you explain that there are no exceptions while just dumping the received values instead of processing them?

gemu2015, 2020-08-03:
i checked the code and found that with a very faulty serial interface a buffer overrun could occur.
this is now fixed.
with working serial connection i never saw an exception in my own sdm530 for month.

you should check your hardware serial to rs485 interface.

bs-github, 2020-08-03:
How do I check that hardware? 
And how can I be sure that the data is faulty in the first place? 
I assume I do not see any faulty data (just no data) due to the way the checksum thingy works.

bs-github, 2020-08-03:
I do have quite some values read now by the script. But I'm not finished yet.
I'll just put that here for completeness/reference, albeit my scripting issues should be discussed somewhere else (new issue / chat?).

```
>D
oldFL1=0
;Fill vars with content on teleperiod
>T
oldFL1=#L1ForwardActiveEnergy
>B
=>sensor53 r
;Set teleperiod to 10sec
tper=10
>S
>M 1
+1,3,M,0,9600,ORNO516,1,12,0103010A,0103010C,0103010E,01030112,01030114,01030116,01030014,0103000E,01030010,01030012,0103001C,0103001E,01030020,01030022,01030102,01030104,01030106,01030100
1,010304ffffffff@i0:1,L1 Forward Active Energy,kWh,L1ForwardActiveEnergy,2
1,010304ffffffff@i1:1,L2 Forward Active Energy,kWh,L2ForwardActiveEnergy,2
1,010304ffffffff@i2:1,L3 Forward Active Energy,kWh,L3ForwardActiveEnergy,2
1,010304ffffffff@i3:1,L1 Reverse Active Energy,kWh,L1ReverseActiveEnergy,2
1,010304ffffffff@i4:1,L2 Reverse Active Energy,kWh,L2ReverseActiveEnergy,2
1,010304ffffffff@i5:1,L3 Reverse Active Energy,kWh,L3ReverseActiveEnergy,2
1,=h==================
1,=d 1 60 @0.01,Delta FAEP1,W,DFAEP1,2
1,=d 2 60 @0.01,Delta FAEP2,W,DFAEP2,2
1,=d 3 60 @0.01,Delta FAEP3,W,DFAEP3,2
1,=h==================
1,=d 4 60 @0.01,Delta RAEP1,W,DRAEP1,2
1,=d 5 60 @0.01,Delta RAEP2,W,DRAEP2,2
1,=d 6 60 @0.01,Delta RAEP3,W,DRAEP3,2
1,=h==================
1,010304ffffffff@i6:1,Grid Frequency,Hz,GridFrequency,2
1,010304ffffffff@i7:1,L1 Voltage,V,L1Voltage,2
1,010304ffffffff@i8:1,L2 Voltage,V,L2Voltage,2
1,010304ffffffff@i9:1,L3 Voltage,V,L3Voltage,2
1,010304ffffffff@i10:0.01,Total Active Power,W,TotalActivePower,2
1,010304ffffffff@i11:0.01,L1 Active Power,W,L1ActivePower,2
1,010304ffffffff@i12:0.01,L2 Active Power,W,L2ActivePower,2
1,010304ffffffff@i13:0.01,L3 Active Power,W,L3ActivePower,2
1,010304ffffffff@i14:1,Total Energy,kWh,TotalEnergy,2
1,010304ffffffff@i15:1,L1 Total Energy,kWh,L1TotalEnergy,2
1,010304ffffffff@i16:1,L2 Total Energy,kWh,L2TotalEnergy,2
1,010304ffffffff@i17:1,L3 Total Energy,kWh,L3TotalEnergy,2
1,=h==================
1,=d 15 60 @1,Total Active Energy d,W,DTAE,2
#
```

The values for the delta calculations doesn't seem right as of now, but I have to look into this a bit more to understand it.

I think **oldFL1=#L1ForwardActiveEnergy** does not work as intended (so I'm not sure how to access the values read from the meter within the script) and my delta definitions (like **1,=d 15 60 @1,Total Active Energy d,W,DTAE,2**) are probably wrong as well.

I'd like to play around with the delta thing and probably with calculating some values in >T or >S to finally get to the point where I can tell for sure if the energy is actually consumed or generated. The meter itself can count forward and reverse energy, but Active Power is always positive, even when the energy is flowing out.

gemu2015, 2020-08-03:
1. do you still get exceptions ?
2. did you set SML_MAX_VARS to a higher value ?
default is 20. you use more then 20 lines in your descriptor (html only lines dont count)
this also leads to failure.
3. you dont need to get vars via JSON. you may use sml[x] to access sml descriptor lines
anyhow your syntax is wrong, this is the right one
oldFL1=ORNO516#L1ForwardActiveEnergy

i can recommend this rs485 interface. it works perfect
https://www.ebay.de/itm/TTL-RS485-Adapter-485-UART-Seriell-3-3V-5-Volt-Level-Konverter-Modul-Arduino/252852473229?hash=item3adf2e8d8d:g:DQwAAOSw5UFb4H25

bs-github, 2020-08-04:
> 1. do you still get exceptions ?

No, not with ```+1,3,M,0,9600,ORNO516,1,12,...```

> 2. did you set SML_MAX_VARS to a higher value ?
>    default is 20. you use more then 20 lines in your descriptor (html only lines dont count)
>    this also leads to failure.

yes, I did ```#define SML_MAX_VARS 25``` in ```tasmota/user_config_override.h```

> 3. you dont need to get vars via JSON. you may use sml[x] to access sml descriptor lines
>    anyhow your syntax is wrong, this is the right one
>    oldFL1=ORNO516#L1ForwardActiveEnergy

Thanks, I've got values in the script now.

> i can recommend this rs485 interface. it works perfect
> https://www.ebay.de/itm/TTL-RS485-Adapter-485-UART-Seriell-3-3V-5-Volt-Level-Konverter-Modul-Arduino/252852473229?hash=item3adf2e8d8d:g:DQwAAOSw5UFb4H25

That's what I have (from a different seller).

bs-github, 2020-08-08:
Hello,

> > 1. do you still get exceptions ?
> 
> No, not with `+1,3,M,0,9600,ORNO516,1,12,...`

I would be able to get back to smaller values if you want me to test something.
But I do not have to solve this issue because I get the data that I need reliably now.

> > 3. you dont need to get vars via JSON. you may use sml[x] to access sml descriptor lines
> >    anyhow your syntax is wrong, this is the right one
> >    oldFL1=ORNO516#L1ForwardActiveEnergy
> 
> Thanks, I've got values in the script now.

I could not get the values with sml[x] and the delta definitions did not work for me either, but those drawbacks are no issue for me and I'm quite happy with the following version of my script for the ORNO or-we-516 smart meter:
```
>D
;now
F1=0
F2=0
F3=0
R1=0
R2=0
R3=0
;before
f1=0
f2=0
f3=0
r1=0
r2=0
r3=0
;in or out
io1=1
io2=1
io3=1
;active power
A1=0
A2=0
A3=0
a1=0
a2=0
a3=0
;Fill vars with content on teleperiod
>T
F1=ORNO516#L1ForwardActiveEnergy
F2=ORNO516#L2ForwardActiveEnergy
F3=ORNO516#L3ForwardActiveEnergy
R1=ORNO516#L1ReverseActiveEnergy
R2=ORNO516#L2ReverseActiveEnergy
R3=ORNO516#L3ReverseActiveEnergy
A1=ORNO516#L1ActivePower
A2=ORNO516#L2ActivePower
A3=ORNO516#L3ActivePower
if R1>r1
then
r1=R1
io1=-1
endif
if R2>r2
then
r2=R2
io2=-1
endif
if R3>r3
then
r3=R3
io3=-1
endif
if F1>f1
then
f1=F1
io1=1
endif
if F2>f2
then
f2=F2
io2=1
endif
if F3>f3
then
f3=F3
io3=1
endif
;active power with sign
a1=io1*A1
a2=io2*A2
a3=io3*A3
;print F %F1% %F2% %F3% f %f1% %f2% %f3% R %R1% %R2% %R3% r %r1% %r2% %r3% a %a1% %a2% %a3% A %A1% %A2% %A3% i %io1% %io2% %io3%

>B
=>sensor53 r
;Set teleperiod to 10sec
tper=10

>J
,"CALCULATED":{"io_L1":%io1%,"io_L2":%io2%,"io_L3":%io3%
,"L1ActivePowerSigned":%a1%,"L2ActivePowerSigned":%a2%,"L3ActivePowerSigned":%a3%}

>W
CALCULATED
L1 Active Energy: {m} %a1% W
L2 Active Energy: {m} %a2% W
L3 Active Energy: {m} %a3% W

>M 1
+1,3,M,0,9600,ORNO516,1,12,01030014,0103000E,01030010,01030012,0103010A,01030112,01030102,0103010C,01030114,01030104,0103010E,01030116,01030106,01030100,0103001E,01030020,01030022,0103001C
1,010304ffffffff@i0:1,GridFrequency,Hz,GridFrequency,2
1,010304ffffffff@i1:1,L1Voltage,V,L1Voltage,2
1,010304ffffffff@i2:1,L2Voltage,V,L2Voltage,2
1,010304ffffffff@i3:1,L3Voltage,V,L3Voltage,2
1,010304ffffffff@i4:1,L1ForwardActiveEnergy,kWh,L1ForwardActiveEnergy,2
1,010304ffffffff@i5:1,L1ReverseActiveEnergy,kWh,L1ReverseActiveEnergy,2
1,010304ffffffff@i6:1,L1TotalEnergy,kWh,L1TotalEnergy,2
1,010304ffffffff@i7:1,L2ForwardActiveEnergy,kWh,L2ForwardActiveEnergy,2
1,010304ffffffff@i8:1,L2ReverseActiveEnergy,kWh,L2ReverseActiveEnergy,2
1,010304ffffffff@i9:1,L2TotalEnergy,kWh,L2TotalEnergy,2
1,010304ffffffff@i10:1,L3ForwardActiveEnergy,kWh,L3ForwardActiveEnergy,2
1,010304ffffffff@i11:1,L3ReverseActiveEnergy,kWh,L3ReverseActiveEnergy,2
1,010304ffffffff@i12:1,L3TotalEnergy,kWh,L3TotalEnergy,2
1,010304ffffffff@i13:1,TotalEnergy,kWh,TotalEnergy,2
1,010304ffffffff@i14:0.001,L1ActivePower,W,L1ActivePower,2
1,010304ffffffff@i15:0.001,L2ActivePower,W,L2ActivePower,2
1,010304ffffffff@i16:0.001,L3ActivePower,W,L3ActivePower,2
1,010304ffffffff@i17:0.001,TotalActivePower,W,TotalActivePower,2
1,=m 15+16+17 @1,Power L1+L2+L3,W,CalcTotActPow,2
#
```

There are things in there that are not really needed, but if someone wants to play around with this, I thought deleting things is more easy than coming up with more ;-).

For someone who wants to use this, I've compiled my tasmota firmware with the following in ```tasmota/user_config_override.h```:

```
#define USE_PROMETHEUS
#define USE_ADC_VCC

#ifndef USE_SCRIPT
#define USE_SCRIPT  # adds about 17k flash size, variable ram size
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifndef USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_WEB_DISPLAY
#endif
#ifndef USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_JSON_EXPORT
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
// see bellow instructions to set the value N accordingly
//SML_MAX_VARS default 20
#define SML_MAX_VARS 35
//SML_BSIZ serial buffers default 48
#define SML_BSIZ 48
```

Thanks for the help and all the work done!

closed 2020-08-08 by bs-github

bs-github, 2020-08-08:
In case someone stumbles over this, I'm using https://github.com/hikhvar/mqtt2prometheus/ to get the data into prometheus and grafana.

gemu2015, 2020-08-08:
i forgot to mention, to use sml[x]
you must 
#define USE_SML_SCRIPT_CMD

mdiazgoncalves, 2021-09-22:
Hi @bs-github 

In your script,
1,=m 15+16+17 @1,Power L1+L2+L3,W,CalcTotActPow,2
should be 
1,=m 14+15+16 @1,Power L1+L2+L3,W,CalcTotActPow,2
for the sum of all active powers right?