gemu2015/Sonoff-Tasmota #25: How to connect CC1101 to ESP (Wemos)

iz3man, 2020-08-17:
I looked for any docs of this fork, but couldn't find any. Maybe I'm totally wrong here.
I was looking for a way to control a 868Mhz radio device. Ideally with tasmota and the CC1101 I have laying around here.
Is there a connection diagram somewhere how to do it, and how to configure the ESP to make this work?
Thanks!!

gemu2015, 2020-08-17:
@iz3man 
most of my fork is transferred to Tasmota so you may find the docs there.
the cc1101 moritz driver is nearly ready but it is meant for special LORA Devices named MAX (EQ3 Systems)
I will no longer support this because EQ3 did put the MAX system to end of life.
There are many different LORA systems on the market that all differ in the actual protocol used.
You may look into the moritz driver as a template how to adapt such a system to the CC1101 chip (SPI: MOSI,MISO,SCLK,CS )
in Tasmota there is also another driver using CC1101 (Jarolift Keeloq shutter)
currently not other devices are ported in Tasmota.

you may find hints in the sources of Rudolf Koenig (culfw)
who supports a range of 868Mhz devices with his driver.
i also took the moritz routines from there.

iz3man, 2020-08-17:
@gemu2015 Thanks a lot for your long answer. Being not able to code stuff, I may have to look somewhere else. ;)

TheChatty, 2020-11-14:
I'd also like a CC1101 driver in Tasmota. I suggest to implement it like SML does with ModBus (so also similar to culfw): It would then offer a device descriptor and several device commands. Also it would offer to just dump received packets to enable someone to create his own descriptor/commands set.

gemu2015, 2020-11-16:
i definitely will do no more development with cc1101. 
the only LORA devices which i would be interested (Homematic) have proprietary encrypted protocols which can not be used here.
When i have to replace my EQ3 devices i will choose Zigbee replacements. Tasmota now supports all kinds of Zigbee devices
Tasmota also supports RF transmitters and receivers for some protocols

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015