gemu2015/Sonoff-Tasmota #26: May I add TCS34725 to the main TASMOTA project

stefanbode, 2020-10-08:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
How far are you with bugfixing this driver? If everything is ok I can help you to integrate this driver into the TASMOTA main project. OK?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-10-08:
ok if you think this would be interesting i can make a pr for it.
i just updated the driver and tested with my hardware, it works fine.
i will provide a pr in the next few days.

i still have another interresting driver  for the eQ3 max devices (window sensors and thermostats).
however they recently were declared end of life. so it makes no sense to pr

gemu2015, 2020-10-09:
@stefanbode 
ok, made a branch ready for pr.
But i am not happy with the adafruit library because it works with delay() for integration time.
it has to be redesigned to circumvent this.
i am currently busy with other things. feel free to rewrite the code without delays and pr

stefanbode, 2020-10-11:
Ok, 
I will take a look at it. It could be that for the light measurement the delay is mandatory because of the way it gets measured. I already see this on other devices that took on low light up to 1 second

gemu2015, 2020-10-11:
there is the option to use an extra IRQ pin but i would prefer to use only SDA,SCL

gemu2015, 2020-10-11:
i use this in a battery operated luxmeter with a small OLED display.  In this use case i do not mind the delay. it works well.
in Tasmota however for general use we must get rid of it

hmjvaline, 2022-08-03:
Does tasmota support TCS34725 now? Or not?

gemu2015, 2022-08-03:
Tasmota does not support TCS34725. as already said i use this driver in a lux meter and it works well.
however since it is synchronous it blocks the tasmota loop for quite some time while getting values.
this is not allowed in Tasmota. You will have to rewrite the driver for asynchronous requests before it may be added to Tasmota.

dragospascale, 2024-01-29:
You can use TCS34725 async library to write the driver: [https://github.com/ysard/TCS34725]

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015