gemu2015/Sonoff-Tasmota #28: Arduino_ST7789: 135*240 TFT

jeroenvermeulen, 2020-12-28:
### PROBLEM DESCRIPTION
A 135*240 ST7789 TFT is not working correct, caused by hard-coded [_xstart](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L339) and [_ystart](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L346-L347) in  [Arduino_ST7789.cpp](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L347)

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [LILYGO® TTGO T-Display ESP32 WiFi And Bluetooth Module Development Board 1.14 Inch LCD](https://www.aliexpress.com/i/33050639690.html)
- [x] Tasmota binary firmware version number used: 9.2.90
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: `esptool.py --chip esp32 --port /dev/tty.usbserial-01F93A32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 build_output/firmware/bootloader_dout_40m.bin 0x8000 build_output/firmware/partitions.bin 0xe000 build_output/firmware/boot_app0.bin 0x10000 build_output/firmware/tasmota32.bin`
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:31:52 MQT: stat/ttgo-garage/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"928":"SSPI CS"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"960":"SSPI DC"},"GPIO17":{"0":"None"},"GPIO18":{"896":"SSPI SCLK"},"GPIO19":{"864":"SSPI MOSI"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"1024":"OLED Reset"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"}}}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:32:36 MQT: stat/ttgo-garage/STATUS = {"Status":{"Module":1,"DeviceName":"ttgo-garage","FriendlyName":["TTGO Garage","Tasmota2"],"Topic":"ttgo-garage","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:32:36 MQT: stat/ttgo-garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:09:31","StartupUTC":"2020-12-27T23:23:05","Sleep":50,"CfgHolder":4617,"BootCount":72,"BCResetTime":"2020-12-27T20:47:59","SaveCount":191}}
00:32:36 MQT: stat/ttgo-garage/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-27T23:51:58","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"449/699"}}
00:32:36 MQT: stat/ttgo-garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GRS-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
00:32:37 MQT: stat/ttgo-garage/STATUS4 = {"StatusMEM":{"ProgramSize":1408,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FFAC78F","001403A1","000000CF","010013C0","40000981","00004020","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,13,20,21,24,26,27,29","Sensors":"1,2,3,5,6,62"}}
00:32:37 MQT: stat/ttgo-garage/STATUS5 = {"StatusNET":{"Hostname":"ttgo-garage","IPAddress":"192.168.7.62","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.1","Mac":"10:52:1C:66:D8:B0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:32:37 MQT: stat/ttgo-garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.2","MqttPort":1883,"MqttClientMask":"ttgo-garage_%06X","MqttClient":"ttgo-garage_66D8B0","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:32:37 MQT: stat/ttgo-garage/STATUS7 = {"StatusTIM":{"UTC":"2020-12-27T23:32:37","Local":"2020-12-28T00:32:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
00:32:37 MQT: stat/ttgo-garage/STATUS10 = {"StatusSNS":{"Time":"2020-12-28T00:32:37"}}
00:32:37 MQT: stat/ttgo-garage/STATUS11 = {"StatusSTS":{"Time":"2020-12-28T00:32:37","Uptime":"0T00:09:32","UptimeSec":572,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"GRS-IOT","BSSId":"02:9F:C2:2D:4C:FD","Channel":9,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure in [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L46):
```
#define ST7789_TFTWIDTH 	135
#define ST7789_TFTHEIGHT 	240

#define ST7789_240x240_XSTART 53
![IMG_4501](https://user-images.githubusercontent.com/658024/103182007-ca8e6500-48a7-11eb-811f-07204df30c29.jpg)

#define ST7789_240x240_YSTART 40
```
Compile `tasmota32`, flash.
Enter in console:
```
BackLog  DisplayRotate 0; DisplayFont 1; DisplayCols 19; DisplayRows 20; DisplayMode 0
DisplayText [zl1c1]Hello, World!
```
### EXPECTED BEHAVIOUR
A black display with only the text `Hello World!`

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Next step would be to make the width, height, xstart and ystart configurable.
When I undo the two hard-coded overrides in  [Arduino_ST7789.cpp](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L347), and with my custom settings in [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L46) it works fine:
[image]

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-12-28:
ok did you load my last commit? you need only the driver file (Arduino_ST7789.cpp)
do not change anything in the driver, just report what is happening
you should only set display width in console to 135

the offset should now be set by the driver automatically to 40, 52

i will adjust things when it goes wrong

jeroenvermeulen, 2020-12-28:
I have tested 403dc9ba, I just build your development branch.
It is almost right!

Rotation 0 is one pixel too far to the left:
[image]

Rotation 1 is perfect:
[image]

Rotation 2 is perfect:
[image]

Rotation 3 is one pixel too far to the top:
[image]

gemu2015, 2020-12-28:
i am bit confused about where to set 52 or 53

there are 3 places to check

please iterate the 52,53 at these places and check the result.

we then can fix the remaining rotations.

if (width==240 && height==240) {
    _colstart = ST7789_240x240_XSTART;
    _rowstart = ST7789_240x240_YSTART;
  } else {
    _colstart = 40;
    _rowstart = 52;
  }

void Arduino_ST7789::setRotation(uint8_t m) {

  writecommand(ST7789_MADCTL);
  rotation = m % 4; // can't be higher than 3
  switch (rotation) {
   case 0:
     writedata(ST7789_MADCTL_MX | ST7789_MADCTL_MY | ST7789_MADCTL_RGB);

     _xstart = _colstart;
    // _ystart = _rowstart;
      if (_width==240 && _height==240) {
        _ystart = 80;
      } else {
        _ystart = 53;
      }
     break;
   case 1:
     writedata(ST7789_MADCTL_MY | ST7789_MADCTL_MV | ST7789_MADCTL_RGB);

     _ystart = _colstart;
    // _xstart = _rowstart;
     if (_width==240 && _height==240) {
       _xstart = 80;
     } else {
       _xstart = 53;
     }
     break;
  case 2:
     writedata(ST7789_MADCTL_RGB);

     _xstart = _colstart;
     _ystart = _rowstart;
     break;

   case 3:
     writedata(ST7789_MADCTL_MX | ST7789_MADCTL_MV | ST7789_MADCTL_RGB);

     _ystart = _colstart;
     _xstart = _rowstart;
     break;
  }
}

jeroenvermeulen, 2020-12-28:
I think my previous message was based on the firmware, I am sorry. 
Your dev branch seems to output the firmware in another dir.
Currently testing further.

gemu2015, 2020-12-28:
you can use original tasmota and only take the driver file from my fork

jeroenvermeulen, 2020-12-28:
Why don't you merge from upstream?

gemu2015, 2020-12-28:
i want to take control what i merge and keep my own project with some special settings.

gemu2015, 2020-12-28:
ok try my very last version.

i copied the rotation settings from a working driver.  it has a different coding of rotations but i think i reordered it right

jeroenvermeulen, 2020-12-28:
Maybe you can check if my version still works on 240*240. It is based on your version dd9567a

closed 2020-12-28 by gemu2015

jeroenvermeulen, 2020-12-28:
This are my test commands:
```
BackLog  DisplayRotate 0; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
BackLog  DisplayText [zs1l1c1]# Hello how are you? #; DisplayText [s1l29c1]1234567890123456789012

BackLog  DisplayRotate 1; DisplayWidth 240; DisplayHeight 135; DisplayFont 0; DisplayCols 40; DisplayRows 16;
BackLog  DisplayText [zs1l1c1]########## Hello how are you? ##########; DisplayText [s1l16c1]1234567890123456789012345678901234567890

BackLog  DisplayRotate 2; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
BackLog  DisplayText [zs1l1c1]# Hello how are you? #; DisplayText [s1l29c1]1234567890123456789012

BackLog  DisplayRotate 3; DisplayWidth 240; DisplayHeight 135; DisplayFont 0; DisplayCols 40; DisplayRows 17;
BackLog  DisplayText [zs1l1c1]########## Hello how are you? ##########; DisplayText [s1l16c1]1234567890123456789012345678901234567890
```

gemu2015, 2020-12-28:
are you sure about your modifications?  there is a discussion in a driver forum with a sample driver that definitely has also offsets of 53 ?

gemu2015, 2020-12-28:
https://github.com/Bodmer/TFT_eSPI/issues/379

jeroenvermeulen, 2020-12-28:
I have just built your `universal` branch, commit ebfb6a21

[image]

[image]

[image]

[image]

gemu2015, 2020-12-28:
probably you won't see 1 pixel offsets with text.
you should draw rectangles with full size to test like in the link

jeroenvermeulen, 2020-12-28:
You are right, I found 1 pixel off in Rotate 1 using a rectangle.
Working on a fix.

gemu2015, 2020-12-28:
2 rotates should have 53 and 2 52

jeroenvermeulen, 2020-12-28:
Test `universal` branch with PR #30 changes, Tasmota 9.1.0.2 
```
BackLog  DisplayRotate 0; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
DisplayText [zr135:240s2l2c2] Tasmota [s1l7c2] IPS ST7789V 1.14" [l9c2] 135 x 240 [l11c2] Rotate 0

BackLog  DisplayRotate 1; DisplayWidth 240; DisplayHeight 135; DisplayFont 0; DisplayCols 40; DisplayRows 16;
DisplayText [zr240:135s2l2c7] Tasmota [s1l7c2] IPS ST7789V 1.14" 135 x 240 [l9c2] Rotate 1

BackLog  DisplayRotate 2; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
DisplayText [zr135:240s2l2c2] Tasmota [s1l7c2] IPS ST7789V 1.14" [l9c2] 135 x 240 [l11c2] Rotate 2

BackLog  DisplayRotate 3; DisplayWidth 240; DisplayHeight 135; DisplayFont 0; DisplayCols 40; DisplayRows 17;
DisplayText [zr240:135s2l2c7] Tasmota [s1l7c2] IPS ST7789V 1.14" 135 x 240 [l9c2] Rotate 3
```
[image]

[image]

[image]

[image]

gemu2015, 2020-12-29:
ok well done,  had to change one define for my display but now it works for my display too. i also removed  some unnecessary code

please test gain and then you can make a pr to tasmota.

jeroenvermeulen, 2020-12-29:
Thanks, will do it this evening (in Belgium)

jeroenvermeulen, 2020-12-29:
@gemu2015 Can you maybe take a look at the compiler errors? https://github.com/gemu2015/Sonoff-Tasmota/pull/30#issuecomment-751878988
To repeat the warning you need to remove the `.cache` dir and compile.

jeroenvermeulen, 2020-12-29:
Never mind, I see you already fixed it

blakadder, 2020-12-29:
I have version using
```
backlight_pin: GPIO5
cs_pin: GPIO22
dc_pin: GPIO21
reset_pin: GPIO18
clk_pin: GPIO19
mosi_pin: GPIO23
miso_pin: GPIO25
TFT_WIDTH=240
TFT_HEIGHT=320
TFT_INVERSION_OFF     ; for normal colors
TFT_RGB_ORDER=TFT_BGR ; Colour order Blue-Green-Red
```
with an FT6336 Utouch chip

jeroenvermeulen, 2020-12-29:
@blakadder Can you test my PR https://github.com/arendst/Tasmota/pull/10310 ?

blakadder, 2021-01-01:
couldn't even get it to display anything

jeroenvermeulen, 2021-01-01:
@blakadder Do you have a link to the product you have?

blakadder, 2021-01-01:
https://blakadder.com/lanbon-L8/

gemu2015, 2021-01-01:
@blakadder 
did you use the very last dev version from today ?

and how did you connect your pins ?
i use SSPI on both devices i have and did not test with spi (spi should work also i assume).

then in console type display and check if the driver was found.

if found, set the width to 320 for your display

since your display is the max possible resolution of the chip the zero offsets should work.

at least you should be able to switch backlight On and Off and see partial pictures

blakadder, 2021-01-01:
pins are not changeable and are as follows
```
backlight_pin: GPIO5
cs_pin: GPIO22
dc_pin: GPIO21
reset_pin: GPIO18
clk_pin: GPIO19
mosi_pin: GPIO23
miso_pin: GPIO25
```

gemu2015, 2021-01-01:
i meant what was your pin function selection in Tasmota ?

i would suggest:

backlight_pin: GPIO5  Backlight
cs_pin: GPIO22 ST7789_CS
dc_pin: GPIO21  ST7789_DC
reset_pin: GPIO18 OLED_Reset
clk_pin: GPIO19 SSPI_SCLK
mosi_pin: GPIO23 SSPI_MOSI
miso_pin: GPIO25 not needed for display

blakadder, 2021-01-01:
ok got it working with that config... looks fine out of the box

gemu2015, 2021-01-01:
nice to hear

gemu2015, 2021-01-01:
to get the touch screen working you need to (only with scripting)
#define USE_FT5206
#define USE_TOUCH_BUTTONS

you then in a script can define touch buttons directly linked with relays or virtual buttons

blakadder, 2021-01-01:
one more hiccup: my display has inverted colors

gemu2015, 2021-01-02:
this driver inverts your display  (if width or height == 320)

[Arduino_ST7789.cpp.zip](https://github.com/gemu2015/Sonoff-Tasmota/files/5759897/Arduino_ST7789.cpp.zip)

virtualizer-ham, 2021-01-07:
Hello gemu2015, respect for your work, just came along since I bought the same device and am trying to get something drawn/written to this nice little board with display!
But I'm going nearly crazy, I already checked out your Repo and compiled it, checked also the dev-repo from tasmota / Theo Arends out and compiled it (9.2.0.3), tried to use the nightly esp32 builds provided by Theo Arends and tried also nearly each combination of gpio setup for this device, only thing that works: Switching backlight on/off !

I don't know what I'm doing wrong, but trying to write out some text, also using your examples above, ends up with no output at all.
Also playing with the fore- and background colours makes no difference.

When the device freshly powers up you see some kind of initialization (white stripes going down), then it goes to black screen and stays there. Playing with DisplayDimmer only makes the background look brighter or darker, that's it.

Actual configuration:
GPIO4: Backlight
GPIO5: ST7789CS
GPIO16: ST7789DC
GPIO18: SSPI SCLK
GPIO19: SSPI MOSI
GPIO23: OLED Reset

Program Version 9.2.0.3(display)
Core/SDK Version 1_0_5_RC4/v3.3.4-286-gb4c075169

But trying out all other combinations with SPI, SSPI etc. only lead to no recognition of the display.
Please shed some light on me...

Greetings
joe

gemu2015, 2021-01-07:
what hardware do you have ?
if you have the 135x240 board  you should ask @jeroenvermeulen

virtualizer-ham, 2021-01-07:
Hi, I also have the TTGO T-Display Dev-Board with 1.14" TFT. OK, so now my question goes to jeroenvermeulen :-)

jeroenvermeulen, 2021-01-07:
These are my notes:
```
GPIO19 = SSPI MOSI/SID
GPIO18 = SSPI SCLK
GPIO5  = ST7789 CS
GPIO16 = ST7789 DC
GPIO23 = OLED Reset
GPIO4  = Backlight
```
Template for Software SPI:
```
{"NAME":"TTGO T-Display v1.1","GPIO":[1,1,1,1,992,6592,1,1,1,1,1,1,6624,1,896,864,0,640,608,1024,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
```
To test I enter in console:
```
BackLog  DisplayModel 12; DisplayRotate 0; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
DisplayText [zr135:240s2C63488l2c2] Tasmota [s1C31l7c2] IPS ST7789V 1.14" [l9c2] 135 x 240 [l11c2] Rotate 0
```
Does this help?

virtualizer-ham, 2021-01-07:
Hi Jeroenvermeulen, thank you for your very fast reply also! It seems, the Template did the trick!!! I was using before the standard ESP32DEV template and had the GPIO's configured same like you (and all other combinations...), without success, now I activated the template and copy/pasted your test strings: That's it! Works!

Thank you so much! I was already going crazy about no succes even doing (nearly) everything like stated in this thread and all others I found on the Net.

Have a nice evening and thank you again for your help, also to Gemu2015!
Greetings
Joe

jeroenvermeulen, 2021-01-07:
You're welcome.
To make this help complete:

I build `tasmota32`, this should be possible using the current [development](https://github.com/arendst/Tasmota/tree/development) branch.
You need to have [PlatformIO Core](https://docs.platformio.org/en/latest/core/installation.html) and [esptool](https://github.com/espressif/esptool) installed.

I use these settings in `tasmota/user_config_override.h`:
```
#define USE_DISPLAY
#define USE_DISPLAY_ST7789
#define USE_SPI
#define SHOW_SPLASH
```
Build command:
```
platformio run -e tasmota32
```
Flash command:
```
BINDIR="build_output/firmware"
SERIAL="/dev/tty.usbserial-01F93A32"
wget https://github.com/arendst/Tasmota/raw/firmware/firmware/tasmota32/ESP32_needed_files/bootloader_dout_40m.bin -O $BINDIR/bootloader_dout_40m.bin
wget https://github.com/arendst/Tasmota/raw/firmware/firmware/tasmota32/ESP32_needed_files/partitions.bin -O $BINDIR/partitions.bin
wget https://github.com/arendst/Tasmota/raw/firmware/firmware/tasmota32/ESP32_needed_files/boot_app0.bin -O $BINDIR/boot_app0.bin
esptool.py \
  --chip esp32 \
  --port $SERIAL \
  --baud 921600 \
  --before default_reset \
  --after hard_reset \
  write_flash \
  -z \
  --flash_mode dout \
  --flash_freq 40m \
  --flash_size detect \
  0x1000 $BINDIR/bootloader_dout_40m.bin \
  0x8000 $BINDIR/partitions.bin \
  0xe000 $BINDIR/boot_app0.bin \
  0x10000 $BINDIR/tasmota32.bin
```

virtualizer-ham, 2021-01-07:
OK, thank you very much for this information, PlatformIO is already setup, GIT and ESPTOOL also, but since I'm quite new to this I was already wondering how to not build the whole platforms and derivates, flashing was no problem, using Tasmota already some time longer, but selfcompiling was the challenge...will try your settings!
Thank's a lot!

no-Legend, 2021-12-20:
Hi together is there still the need to compile by my self?
Or is the Ttgo display already in the stable branch or the development branch?

Cheers Robert

jeroenvermeulen, 2021-12-20:
On https://tasmota.github.io/docs/Firmware-Builds/#available-features-and-sensors you can search for `USE_DISPLAY_ST7789`.
You will see it is not included by default in any build.

There is a [Universal Display Driver](https://tasmota.github.io/docs/Displays/#universal-display-driver) [here](https://github.com/arendst/Tasmota/blob/development/tasmota/displaydesc/ST7789_display.ini). 
The last time I have worked with a universal display driver it did not work as good as a compiled driver.

no-Legend, 2021-12-20:
So at the moment I got some information on the display.
Could it be that I nee first to Set the Displaymode to 0 (DisplayText)?

jeroenvermeulen, 2021-12-20:
Can you send a photo?

no-Legend, 2021-12-20:
I just figured out, that I need to to configure a button.
An after I pushed the button, the display shows something. A second push let the display turn off. Is this behaviour correct?

jeroenvermeulen, 2021-12-20:
Yes, kind of.
You can also use the Tasmota web interface.
When there is a pin configured for the Display Background / Led, the last 'Toggle' button can be used to switch the display on/off. If you wish you can connect a physical button to this toggle.

no-Legend, 2021-12-21:
I did not understand the button behaviour. 
When I configure a button in "Configure Module" should it normaly appears in on the first page of the webgui?
It looks like every button number I choose is for turning the display on or off.
I also didn't do any rules.

jeroenvermeulen, 2021-12-21:
Can you maybe try to **not** configure a pin for `Backlight`?
If you need a button to switch something else you can add a Relay with the same number.

PR #29 (2020-12-28): Fixed 135*240 TFT
## Description:

**Related issue (if applicable):** fixes #28

## Checklist:
  - [x] The pull request is done against the latest ~dev~ universal branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.2.1.0
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Your PR cannot be mer [... cut]
PR #30 (2020-12-29): Support 135*240 and 240*240 TFT
Cleaned up code, defines instead of overrides.

## Description:

**Related issue (if applicable):** fixes #28

## Checklist:
  - [x] The pull request is done against the latest ~dev~ universal branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.2.1.0
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOT [... cut]