gemu2015/Sonoff-Tasmota #31: M5 core2 dont use 8MB PSRAM

Jason2866, 2021-01-01:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Checking the possibilty if implementing https://github.com/espressif/esp-idf/tree/master/examples/system/himem is possible to use the upper 4MByte of the PSRAM
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-01-01:
esp32 supports only 4 mb mapped into memory space. if you want to use the remaining 4mb you must use bank switching. This makes only sense for very special cases like e.g. memory intensive matrix calculations or bitmap manipulations.

currently i don't see a use case in tasmota for that.

i personally use HTTPS and TLS emails which uses a lot of sram memory and it would be of help if this ram could be allocated in PSRAM. this would require the libraries to detect PSRAM and use it when available. This is in the ESP32 SDK however.

in all applications i coded (e.g. microphone recording) i use PSRAM when possible.

Jason2866, 2021-01-01:
Understand. Thx for the answer.

closed 2021-01-01 by Jason2866