gemu2015/Sonoff-Tasmota #33: Hello,

kurczak2, 2021-04-09:
Hello,

> > 1. do you still get exceptions ?
> 
> No, not with `+1,3,M,0,9600,ORNO516,1,12,...`

I would be able to get back to smaller values if you want me to test something.
But I do not have to solve this issue because I get the data that I need reliably now.

> > 3. you dont need to get vars via JSON. you may use sml[x] to access sml descriptor lines
> >    anyhow your syntax is wrong, this is the right one
> >    oldFL1=ORNO516#L1ForwardActiveEnergy
> 
> Thanks, I've got values in the script now.

I could not get the values with sml[x] and the delta definitions did not work for me either, but those drawbacks are no issue for me and I'm quite happy with the following version of my script for the ORNO or-we-516 smart meter:
```
>D
;now
F1=0
F2=0
F3=0
R1=0
R2=0
R3=0
;before
f1=0
f2=0
f3=0
r1=0
r2=0
r3=0
;in or out
io1=1
io2=1
io3=1
;active power
A1=0
A2=0
A3=0
a1=0
a2=0
a3=0
;Fill vars with content on teleperiod
>T
F1=ORNO516#L1ForwardActiveEnergy
F2=ORNO516#L2ForwardActiveEnergy
F3=ORNO516#L3ForwardActiveEnergy
R1=ORNO516#L1ReverseActiveEnergy
R2=ORNO516#L2ReverseActiveEnergy
R3=ORNO516#L3ReverseActiveEnergy
A1=ORNO516#L1ActivePower
A2=ORNO516#L2ActivePower
A3=ORNO516#L3ActivePower
if R1>r1
then
r1=R1
io1=-1
endif
if R2>r2
then
r2=R2
io2=-1
endif
if R3>r3
then
r3=R3
io3=-1
endif
if F1>f1
then
f1=F1
io1=1
endif
if F2>f2
then
f2=F2
io2=1
endif
if F3>f3
then
f3=F3
io3=1
endif
;active power with sign
a1=io1*A1
a2=io2*A2
a3=io3*A3
;print F %F1% %F2% %F3% f %f1% %f2% %f3% R %R1% %R2% %R3% r %r1% %r2% %r3% a %a1% %a2% %a3% A %A1% %A2% %A3% i %io1% %io2% %io3%

>B
=>sensor53 r
;Set teleperiod to 10sec
tper=10

>J
,"CALCULATED":{"io_L1":%io1%,"io_L2":%io2%,"io_L3":%io3%
,"L1ActivePowerSigned":%a1%,"L2ActivePowerSigned":%a2%,"L3ActivePowerSigned":%a3%}

>W
CALCULATED
L1 Active Energy: {m} %a1% W
L2 Active Energy: {m} %a2% W
L3 Active Energy: {m} %a3% W

>M 1
+1,3,M,0,9600,ORNO516,1,12,01030014,0103000E,01030010,01030012,0103010A,01030112,01030102,0103010C,01030114,01030104,0103010E,01030116,01030106,01030100,0103001E,01030020,01030022,0103001C
1,010304ffffffff@i0:1,GridFrequency,Hz,GridFrequency,2
1,010304ffffffff@i1:1,L1Voltage,V,L1Voltage,2
1,010304ffffffff@i2:1,L2Voltage,V,L2Voltage,2
1,010304ffffffff@i3:1,L3Voltage,V,L3Voltage,2
1,010304ffffffff@i4:1,L1ForwardActiveEnergy,kWh,L1ForwardActiveEnergy,2
1,010304ffffffff@i5:1,L1ReverseActiveEnergy,kWh,L1ReverseActiveEnergy,2
1,010304ffffffff@i6:1,L1TotalEnergy,kWh,L1TotalEnergy,2
1,010304ffffffff@i7:1,L2ForwardActiveEnergy,kWh,L2ForwardActiveEnergy,2
1,010304ffffffff@i8:1,L2ReverseActiveEnergy,kWh,L2ReverseActiveEnergy,2
1,010304ffffffff@i9:1,L2TotalEnergy,kWh,L2TotalEnergy,2
1,010304ffffffff@i10:1,L3ForwardActiveEnergy,kWh,L3ForwardActiveEnergy,2
1,010304ffffffff@i11:1,L3ReverseActiveEnergy,kWh,L3ReverseActiveEnergy,2
1,010304ffffffff@i12:1,L3TotalEnergy,kWh,L3TotalEnergy,2
1,010304ffffffff@i13:1,TotalEnergy,kWh,TotalEnergy,2
1,010304ffffffff@i14:0.001,L1ActivePower,W,L1ActivePower,2
1,010304ffffffff@i15:0.001,L2ActivePower,W,L2ActivePower,2
1,010304ffffffff@i16:0.001,L3ActivePower,W,L3ActivePower,2
1,010304ffffffff@i17:0.001,TotalActivePower,W,TotalActivePower,2
1,=m 15+16+17 @1,Power L1+L2+L3,W,CalcTotActPow,2
#
```

There are things in there that are not really needed, but if someone wants to play around with this, I thought deleting things is more easy than coming up with more ;-).

For someone who wants to use this, I've compiled my tasmota firmware with the following in ```tasmota/user_config_override.h```:

```
#define USE_PROMETHEUS
#define USE_ADC_VCC

#ifndef USE_SCRIPT
#define USE_SCRIPT  # adds about 17k flash size, variable ram size
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifndef USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_WEB_DISPLAY
#endif
#ifndef USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_JSON_EXPORT
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
// see bellow instructions to set the value N accordingly
//SML_MAX_VARS default 20
#define SML_MAX_VARS 35
//SML_BSIZ serial buffers default 48
#define SML_BSIZ 48
```

Thanks for the help and all the work done!

_Originally posted by @bs-github in https://github.com/gemu2015/Sonoff-Tasmota/issues/24#issuecomment-670924323_

kurczak2, 2021-04-09:
Hello, sorry to bother you. I see you've run ORNO516 with Tasmota. I bought this energy meter hoping to somehow read it with a Tasmota. Unfortunately, I'm a beginner and compiling a script is too much for me. You might be interested in including instructions for this Orno 516 in the Tasmota documentation - https://tasmota.github.io/docs/Smart-Meter-Interface/. It would help many novice users like me. Or maybe you can share the compiled file? Thank you in advance for the enormous amount of work and time invested in the development of this software. Regards Mirek

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015