gemu2015/Sonoff-Tasmota #35: abs() is part of USE_ANGLE_FUNC

camel1cz, 2021-09-12:
Hi,

I spend lot of time to find out abs() function is part of USE_ANGLE_FUNC and it's not mentioned in the docs.

I bet, it's just wrong placement of #endif in xdrv_10_scripter.ino around line 1759?

gemu2015, 2021-09-18:
this has been fixed several month ago. you seem to use an outdated version

camel1cz, 2021-09-19:
Ah, sorry - it's not in stable tasmota (9.5.0) yet and I overlooked it in devel branch.

Btw. thanks for your great work. I switched from bare MCU to tasmota since I discovered your scripting language.
Just one note - looking for errors in code is pain - it would be nice to have build with more debug information to make evelopment of larger projects easier.
Also some script to upload code into tasmota with comments stripping, versioning and replacement of variable names would be cool. I have such script for my personal use, but it's just quick and dirty hack.

One more time: thank you!

gemu2015, 2021-09-19:
This is already there! Just look at tasmota docs scripting language for script editor

gemu2015, 2021-09-19:
https://www.dropbox.com/sh/0us18ohui4c3k82/AACcVmpZ4AfpdrWE_MPFGmbma?dl=0

camel1cz, 2021-09-19:
Thank you, will give it a try... I saw it but wasn't aware of the features.
Btw. does your cross compiler support also linux? Would be nice to have native app w/o fiddling with emulators.

gemu2015, 2021-09-19:
Sorry my crosscompiler only supports windows

closed 2021-12-10 by camel1cz