gemu2015/Sonoff-Tasmota #38: Add Kamstrup protocol to SML

TheChatty, 2022-09-05:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
Native support for [Kamstrup protocol](https://github.com/nabovarme/MeterLogger/wiki/Kamstrup-Protocol) as it requires CRC calc and a few char escapings.

**Describe the solution you'd like**  
Like o is for OBIS I suggest k for Kamstrup.

**Describe alternatives you've considered**  
Non-Tasmota python script. I'd rather avoid this.

**Additional context**  
There is plenty of python source code: [1](https://github.com/bsdphk/PyKamstrup/blob/master/kamstrup.py) [2](https://programtalk.com/vs2/?source=python/8914/conpot/conpot/protocols/kamstrup/meter_protocol/decoder_382.py) [3](https://community.home-assistant.io/t/running-python-script-in-hass/326876)

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-09-06:
normally i only implement drivers for devices i personally use.
however at least i would need a hexdump of meter values to evaluate how much effort it would be.
since berry is very similar to python you may also consider to try that path.

TheChatty, 2022-09-09:
@gemu2015 I added the infos to the issue description.

gemu2015, 2022-09-09:
ok i added kamstrup, give it a try i can not test since i don't have such a meter,
at least the transmission is correct
the decoder code is kstr
it works similar to modbus and you must also use indexes i0 ...

\>M 1
+1,3,k,0,9600,KS,1,10,3F1001003C
1,3F1001003Ckstr@i0:1,Volts,V,VP,2
#

TheChatty, 2022-09-09:
Incredible but it will take two weeks till I can test. I‘ll report.

TheChatty, 2022-10-01:
I eventually found time to test your code which was obviously already merged upstream.

I configured the meter as:
```
+1,3,kN2,0,1200,KSMC403,1,10,3F1001003C
1,3F10003Ckstr@i0:1,Heat Energy (E1),MWh,HeatEnergyE1,3
```
Because it runs ar 1200 8N2 and 3C returns total heat energy used in MWh at a precision of 3. 

The request is created in the correct manner and answered from the meter like this:
```
> 80 3F 10 01 00 3C B2 5F 0D
< 40 3F 10 00 3C 03 04 43 00 00 D9 2B AC A6 0D
```

But console shows no receiption / webUI shows "null".

gemu2015, 2022-10-02:
try new version in my repo

TheChatty, 2022-10-02:
1. That worked to some extend. I can now define all measures and they are requested one after the other but only the first requested measure is decoded correctly (e.g. actual value shown in UI).
2. I still only see transmits in console with `define MODBUS_DEBUG` - no responses even in the case of the first telegram.
3. Some values represent a date. The number 221001 (YYMMDD) should be shown as 01.10.22... is this achievable?

Current meter definition is:
```
>D
>B
=>sensor53 r
>M 1
+1,3,kN2,0,1200,KSMC403,1,10,3F1001003C,3F10010056,3F10010057,3F10010059,3F1001004A,3F1001008A
1,3F10003Ckstr@i0:1000,Heat Energy (E1),MWh,HeatEnergyE1,3
1,3F100056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
1,3F100057kstr@i0:100,Rücklauftemp.,°C,AftTemp,2
1,3F100059kstr@i0:100,Temp.diff.,°C,DifTemp,2
1,3F10004Akstr@i0:1,Fließgeschw.,l/h,Flow,2
1,3F10008Akstr@i0:1,Datum max. Fluss,d,maxFlowDate
#
```

All values remain zero except the one from the first telegram.

gemu2015, 2022-10-02:
You must increment the index
i0, i1 etc like with modbus

give an example response for time

TheChatty, 2022-10-02:
`403f10008a30040000035ee50a580d` decodes to 220901 meaning 01.09.22.

TheChatty, 2022-10-02:
I found that I can request multiple values at once. But `skip N bytes`does not work whereas enough xx work. Following entries should be similar but are not.
```
+1,3,kN2,0,1200,KSMC403,1,10,3F1002003C0056
1,3F10003Ckstr@i0:1000,Heat Energy (E1),MWh,HeatEnergyE1,3
1,3F10x90056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
1,3F10xxxxxxxxxxxxxxxxxx0056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
```

gemu2015, 2022-10-02:
1. the decoder skip with xn requires a not number delimiter, so x90056kstr would assume you would skip 90056 chars while x11kstr would work
2. the serial buffer is limited to SML_BSIZ with default = 48 bytes, you may increase this value in user config override to fit more data

TheChatty, 2022-10-02:
I just noticed those "escape bytes" make the fixed number skip algo somewhat useless.

gemu2015, 2022-10-02:
no, what you see is the raw received value, the decoder gets the corrected data without escapes

TheChatty, 2022-10-02:
So it should work already? I will try.

TheChatty, 2022-10-02:
I raised **SML_BSIZ to 90** and are now able to request at least 6 values. A few examples would be:
```
> 80 3f 10 1b f9 00 3c 00 56 00 57 00 59 00 4a 00 44 03 1b bf 0d
< 403F10003C0304430000D949005625024217C400572502420AC000592602421BF204004A290200000700442804420002ECA858770D
< 403F10003C0304430000D949005625024217C100572502420A4B00592602421BF276004A290200000900442804420002ECA85F690D
< 403F10003C0304430000D949005625024217CA00572502420A4100592602421BF289004A290200000900442804420002ECA875710D
        <--------09------><-----07-----><-----07-----><-----07--^^---><-----07-----><--------09------><CR>
```

But the following meter definition only succeeds to decode the first four values or none (with above examples):
```
>D
>B
=>sensor53 r
>M 1
+1,3,kN2,0,1200,KSMC403,1,10,3F1006003C005600570059004A0044
1,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
1,3F10x08xx0056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
1,3F10x15xx0057kstr@i0:100,Rücklauftemp.,°C,AftTemp,2
1,3F10x22xx0059kstr@i0:100,Temp.diff.,°C,DifTemp,2
1,3F10x29xx004Akstr@i1:1,Fließgeschw.,l/h,Flow,0
1,3F10x36xx0044kstr@i1:100,Volumenstrom,m³,Volume,2
#
```

TheChatty, 2022-10-02:
Now all that is missing are the date values. Shall I update the docs already (without date values atm)?

gemu2015, 2022-10-02:
i implemented the # indicator for kamstrup as a date string. (see repo)
like this

1,3F10003Ckstr@#,Datum,,date,0

however since the driver supports only one string per meter you may only define one  such a line per meter

yes update the docs together with an example, i will male a pr in the next few days

TheChatty, 2022-10-03:
How to specify index in case of date? "@i0:#" doesn't work.

gemu2015, 2022-10-03:
no index with this decoder line

TheChatty, 2022-10-03:
It will always be applied to first telegram or on every telegram?

gemu2015, 2022-10-03:
yes to any, but the decoder will select the right one, because of the register address given.

in theory  we would not need an index in  kamstrup decoder since the reply repeats the register address which is not the case on MODBUS.

i guess the decoder will work also whitout any index because the register selects the decoder line.

gemu2015, 2022-10-03:
ok, then leave the indexes and only for the string use without index.

TheChatty, 2022-10-03:
There is still some instability... now the formerly working script decodes nothing (all shown as 0)... `sensor53 d1` shows data is sent and received.

gemu2015, 2022-10-03:
post the script

TheChatty, 2022-10-03:
```
>D
>B
=>sensor53 r
>M 1
+1,3,kN2,0,1200,KSMC403,1,15,3F1007003C005600570059004A0044007B
1,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
1,3F10x08xx0056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
1,3F10x15xx0057kstr@i0:100,Rücklauftemp.,°C,AftTemp,2
1,3F10x22xx0059kstr@i0:100,Temp.diff.,°C,DifTemp,2
1,3F10x29xx004Akstr@i0:1,Fließgeschw.,l/h,Flow,0
1,3F10x36xx0044kstr@i0:100,Volumenstrom,m³,Volume,2
1,3F10x45xx007Bkstr@#,max. Fluss am,,MaxFlowDate,0
#
```

TheChatty, 2022-10-03:
Changing telegram to read only four values it works (3F1004003C005600570059).

gemu2015, 2022-10-03:
i did put a hexdump in normal mode, (not dump mode) to show the decode string with removed escapes.
see repo

TheChatty, 2022-10-03:
I split the telegram into two pieces. So sometimes the first, sometimes the second is working.

That point of code only gets called when decoding works. Thus the error must be happening before.

gemu2015, 2022-10-03:
ok seems to need escaping BEFORE checksum is checked

try repo

TheChatty, 2022-10-03:
All CRCs are verified ok. All values are interpreted ok. Thanks a million!
[image]

TheChatty, 2022-11-21:
All done.

closed 2022-11-21 by TheChatty