gemu2015/Sonoff-Tasmota #39: Parallel display support in uniDisplay driver

blakadder, 2022-09-19:
Support for parallel display to be used with LVGL via uniDisplay driver

With the proliferation of ESP32-S3 chinese manufacturers have started making display boards with parallel display interface

f.e. https://github.com/HASwitchPlate/openHASP/discussions/383 or https://www.alibaba.com/product-detail/Good-heat-dissipation-IPS-LCD-Color_1600148110318.html

- 

-->

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  

**Describe the solution you'd like**  

**Describe alternatives you've considered**  

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-09-19:
looks very promising!

gemu2015, 2022-09-23:
i had a parallel ili9341 laying around. (https://www.amazon.de/ARCELI-ILI9341-240X320-Display-Arduino/dp/B09DYDYSL3/ref=asc_df_B09DYDYSL3/?tag=googshopde-21&linkCode=df0&hvadid=579320962303&hvpos=&hvnetw=g&hvrand=7919256257140488711&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9062618&hvtargid=pla-1440065907977&psc=1&th=1&psc=1)

i added an esp32-s3 WROOM and now i am porting the universal driver.
will have options for 8 and 16 bit interface

will take a few days since i am short of time currently.

gemu2015, 2022-09-30:
@blakadder 
parallel support in Universal display driver is ready including simple resistive touch (DMA not yet) 
for above display. (esp32-s3 only)

needed a large number of changes, must be testet for regressions in other displays.
will not do a pr until better tested.

if you get hands on a 16 bit parallel display you may give it a try, should work also,

files affected:
uDisplay.cpp
uDisplay.h
renderer.cpp
renderer.h
xdrv_55_touch.ino
xdsp_17_universal.ini

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015