gemu2015/Sonoff-Tasmota #40: Support ESP-IR

TheChatty, 2022-10-03:
SML works fine with some ESP8266 and D0 transceiver on hardware and software serial.
[image]

Now I'd like to change to ESP-IR (integrated board with emitter and receiver):
[image]

- First solution configured to `Serial TX/RX`:  transmitter is always off and only blinking on `SerialSend ...`.
- Second solution configured to `Serial TX/RX`: transmitter is always on and blinking off on `SerialSend ...`. Both diodes are wired to GPIO4/14 thus software serial.

I guess there is an [inverting Schmitt trigger](https://de.wikipedia.org/wiki/Schmitt-Trigger#Invertierender_Schmitt-Trigger) in the [D0 head](https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-ttl-ausgang). For obvious reasons that cannot be easily integrated into ESP-IR.
Is there an easy solution to invert their behavior for SML? (e.g. change all lines in [TasmotaSerial.cpp](https://github.com/arendst/Tasmota/blob/development/lib/default/TasmotaSerial-3.5.0/src/TasmotaSerial.cpp) containg `digitalRead` and `digitalWrite` in a clever way)

gemu2015, 2022-10-04:
this took me 15 minutes to make a compile option to invert software serial, would be some more effort to make that configurable with a command

#define TS_SWS_INVERT

see repo

TheChatty, 2022-10-04:
You [forgot one digitalWrite](https://github.com/gemu2015/Sonoff-Tasmota/blob/universal/lib/default/TasmotaSerial-3.5.0/src/TasmotaSerial.cpp#L82).

But even with this fixed the transmitter is always on and flashing when sending. What's happening?

gemu2015, 2022-10-04:
this is only working for software serial , hardware serial can not be inverted.

the pin i forgot should set the trx GPIO to the right level.

it should definitely work for sending.

for receiving i forgot the level change interrupt active transition, i will check that later

TheChatty, 2022-10-04:
The problem was `#define`s from `user_config_override.h` are not incorporated into (all?) lib(s).

Now enforcing that `#define  TS_SWS_INVERT` it works visually.

gemu2015, 2022-10-04:
yes i forgot that we are in a library.
i fixed the irq , try receiving

TheChatty, 2022-10-04:
You need to check it in ;-)

TheChatty, 2022-10-04:
Sending strings from ESP-IR with up to 14.400 baud works. Starting from 19.200 things get messy. The meter I have which requires sending is only at 1.200 baud thus ok.

Receiving string with ESP-IR does not work even at 1.200 baud. Do you have any ideas how to analyze this?

gemu2015, 2022-10-05:
first check if the non inverse version works both for sending and receiving.
ich checked the code again and can't find a reason why it should not work.

TheChatty, 2022-10-07:
Using **original** Tasmota on both ends leads to this:

Both modules **configured to IRsend/IRrecv**

- ESP-IR transmitter **off** when idle
- D0 transmitter **lit** when idle

`IRsend {"Protocol":"NEC","Bits":32,"Data":0x1248FEB7}` is received as `"IrReceived":{"Protocol":"UNKNOWN","Bits":32,"Hash":"0x1248FEB7"}` **in both directions**.
This library seems to handle inverted diodes transparently.

Both modules **configured to TCP Tx/Rx with `TCPBaudrate 1200`**

- ESP-IR transmitter **lit** when idle
- D0 transmitter **off** when idle

Data sent from D0 to ESP-IR is **not received**. 
Data sent from ESP-IR to D0 is **kind of received**. Long strings of the same character are "**translated**" into different characters (with some garbage).
```
0x65 -> 0x4D
0x67 -> 0x4C
0x6B -> 0x4A
0x79 -> 0x43
0x31 -> 0x67
0x33 -> 0x66
0x35 -> 0x65
0x3F -> 0x60
0x5F -> 0x50
```

TheChatty, 2022-10-09:
@s-hadinger Are you able to to give some insight why a modded TasmotaSerial.cpp installed on ESP-IR is working for sending but not for receiving?Maybe you've got some idea where to continue investigation?

s-hadinger, 2022-10-09:
This looks like a duplicate from https://github.com/arendst/Tasmota/discussions/16447

TheChatty, 2022-10-09:
The discussion at the main repo got stalled so sorry for the duplicate. Once solved I'll close it with a link to the solution.

The IR communication both threads are dealing with is about reading certain energy meters via standardized D0 interface. It's not about signals from TV remotes and so on.

As I said above the D0 IR head probably sports a Schmitt trigger which inverts plain TTL signals thus the head is **working as is** on hardware and software serial from Tasmota or on a FTDI.

The factory-made ESP-IR has its diodes connected to GPIO 4 and 14 thus software serial. As try and error has shown switching HIGH and LOW **did work for sending**. This makes me wonder what is missing for receiving.

gemu2015, 2022-10-10:
@TheChatty 
sorry i did not realize what you are trying to do.
i just thought you wanted an inverted serial driver to save a hardware inverter.
but the esp-ir you are using is an ir-remote controller which operates with modulated ir light (38kHz). Thus the ir receiver is not just a photodiode but a receiver IC.
it cannot operate with energy meters at all.

TheChatty, 2022-10-10:
The receiving IC is a [VS838](https://datasheetspdf.com/pdf-file/796250/ETC/VS838/1). Thus it would need to be desoldered and replaced with a [SFH309FA](https://docs.rs-online.com/0d92/0900766b814f56a7.pdf), for instance?

gemu2015, 2022-10-10:
yes that would work. if you stay with inverted logic you should wire the photodiode so that the logic level is zero when no light is present.

TheChatty, 2022-10-10:
I'd try to simply connect SFH309FA to the same pin where VS838 is connected at the moment. I assume it's directly hooked up to GPIO14 from ESP.

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015