gemu2015/Sonoff-Tasmota #41: Order of meter descriptions defines whether meter is (not) working

TheChatty, 2022-11-04:
### PROBLEM DESCRIPTION
I have got two meters requiring RX/TX and one meter requiring only RX - all connected at the same time.

Whether SML script works depends on the order of their meter descriptions.

I found one working order - all other orders have at least one meter showing only 0's.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-12F in development board
- [ ] Tasmota binary firmware version number used: 12.2.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

Only wokring script:
```
>D
>B
=>sensor53 r
>M 3
+1,12,o,0,9600,Strom
1,1-0:1.7.0*255(@1,P_in,W,P_in,2
1,1-0:21.7.0*255(@1,L1,W,L1,2
1,1-0:41.7.0*255(@1,L2,W,L2,2
1,1-0:61.7.0*255(@1,L3,W,L3,2
1,1-0:1.8.0*255(@1,E_in,kWh,E_in,3
1,1-0:2.8.0*255(@1,E_out,kWh,E_out,3
1,=h<hr/>
+2,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
2,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
2,F7031Cx06SSss@i0:10,Vorlauftemp.,°C,Temp_Flow,1
2,F7031Cx14SSss@i0:10,Rücklauftemp.,°C,Temp_Return,1
2,F7031Cx26SSss@i0:10,Speichertemp.,°C,Temp_Vessel,1
2,F70308xxxxUUuu@i1:0.1,Messwertm3-h,l/h,Metric_M3H,0
2,F70304UUuu@i2:1,StellsignalRk1,%,CtrlSig_RK1,0
2,=h<hr/>
+3,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
3,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
3,3F10x29xx0044kstr@i0:100,Volumenstrom,m³,Volume,2
3,3F10x08xx0056kstr@i0:100,Vorlauftemp.,°C,PreTemp,2
3,3F10x15xx0057kstr@i0:100,Rücklauftemp.,°C,AftTemp,2
3,3F10x22xx004Akstr@i0:1,Fließgeschw.,l/h,Flow,0
#
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
- Flash Tasmota
- Configure & activate script
- Notice all meters are working
- Change order of meter descriptions
- Notice some non-working meters

### EXPECTED BEHAVIOUR
All meters are working no matter in which order they appear in the script.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TheChatty, 2022-11-05:
Some examples:

Meter 1 delivers no values
```
>D
>B
=>sensor53 r
>M 3
+1,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
1,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
+2,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
2,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
+3,12,o,0,9600,Strom
3,1-0:1.7.0*255(@1,P_in,W,P_in,2
#
```

Meter 1 delivers no values
```
>D
>B
=>sensor53 r
>M 3
+1,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
1,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
+2,12,o,0,9600,Strom
2,1-0:1.7.0*255(@1,P_in,W,P_in,2
+3,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
3,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
#
```

Meter 2 delivers no values
```
>D
>B
=>sensor53 r
>M 3
+1,12,o,0,9600,Strom
1,1-0:1.7.0*255(@1,P_in,W,P_in,2
+2,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
2,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
+3,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
3,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
#
```

Meter 2 delivers no values
```
>D
>B
=>sensor53 r
>M 3
+1,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
1,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
+2,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
2,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
+3,12,o,0,9600,Strom
3,1-0:1.7.0*255(@1,P_in,W,P_in,2
#
```

Meter 1 delivers no values
```
>D
>B
=>sensor53 r
>M 3
+1,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
1,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
+2,12,o,0,9600,Strom
2,1-0:1.7.0*255(@1,P_in,W,P_in,2
+3,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
3,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
#
```

All meters working (different configuration from above)
```
>D
>B
=>sensor53 r
>M 3
+1,12,o,0,9600,Strom
1,1-0:1.7.0*255(@1,P_in,W,P_in,2
+2,4,m,0,19200,SekHK,2,5,rF7030009000E,rF703001C0004,F703006A
2,F7031CSSss@i0:10,Außentemp.,°C,Temp_Outside,1
+3,14,kN2,0,1200,PriHK,13,10,3F1005003C00560057004A0044
3,3F10003Ckstr@i0:1000,Wärmemenge,MWh,HeatEnergyE1,3
#
```

gemu2015, 2022-11-15:
you are on an esp8266 and thus use all channels in software serial mode
this will not work.
software serial blocks when sending and so prevents any receiving (no full duplex)
if is also critical to use higher baudrates.

esp32 has 3 hardware serial UARTS and thus should work without issues

TheChatty, 2022-11-15:
But there are **two working configurations**. And whether it works or not only depends on the order of meter descriptions - there is **no** reconfiguration on the hardware side.

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015