gemu2015/Sonoff-Tasmota #44: Script problems when using the latest version

Sinned42, 2023-02-22:
Hallo gemu2015,

ich habe ein seltsames Verhalten in der letzten Version 12.4.0 mit Script und Size 8192 im 1M Esp01.
Im Gegensatz zu früheren Versionen gibt es einen Unterschied mit der Klammersetzung im mathematischen Bereich.

Hier als Beispiel (ich programmiere an ottelos Stromzählervisualisierung etwas mit):

Folgender Code (2x identische Anweisungen):

cstr2="cnth"+s(1.0(((hours+20)*120)+(mins*2)+int(secs/30))%2880+1)+"/120"
print %cstr2%

cstr2="cnth"+s(1.0(((hours+20)*120)+(mins*2)+int(secs/30))%2880+1)+"/120"
print %cstr2%

Ausgabe der Print Anweisungen:

16:22:00.376 cnth1/120
16:22:00.380 cnth1485/120

16:22:30.393 cnth2/120
16:22:30.398 cnth1486/120

Manchmal läuft alles gut durch, aber meistens zeigt sich diese komische Berechnung.

Als workaround habe ich folgendes gefunden (mehr Klammern):

cstr2="cnth"+s(1.0((((hours+20)*120)+(mins*2)+(int(secs/30)))%2880+1))+"/120"

Sollte aber eigentlich nicht sein, dass beim ersten Beispiel verschiedene Werte herauskommen, oder?

Vielleicht hast Du ja bereits eine Ahnung, was sich im Code verändert hat. Ansonsten Daumen hoch, gerade für die letzte Änderung mit 8192 bei 1M.

Viele Grüße,
Dennis

gemu2015, 2023-02-22:
Hi Dennis,
Also habe an diesem Teil von scripter seit einiger Zeit nichts verändert.
hast du mal versucht die Berechnung außerhalb der s() Klammer zu machen ?
kann mir den Fehler erst mal nicht erklären.
Wenn er aber nur manchmal, also zufällig auftritt klingt das nach Stack Problemen.
Beim ESP8266 ist der Stack und Heap sehr knapp bemessen und mit den 8kb nehmen wir einiges an Heap weg.
Druck dir im Zweifelsfall Stack und Heap aus um zu sehen wo du liegst, vermeide Subroutinen denn die kosten ca 600 Bytes Stack.
Falls du aber einen reproduzierbare Fehler findest versuche ich das zu reparieren.

Gruß
Gerhard

Sinned42, 2023-02-22:
Danke für deine schnelle Antwort :) 

Okay, ich versuche es mal die Tage, es außerhalb zu machen, wobei mein Ziel natürlich ein Einzeiler ohne weitere Variablen ist, was mir ja mit dem Workaround auch gelungen ist.

Heap ist bei mir bei 12-13k, Stack 1,2-1,3k und ram bei 12k. Ich kann allerdings mit den Angaben nicht so viel anfangen, was okay ist und was nicht, beim c64 hätte ich mehr Ahnung ;)

Viele Grüße, Dennis

Sinned42, 2023-02-22:
Hat mir jetzt keine Ruhe gelassen... ich habe es sehr vereinfacht....

Folgendes Beispiel:

tmp=((hours+20)*120)

print ohne minuten %tmp%

tmp=((hours+20)*120)+(mins*2)

print mit minuten %tmp%

Zuverlässige Ausgabe zur Reproduktion:

23:45:25.725 ohne minuten 5160.00
23:45:25.728 mit minuten -0.00

Die Variante mit Minuten ist hier 0. Auch wenn man weitere Klammern setzt, z.B. tmp=(((hours+20)*120)+(2*mins))

Mache ich hier einen Denkfehler?

Viele Grüße, ich hoffe ich träume jetzt nicht von Stunden und Minuten :) :)

Sinned42, 2023-02-22:
Nanu - der Code sollte lauten:

tmp=((hours+20)*120)

print ohne minuten %tmp%

tmp=((hours+20)*120)+(mins*2)

print mit minuten %tmp%

Hat der Editor was gefressen?

Sinned42, 2023-02-22:
`tmp=((hours+20)*120)

print ohne minuten %tmp%

tmp=((hours+20)*120)+(mins*2)

print mit minuten %tmp%`

Sinned42, 2023-02-22:
Okay, ich gebs auf... hier der Code

https://pastebin.com/1Eanb5KN

gemu2015, 2023-02-23:
ok prima das kann ich reproduzieren.

Kannst du sagen ab welcher Version der Fehler auftritt ?

gemu2015, 2023-02-23:
hab die letzte release gefunden die funktioniert 15.2.23
Da habe ich den integermode eingebaut.
Werd jetzt suchen was da schief läuft.

Danke für die Rückmeldung!

gemu2015, 2023-02-23:
Fehler gefunden und gefixt.
War eine uninitialisierte Variable, deshalb der Zufall je nachdem was da auf dem Stack liegt.

Sinned42, 2023-02-23:
Super, das ging ja schnell, danke 😃

Sinned42, 2023-02-23:
Gleich mal geflasht, super Arbeit, läuft wie es soll...

11:10:45.166 ohne minuten 3720.00
11:10:45.170 mit minuten 3740.00

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015