gemu2015/Sonoff-Tasmota #47: I2C display not showing end of dt command correctly in scripting

pkkrusty, 2023-04-15:
### PROBLEM DESCRIPTION
Using OLED SSD1307 display with dt commands results in flashing of final character when same line is written to multiple times

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP32-WROOM32
- [ x] Tasmota binary firmware version number used: 12.4.0.4
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: VSC
- [ x] Flashing tools used: ESP-Flasher
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
 00:08:25.431 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:08:25.504 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:08:25.611 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"225":"Relay2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"672":"SPI MISO1"},"GPIO13":{"704":"SPI MOSI1"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"226":"Relay3"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
00:09:24.240 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Ousseynou BMS","FriendlyName":["Solar","Load","Display",""],"Topic":"pkkrusty_E56B00","ButtonTopic":"0","Power":11,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:09:24.250 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:09:25","StartupUTC":"","Sleep":50,"CfgHolder":4802,"BootCount":28,"BCResetTime":"1970-01-01T00:00:00","SaveCount":64}}
00:09:24.258 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0.4(tasmota32)","BuildDateTime":"2023-04-10T14:12:58","Core":"2_0_7","SDK":"4.4.4.20230218","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"377/699"}}
00:09:24.264 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":60,"Resolution":"558181C0","SetOption":["00208000","2805320001000600003C5A00192800000000","00804009","00006000","00004000","00000000"]}}
00:09:24.274 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":977,"Free":1856,"Heap":166,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","0520048E","00080121","00000043","00000400","40000810","00000080","40200000","00000000","00000000"],"Drivers":"1,2,4,9,10,13,29,38,50,56","Sensors":"1,2,5,54,127","I2CDriver":"4,26,35"}}
00:09:24.282 RSL: STATUS5 = {"StatusNET":{"Hostname":"ousseynou-bms","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"C0:49:EF:E5:6B:00","IP6Global":"","IP6Local":"","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
00:09:24.285 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:09:24","Local":"1970-01-01T00:09:24","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00"}}
00:09:24.292 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:09:24","Battery":{"Cell-1":3.645,"Cell-2":3.648,"Cell-3":3.652,"Cell-4":3.578,"Cell-5":0.000,"Cell-6":0.000,"Cell-7":0.000,"Cell-8":0.000,"Cell-9":0.000,"Cell-10":0.000,"Cell-11":0.000,"Cell-12":0.000,"Max":3.652,"Min":3.578,}}}
00:09:24.295 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:09:24","Uptime":"0T00:09:25","UptimeSec":565,"Heap":166,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":0,"POWER1":"ON","POWER2":"ON","POWER3":"OFF","POWER4":"ON"}}
```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
No applicable weblog output
```

### TO REPRODUCE
Enter the following in scripting:
```
dp3
dt [s1c1l1f5] A=%curr%, V=%volt% 
dp0
dt [c1l2] M=%maxc%
dp3
dt [c4l2] :%max%
dp0
dt [c10l2] , m=%minc%
dp3
dt [c15l2] :%min%
```
Display shows everything correctly except the last character of 2nd, 3rd and 4th dt command. It flashes and disappears.

### EXPECTED BEHAVIOUR
Display shows all text correctly without disappearing.

### SCREENSHOTS
https://user-images.githubusercontent.com/79770016/232209317-3f9f1313-bcb3-4e6c-bd18-2ec1ed99beef.MOV

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-04-15:
you better should disable auto draw with this display to avoid flickering
also you can set the number of decimals by a number preceding the variable

an example from a device in my house:

>B
; clear display disable auto draw
dt [zf1s1D0]

>S
dt [l1c1] temp=%2aztemp%
dt [l3c1] hum=%0azhumi%
dt [l5c1] tvoc=%1aztvoc%
dt [l7c1] eco2=%3azeco2%
; draw
dt [d]

pkkrusty, 2023-04-15:
Preceding the variable works.

```
dt [s1c1l1f5] A=%3curr% V=%2volt% 
dt [c1l2] M=%0maxc%:%3max% m=%0minc%:%3min%
dt [c1l3] %0updy%T%tmvar% H=%0heap%
```

Autodraw enabled works fine, no flickering.

Thanks

closed 2023-04-15 by pkkrusty