gemu2015/Sonoff-Tasmota #48: Rolling log using scripting file system

pkkrusty, 2023-04-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Would like to log data to flash space, but need to limit file size to available flash in ESP32 (300KB). Thus need rolling log to add new data and delete oldest data. Data is tab/carriage return delimited.

**Describe the solution you'd like**  
Some help with the code for that. Beyond my abilities at the moment, but maybe needs additional functions? If you could add an example in the scripting docs, I think that would be a very helpful starting point for people to use for their customs logging code.

**Describe alternatives you've considered**  
I could delete the whole file and start over when it gets too big, but that isn't as elegant.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-04-27:
this is not supported by standard file systems and can only be used by copying to a new file
here is a working sample
(you need my latest fork since i had to implement read line until LF: fr(str fdr 1), the 1 indicates read entire line until LF, not only until TAB or comma like standard string read)

probably i will create a special cmd later to make this easier.

```
>D 40
IP=192.168.188.106
;must assure that longest line fits in one string
fdr=0
fdw=0
res=0
cnt=0
str=""

>B

; subroutine write to log
#tolog(str)
fdr=fo("/log.txt" a)
if fdr>=0 {
	; append new entry from str
	fw(str fdr)
	res=fz(fdr)
	if (res>3000) {
		; shorten file by 1. entry
		; seek to beginning of file
		fs(fdr 0)
		; read first string until LF and throw away
		fr(str fdr 1)
		fdw=fo("/tmp.txt" w)
		for cnt 1 1000 1
			res=fa(fdr)
			if res>0 {
				fr(str fdr 1)
				fw(str+"\n" fdw)
			} else {
				break
			}
		next
		; close files
		fc(fdr)
		fc(fdw)
		; delete log
		fd("/log.txt")
		; rename tmp to log
		frn("/tmp.txt" "/log.txt")
	} else {
		fc(fdr)
	}
}

>S
=#tolog(tstamp+"\n")

```

pkkrusty, 2023-04-27:
Thanks, will try it.

pkkrusty, 2023-04-27:
What's the limit for the for...loop in terms of cycling through the file? If I have 8000 lines, for a total of ~300k, (38 bytes per line), will I run into buffer/RAM/speed issues? I'll try it, but you may have some thoughts...

pkkrusty, 2023-04-27:
Also need #define USE_SCRIPT_FATFS?

pkkrusty, 2023-04-27:
Compiling for ESP32 gives error:

```
/Users/andrew/Tasmota/Tasmota-12-5-0-1/tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino: In function 'int32_t script_logfile_write(char*, char*, uint32_t)':
/Users/andrew/Tasmota/Tasmota-12-5-0-1/tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino:5930:17: error: invalid conversion from 'char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
```

gemu2015, 2023-04-27:
fixed

closed 2023-07-01 by pkkrusty