gemu2015/Sonoff-Tasmota #51: I2S Audio driver questions and suggestions

blakadder, 2023-06-07:
### PROBLEM DESCRIPTION

Currently I2S Audio driver requires configuring I2S DOUT even if you only have a microphone connected, otherwise it crashes Tasmota. As it is not intuitive to users and wastes a GPIO pin this requirement should be removed because there are a number of devices that have only a microphone.

I2S Audio is very complex but the current driver selections are limited and bogged down by device specific defines that might or might not apply to other devices as well.

In my case I have devices (like [Seeed XIAO Sense](https://templates.blakadder.com/seeedstudio_XIAO_ESP32S3_SENSE.html)) with a PDM microphone for which I submitted a [PR](https://github.com/arendst/Tasmota/pull/18807) with a rough enable that tries not to break existing setup

In another unrelated case I have an INMP441 I2S MEMS microphone that only works with specific settings and one of them is `I2S_CHANNEL_FMT_ONLY_LEFT`. When trying to add it as an option with MIC_CHANNELS I ran into the issue of `mic_channels` being used to calculate other values so I couldn't use number 0 for mono left only option. 
In my local code I've added the microphone with a new #define section 

```c
#ifdef INMP441
  i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX);
  i2s_config.communication_format = I2S_COMM_FORMAT_I2S;
  i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
#endif
```
positioned below [line 100](https://github.com/arendst/Tasmota/blob/73c48ff5f0b3a1c07799196711ad59a60f9cc667/tasmota/tasmota_xdrv_driver/xdrv_42_1_i2s_mp3mic.ino#L100)

As far as user friendliness goes my view is that creating defines for each microphone model would be best path going forward (I've got a few more devices with different mics I need to get working) and not have them tied at all to I2S input.

I've also have a few analog microphones that I read can also be used through the I2S component, do you have info on how to use them with Tasmota's I2S driver before I do a deep dive on those as well?

Also, can you go through the revised docs and see if the information is correct? https://tasmota.github.io/docs/I2S-Audio/

On a semi-related note it would be nice to have a "sound sensor" feature for I2S microphones to make them more useful in general home use, something like https://github.com/ikostoski/esp32-i2s-slm but without the need for graphical display, just a text dB(A) sensor

gemu2015, 2023-06-09:
i am currently extremely busy with other things, so it will take some time until i can solve some issues with this driver.
1. since i always use i2s output in my devices i didn't design a mic only driver up to now. 
it is not completely trivial to change this, but i will think of it.
2. the selection of the used microphone with defines as you did seems a be the best solution. 
3. i also use analog microphones (in S3 box and external audio board) they always need an i2s audio codec (ADC,DAC). i implemented 5 of these codecs (see libesp32_audio). analog mics have better audio quality and you may adjust analog audio gain.
4. a sound sensor would be nice to have. it should be not too difficult to implement. it could use strong undersampling to save processing time, as well as dma.  the above example project  goes much further and would need much more processing time.
 i think this is effort is not really needed.

blakadder, 2023-06-09:
> i am currently extremely busy with other things, so it will take some time until i can solve some issues with this driver.

No worries, I've managed to add the necessary features. Whenever you get round to it. There's also all of the defines and references to some device specific functions that don't exist in Tasmota's code anymore (specifically CORE2) which need to be revised.

> 1. since i always use i2s output in my devices i didn't design a mic only driver up to now.
>    it is not completely trivial to change this, but i will think of it.

Ok, I've noted that DOUT is still required when using only mic in the documentation for now

> 2. the selection of the used microphone with defines as you did seems a be the best solution.

Will continue doing that since I ordered every I2S mic if could find on aliexpress :D

> 3. i also use analog microphones (in S3 box and external audio board) they always need an i2s audio codec (ADC,DAC). i implemented 5 of these codecs (see libesp32_audio). analog mics have better audio quality and you may adjust analog audio gain.

Will explore further with a couple I have on a breakout board.

> 4. a sound sensor would be nice to have. it should be not too difficult to implement. it could use strong undersampling to save processing time, as well as dma.  the above example project  goes much further and would need much more processing time.
>    i think this is effort is not really needed.

Whatever takes the least amount of resources sounds good.

pkkrusty, 2023-10-05:
> Will continue doing that since I ordered every I2S mic if could find on aliexpress :D

Any conclusions about which I2S mic is best?

gemu2015, 2026-10-28:
Closing this as outdated. It dates from before the current state of this fork (the scripting / driver code it refers to has changed a lot, and the fork was just merged with the current Tasmota development branch, 27.10.2026), and there has been no activity for a long time. If the problem still exists with the current `testing` build (https://github.com/gemu2015/Sonoff-Tasmota/releases/tag/testing), please open a new issue with the version and the steps to reproduce.

Wird geschlossen, weil veraltet. Falls das Problem mit dem aktuellen `testing`-Build noch besteht, bitte ein neues Issue mit Version und Schritten zum Nachstellen oeffnen.

closed 2026-10-28 by gemu2015