gemu2015/Sonoff-Tasmota #55: ModbusSend with SML driver

pkkrusty, 2023-10-19:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Would like to write certain registers in addition to reading using the Smart Meter scripting section >M

**Describe the solution you'd like**  
A way to write a register while not disturbing the regular polling of registers that the >M section is doing.

**Describe alternatives you've considered**  
I tried in the >S section
```
so(3 1 9600 8N1)
smw(40126 1 1000)
sc()
```
but nothing happened.

**Additional context**  
Documentation states: smw(ADDR MODE NUMBER) send a value with checksum to MODBUS Address, MODE 0 = uint16, 1 = uint32, 3 = float. Does Modbus Address mean the modbus ID? Or ID plus function plus register? 011640126? Or is it NUMBER that holds the function and register and value? i.e. `smw(1 1 0x01109CBE)`

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-10-19:
this is meant for a modbus slave, but you want to write a cmd in modbus master mode. (sml modbus driver)

here is an example of writing to a modbus register in SML mode

in this case the power limit register of a growatt inverter

```
>D 22

cstr=""
gl=0
tmp=0

>B
=>sensor53 r

>S
if chg[gl]>0 {
	; change limit
	tmp=int(gl/42)
	cstr="r0106000300"+hn(tmp)
	sml(1 3 cstr)
}

>M 1
+1,18,m,0,9600,GRW,19,5,01040026,01040028,01040005,01040009,01030003
1,010404UUuu@i0:10,Netzspannung,V,mainsv,1
1,010404xxxxUUuu@i0:10,Einspeisestrom,A,mainsc,1
1,010404UUuuUUuu@i1:10,Einspeiseleistung,W,mainsw,1
1,010404UUuuUUuu@i2:10,string 1 unten,W,s1w,1
1,010404UUuuUUuu@i3:10,string 2 oben,W,s2w,1
1,010304UUuu@i4:1,limit,%,limit,0
#

>W
<hr>
nm(1000 3600 10 gl "Growatt limit (W) " 80 0) 
```

pkkrusty, 2023-10-19:
In this example, you are using `r0106000300`.  Is `r` necessary, or does it follow the syntax of the meter definition? i.e. I want to send 32-bit uint (in this case 1000) to register 40126 with function code 16, I assume i would enter 
`hexstr="01169CBE" + hn(1000)`
`sml(1 3 hexstr)`

gemu2015, 2023-10-19:
r is necessary in this example also a 16 bit unit is sent