gemu2015/Sonoff-Tasmota #56: Canbus SML driver

pkkrusty, 2023-11-07:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Can the Canbus driver be used as a generic Canbus sniffer? If so, what is the expected interface? MCP2515? Perhaps limited to ESP32? Level-shifted serial direct to the Canbus H/L lines?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-11-07:
Esp32 native canbus with external level shifter is recommended as mcp2515 is to slow and will miss Pakets
It can be used as sniffer
See example Huawei Lipo charger in sml docs

gemu2015, 2023-11-07:
can bus transceiver for esp32

https://eckstein-shop.de/Adafruit-CAN-Pal-TJA1051T-3-CAN-Bus-Transciever-Native-CAN-Peripheral?googlede=1&ws_oss_lieferland=DE&gclid=EAIaIQobChMI2uub_b2yggMVFZiDBx2wpgbyEAQYAiABEgL7ivD_BwE

closed 2024-03-13 by pkkrusty