gemu2015/Sonoff-Tasmota #57: Smart Meter custom registers

pkkrusty, 2023-11-10:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I'd like to iterate through a bunch of registers to map what is available (documentation is not correct for the inverter I am communicating with). So I tried to define the meter with the following, but it doesn't work. `>M 1
+1,5,m,0,9600,modbus,1,10,%astr%`
Where I have built astr to be a normal Modbus command like r01037D500008

Is there already a method to do this?

**Describe the solution you'd like**  
Poll consecutive registers programmatically instead of adjusting the script. 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-11-10:
variable substitution is not supported in meter definition line.

define meter with a least one register like
+1,5,m,0,9600,modbus,1,5, r01037D500008

then e.g. in >S send more register requests with 

>S
; modify cstr programmatically 
cstr="r0103"+hn(v1)+hn(v2)+"0008"
sml(1 3 cstr)

you may then watch the responses in dump mode with sensor53 d1

pkkrusty, 2023-11-11:
That works great, thanks.

closed 2023-11-11 by pkkrusty