gemu2015/Sonoff-Tasmota #67: Verbesserung der Lesbarkeit der Dokumentation für Scripting

mi-hol, 2025-04-27:
_Originally posted by @mi-hol in [#6](https://github.com/ottelo9/tasmota-sml-script/issues/6#issuecomment-2715343278)_

Hier als Erinnerung für weitere Verbesserungen angelegt um das Issue in nicht passenden Repository zu schliessen.

> Dort sind die Scripte "eingeklappt" und stören beim Scrollen nicht.
> Das wäre meiner Meinung nach die bessere Option auf bei der Scripting Doc für alle Scripte.
> Da würden auch die sehr grossen Scripte nicht stören.
> 
> Aus meiner Sicht ist "nicht stören" der falsche Ansatz für das erlernen neuer Fähigkeiten.
> "eingeklappter" code funktioniert für die im Durchschnitt nur ca 15 Zeilen langen Skripte gut, aber schon bei dem folgenden Beispiel nicht mehr gut, da nur Teile des codes in das sichtbare Fenster passen.
> Der Rest wird sowohl vertikal, als auch horizonal, abgeschnitten! 
> Wenn ich aber unbekannten code verstehen will, brauche ich die vollständige und spezielle Ansichten, das geht aber nMn nur sinnvoll in einem Editor (zB VS Code).
> 
> Habe gerade extrem wenig Zeit und kann mich nicht selbst darum kümmern.
> 
> Meine Hilfe habe ich ja angeboten, wie kommen wir da weiter?
> 
> Schlechtes visuelles Beispiel für "embedded code":
> 
> [image]
> 
>

mi-hol, 2025-04-27:
Kleine Verbesserung durch "replace dropbox with github link" siehe Details in https://github.com/ottelo9/tasmota-sml-script/pull/9

gemu2015, 2025-10-20:
es haben sich so gut wie keine user für diese komplexen scripte interessiert. deshalb habe ich das nicht weiter verfolgt zumal das Tasmota Team nur Berry Scripting haben möchte.
Ich habe ja meine IObroker Installation auf einem Raspberry komplett eingestellt und alles auf ESP implementiert inclusive Datenbank. Das läuft bei mir schon seit Jahren. Das hat aber keine Nachahmer gefunden.
Wenn du willst kannst du gerne die scripting Doku aufräumen.
Allerdings kann ich nicht alle Sourcecodes veröffentlichen da dort teilweise interne Referenzen vorkommen die man zuerst unkenntlich machen müsste.

Pewibe, 2025-10-20:
> Wären es denn machbar die [Stromzähler Scripte von Ottelo9](https://github.com/ottelo9/tasmota-sml-script) auf Berry zu konvertieren ohne Funktionen zu verlieren?
Ich bin nicht direkt von eurer Diskussion betroffen, bin aber als follower von @gemu2015 darauf aufmerksam geworden. Ich habe selbst Strom- und Gaszähler per Scripting in Betrieb und beschäftige mich seit Wochen mit Berry. Fazit von mir ist, dass Gerhard (zu der entsprechenden Zeit) tolle Arbeit geleistet hat, aber der Zahn der Zeit Richtung ESP32 und Berry ab geht. Beide (Scripting und Berry) uptodate zu halten wäre Resourcen-Verschwendung. So wird die Transformation nach Berry sinnvoll und dringend. Daher hoffe ich, dass Gerhard die Muße findet, sein Baby mit Berry zukunftsfähig zu machen. 
Das schreibe ich als moralische Unterstützung ohne Kenntnis, ob einfache Scripte schon in einer Berry-Version existieren. Zu den aufwendigeren Scripten von Ottelo sollte man ein Script herauspicken und beim Umschreiben in Berry feststellen, wo es vielleicht hakt.
Just my 2 cents

mi-hol, 2025-10-20:
> Zu den aufwendigeren Scripten von Ottelo sollte man ein Script herauspicken und beim Umschreiben in Berry feststellen, wo es vielleicht hakt.

@pewibe das ist ein sehr guter Vorschlag, ich habe https://github.com/ottelo9/tasmota-sml-script/blob/main/2_SML_Script_Chart_PV.tas im Einsatz
Welches verwendest Du?

Pewibe, 2025-10-20:
Die Frage nach den Scripts führt uns (in meinem Fall) nicht weiter, da ich selbstgeschriebenes (z.B. für einen Gaszähler) verwende, das keine schönen Diagramme produziert, was die meisten wohl haben wollen: 
[Tasmota Gasmeter 3.42 (script).txt](https://github.com/user-attachments/files/23010294/Tasmota.Gasmeter.3.42.script.txt) 
[image]
So kommen wir zurück auf die Suche nach einem Freiwilligen, der ein Script in Berry transferieren könnte. Hier kämen zuerst Personen in Frage, die den größten Druck nach einer Script-Transformation haben oder Gerhard in der Hoffnung, dass es ihm als Insider am leichtesten von der Hand geht. :-) 
Ich selbst bin momentan noch im Berry-Lernstadium und mit einer Steuerung für eine Wasserstandsregelung mit Berry beschäftigt. Danach (Zeitpunkt noch unbekannt) würde ich sogar mein o.a. Script in Berry umsetzen, um das Gelernte anzuwenden und das alte Script loszuwerden. Ich bin sicher, das geht, hilft aber dir in deinem Script nicht komplett.
Schwieriger wird es mit /meinem Stromzählerscript, was zumindest vom Kern eher deinem Script entspricht. Hier weiß ich heute nicht, wie man die Meter-Section >M in Berry umsetzt, bzw. nehme an, es geht nicht; hier wird wohl die Hilfe von @gemu2015 benötigt. Warten wir ab, bis er sich hierzu nochmals meldet.

mi-hol, 2025-10-21:
Das wichtigste neue Dokument für TAS scheint mir ein Äquivalent für [.doc_for_ai/FOR_DEVELOPERS.md](https://github.com/arendst/Tasmota/blob/development/.doc_for_ai/FOR_DEVELOPERS.md) zu schreiben.
@gemu2015 wie siehst Du das?

gemu2015, 2025-10-23:
USE_FEXTRACT braucht ca 2kB

"For ESP32 builds it is recommended to use [Berry](https://github.com/tasmota/docs/blob/master/docs/Berry)"

stammt nicht von mir, ist plötzlich mal drin gewesen. 

Es gibt einige die Berry als Standard durchsetzen wollen und scripter als Konkurrenz empfinden.

mi-hol, 2025-10-23:
> rules sind so gut wie immer dabei. Das hat Theo "erfunden" und deshalb da es nur alternativ zu scripter geht ist er nicht für scripting

Bin nicht sicher, dass ich die Antwort vollständig verstehe. Darf ich dich deshalb bitten, diese Aussage als Korrektur der obigen Tabelle zu posten? Damit "wandert" es dann auch automatisch richtig in die Doku :)

Vergiss bitte nicht auf https://github.com/gemu2015/Sonoff-Tasmota/issues/67#issuecomment-3437246763 zu antworten

gemu2015, 2025-10-23:
es gibt fairerweise auch eine ganze Menge von Funktionen in Berry die scripter nicht kann.
z.B. matter geht nur mit Berry. (auch weil Tasmota die Matter Implementierung von Espressif ignoriert)
da man aber problemlos Berry mit scripter zusammen benutzen kann, kann man das beste von beiden nutzen wenn man will.
Da Berry vom Tasmota Team gepusht wird, werden hauptsächlich nur SML user noch scripter nutzen.
Das ist für mich gar kein Problem. Ich hatte das alles eigentlich nur für mich selbst entwickelt (auf meinem Fork) und erst auf Aufforderung einiger User bei Tasmota selbst integriert.

das Anpassen des sml Treibers an Berry ohne scripter ist erheblich komplexer als @Pewibe sich das vorstellt.
es gibt jede Menge Meter die mehr als nur den einfachen descriptor benötigen.
da das für die meisten user mit scripter problemlos funktioniert und bei dieser Anwendung Berry keine Vorteile bietet, habe ich Null Motivation da Arbeit rein zu stecken. 

Es ist überhaupt gar keine Frage dass Berry ein moderne und tolle Programmiersprache ist und Stephan Hadinger ist ein Programmier Genie.  Dass er allerdings Matter in Berry implementiert hat ist mir ein Rätsel da es eine native Implementierung von Espressif gibt die sicher wesentlich schneller ist.
Man muss jetzt beim Kompilieren auch Berry Code kompilieren und den Bytekode mit den C quellen zusammen linken.

Berry sollte genau wie scripter nur für spezielle Aufgaben zusammen mit der fertigen Binary genutzt werden.

scripter ist extrem kompakt auch in der syntax weil es eben schon lange vor Berry auf ESP8266 mit sehr wenig Resourcen laufen sollte. Für mich ist das allerdings nicht kryptisch da ich  viel mit Linux gemacht habe und so etwas gewohnt bin.
Ich habe schon die allerersten CPUs in den 70 Jahren programmiert damals noch in Maschinenkode nicht mal in Assembler.
Ich konnte den Maschinenkode auswendig. Das war kryptisch!
Dann habe ich fast 10 Jahre nur in Assembler programmiert da die damaligen Hochsprachen einfach nur schlecht waren.
Daneben habe ich sicher einige Hundert Hardwareprojekte entwickelt (digital und analog) mit den verschiedensten CPUs.

Im übrigen mache ich nur noch sehr wenig mit Tasmota da meine Hausautomation ausgereizt ist.
nur auf Anfrage von usern mache ich kleine Änderungen oder Korrekturen zuletzt die Anpassung an diverse Solar Akkus wie Marstek.

Bin auch nicht so engagiert wie einige vom Tasm Team die offensichtlich Tag und Nacht nichts anderes zu tun haben und scheinbar keine Familie haben.

Bin auch nicht mehr der jüngste (74) und habe auch genug andere Hobbies

Pewibe, 2025-10-23:
Gerhard war mit dem letzten Beitrag etwas schneller als ich. Das war sehr ehrlich und gut alles beschrieben. Ich oute mich auch mal als Jüngling mit 72 und ähnlichem Background wie Gerhard. Ich habe auch schon eine Scriptsprache "erfunden" und verwendet, muss aber auch zugeben, dass ich langfristig keinen im Sinne der Sprache, nicht der Funktion begeistern konnte. Deswegen verstehe ich dich sehr gut, dass du keine Lust zum Recoding hast. Irgendwann über kurz oder lang kommt einer und beschenkt uns mit der "moderneren" Version.
 
Sorry für meinen versteckten Humor! :-) Ich habe mal claude.ai die Frage gestellt:
**Gibt es Applikationen, die nur mit scripter, nicht mit berry gehen? Soll man noch scripter lernen oder lieber gleich berry?**

**Benutzte Quellen:** Tasmota Berry vs Scripter comparison capabilities 202410 ErgebnisseBerry - Tasmotagithub.ioScripting Language - Berry Scripting Language differences · arendst/Tasmota · Discussion #18700github.comScripting & Automation | arendst/Tasmota | DeepWikideepwiki.comUse of Berry-Script and Scripting together with tasmota · arendst/Tasmota · Discussion #17086github.comScripting Language - Tasmotagithub.ioFirst impressions using Berry Scripting Languagegoogle.comTasmota and berry — LVGL documentationlvgl.ioTasmota (32, Berry)eichelsdoerfer.netGlobal variables in Rules and Berry Scripting · arendst/Tasmota · Discussion #20961github.comTasmota32 und Berryeichelsdoerfer.net

Basierend auf den aktuellen Informationen hier meine (claude-ai's) klare Empfehlung:
**Lerne Berry, nicht Scripter!**
Gründe für Berry:
Berry ist die nächste Generation des Scriptings für Tasmota und wird aktiv weiterentwickelt [Tasmota](https://tasmota.github.io/docs/Berry/)
Für ESP32-Builds wird empfohlen, Berry zu verwenden [GitHub](https://github.com/arendst/Tasmota/discussions/20503)

Berry kann mehr:

- Objektorientierte Programmierung mit Klassen [Tasmota](https://tasmota.github.io/docs/Berry/)
- Volle LVGL-Unterstützung für GUIs auf ESP32 [Vanderwulp](https://michiel.vanderwulp.be/domotica/Nodes/Thermostat/berry-fragments.html)
- Besserer I2C-Zugriff für Custom-Hardware-Treiber [Tasmota](https://tasmota.github.io/docs/Berry/)
- Moderne Syntax ähnlich Python
- Dateisystem-basiert (.be Dateien)
- Aktive Community und Entwicklung
- 

Scripter ist Legacy:

- Nur für ESP8266 und einfache Aufgaben gedacht
- Begrenzte Variablen (45 numerische, 5 Strings) [GitHub](https://github.com/arendst/Tasmota/discussions/20503)
- Primitiver Syntax
- Wird kaum noch weiterentwickelt
- Scripter und Rules schließen sich gegenseitig aus [Tasmota](https://tasmota.github.io/docs/Buttons-and-Switches/)
- 

Nur Scripter kann (aber selten nötig):

- Smart Meter Language (SML) - direkte Unterstützung für Stromzähler-Protokolle [Tasmota](https://tasmota.github.io/docs/Commands/)[GitHub](https://github.com/arendst/Tasmota/discussions/20503)
- Aber: Berry kann die SML-Daten trotzdem über tasmota.read_sensors() auslesen [Tasmota](https://tasmota.github.io/docs/Commands/), wenn Scripter die Daten verarbeitet hat.

Fazit:

- ESP8266: Rules (einfach) oder Scripter (komplex)
- ESP32:   Berry (immer die beste Wahl!)
- Du hast einen ESP32 → Nutze Berry!
- **Berry kann praktisch alles was Scripter kann, aber viel mehr und besser. Die Zeit, die du in Berry investierst, ist besser angelegt als Scripter zu lernen.**

gemu2015, 2025-10-23:
KI ist sicher lustig, habe schon einiges damit gemacht auch mit lokaler Installation.
Leider sehr vieles einfach falsch.

so auch hier
Besserer I2C-Zugriff für Custom-Hardware-Treiber  ->Falsch geht bei scripter sehr gut
Nur für ESP8266 und einfache Aufgaben gedacht ???
Begrenzte Variablen (45 numerische, 5 Strings) 
falsch, nur durch Speicher begrenzt, mag sein dass die Doku noch veraltet ist ???
fast alles was ich in Berry gesehen habe ginge auch in scripter.

Berry kann praktisch alles was Scripter kann, aber viel mehr und besser. Die Zeit, die du in Berry investierst, ist besser angelegt als Scripter zu lernen.

stimmt nur zum Teil, aber sicher ist es für Neulinge erheblich vorteilhafter Berry zu lernen statt scripter.

Darum geht es ja auch hier nicht.

Aber z.B. Multitasking geht prinzip bedingt nicht mit Berry aber mit scripter.
Nutze ich oft

mi-hol, 2025-10-23:
> Bin auch nicht mehr der jüngste (74)

> Jüngling mit 72

Dann bin ich wohl das Küken mit meinen 64 :)

mi-hol, 2025-10-23:
> Benutzte Quellen:

@Pewibe wärst Du so nett obigen Text zu überarbeiten, so dass die Quellen klickbare links wären?
Dann könnte ich nach manueller Prüfung, das relevante in die Doku übernehmen.

mi-hol, 2025-10-23:
> Die Links - Strafarbeit!

das ist doch was für Claude, oder nicht?

mi-hol, 2025-10-23:
@Pewibe Danke für die Links!

> Benutzte Quellen: https://deepwiki.com/arendst/Tasmota/7-scripting-and-automation

ist ja geil!

DeepWiki beschreibt auch TAS unter https://deepwiki.com/arendst/Tasmota/7-scripting-and-automation#scripter-module mMn nach ganz gut.
@gemu2015 stimmen die gemachten Aussagen mit deinen Kenntnissen überein?

mi-hol, 2025-11-26:
@gemu2015 folgendes widerspricht sich mMn:

vns = total size of name strings in bytes (may not exceed 256) or #define SCRIPT_LARGE_VNBUFF extents the size to 4095 (default)  

was ist da richtig?

gemu2015, 2026-02-19:
@mi-hol , @Pewibe 
Vielleicht interessiert es euch was ich mit Claude OPUS 4.6 innerhalb von 12 Stunden zu Stande gebracht habe.
Habe Claude angewiesen einen C Compiler in Javascript zu schreiben mitsamt Editor, Syntaxcolorierung etc der problemlos auf jeden ESP32 und sogar auf ESP8266 läuft (Bytecode Compiler).  Das Ergebnis ist extrem beindruckend. Ich hätte niemals vermutet wie leistungsfähig die KI im Bezug auf Programmierung bereits ist. Tinyc ist einen neue Option für Leute denen scripter zu kryptisch ist oder die wie ich Python und damit Berry nicht mögen. Und die KI erstellt gleichzeitig eine vollständige Dokumentation und Beispieldateien. Probiert es aus und wundert euch. Das wird Folgen haben für die Art wie in Zukunft programmiert wird. Das wäre normalerweise ein Arbeit von Wochen oder Monaten gewesen.